revk/xESP32-Faikout #110: Work With ESP-Home

fantasiataiwan, 2023-09-19:
Can Faikin work with ESP-Home?

revk, 2023-09-19:
The hardware is just an ESP32 with serial at right levels. If esp home can run on ESP32 and handle inverted UART to a Daikin then I see no reason why not. You would need a suitable serial lead to load code (see Tasmotiser).

joshbenner, 2023-09-19:
I use [esphome-daikin-s21](https://github.com/joshbenner/esphome-daikin-s21) on a Faikin board. README there includes the config I use for dealing with the serial invert.

fantasiataiwan, 2023-09-28:
I tried to flash esphome and your v1.bin but it's not working.
I haven't see the hotspot SSID maybe I have something wrong haha.

revk, 2023-10-28:
The board is pretty generic, so maybe raise this with the esphome designers. The schematic for this board is published.

closed 2023-10-28 by revk

shikousa, 2023-11-26:
@revk What is the reason that your board is inverting the TTL-UART?

revk, 2023-11-26:
The reason is down to the fact the Daikin works at 5V. Using a FET each way is an inexpensive solution to allow that to work with the 3.3V of the ESP32. It has the side effect of inverting, but as the ESP32 has hardware tx/rx invert options it is no problem.

shikousa, 2023-11-26:
I was thinking of an ADUM1201 to also get isolation. There is the 3,3V from the ESP32 for VDD1 of the ADUM1201. On PIN1 of the S21 socket is there always 5V? This would be for VDD2 of the ADUM1201. Then the ADUM1201 might also be an approach, what do you think?

revk, 2023-11-26:
Most 3.3-5V converters expect a 5V supply. The issue is the Daikin units and wiring vary so I can't be sure to always have a 5V available. It does seem the signal to the Daikin does have internal pull up though, so this circuit has been found to work well.

shikousa, 2023-11-26:
In your schmatic you take PIN1 for having the 5V for PIN3 (RX). What if these 5V are not available?

revk, 2023-11-26:
Yes, that is just in case. What seems to be the case is the data to the Daikin always has a pull up, so just the FET pulls down. And sometimes there is 5V (not on my units, for example). By having both there is a good chance of it working with all models (that have S21 or X50A). So far it seems to be working well.

shikousa, 2023-11-26:
Thanks. It would be interesting how other gateways have solved this. The Zennio KLICDDv3 has KNX, actually it should have a galvanic isolation.

revk, 2023-11-26:
I did not consider isolation necessary, but I understand some Daikin have non isolated S21 like ports but they then sell an adaptor to isolate. It would obviously be more complex to add isolation.

shikousa, 2023-11-26:
The power consumption of an ESP32 is pretty high, recommended is a supply with 500mA. Do you know if PIN4 (>5V) can deliver this long term.

revk, 2023-11-26:
My understanding is that the 5V cannot deliver the power needed.

shikousa, 2023-11-26:
Yes, but you are using PIN4 with e.g. 12V. Do you know what this PIN can deliver?

revk, 2023-11-26:
We have had no problem with the 12V (often reported and 14V, or some such) powering the ESP32.