revk/xESP32-Faikout #111: Temperature of external sensor not shown/used

andreas-bulling, 2023-09-19:
I've flashed one of the "Mi Temperature and Humidity Monitors" and it is found by Faikin and showing the measured temperature. However, this temperature is not forwarded to HomeAssistant it seems and, consequently, also not shown nor can it be used as a reference when setting temperatures.

Would be great if that worked :)

revk, 2023-09-19:
There is a setting to use it as reference, which you also have to pick (asks for tolerance) and what target you want. It then tells the Daikin high or low to get the target temp, so what goes to HA may be rather confusing.

There is probably scope for improving the links to HA. At present HA just sees what the Daikin sees, which is its view of temp and target, not the external temp or logical target.

It may make more sense when in this mode for HA to see, and be able to set, the "logical" target we are using, and see the external reference.

Maybe when I am over this bout of covid.

andreas-bulling, 2023-09-19:
you mean "Remote BLE temperature sensor"? This I have set.

revk, 2023-09-19:
I mean "Track" and when it appears the temperature target slider.

andreas-bulling, 2023-09-19:
Is there a documentation somewhere what "Track" does?

revk, 2023-09-20:
I thought so, but I will look.

It is the "Faikin auto mode" in effect. It controls the Daikin setting hot/cold and a high/low target to aim for a specific temperature. It can use a BLE sensor as the target for that or the built in temp sensor if no BLE sensor set.

andreas-bulling, 2023-09-20:
Oh, that is the auto feature?! Good to know but curious that it's disabled by default (if I am not mistaken). Isn't that the most important feature/what users want - set a target temperature and the AC maintains that temperature?

FYI: When using the thermostat card/entity/whatever in HA and set a target temperature there it's not this one. It's the temperature in the top part of the Faiking configuration screen. Or do I misunderstand?

revk, 2023-09-20:
Your aircon has an auto setting, and it uses its idea of room temperature and probably has a very wide margin on that.

The "track" is a feature in my code that allows controlling the aircon much more tightly and even using an external (BLE) reference.

The HA side only knows of the aircon itself, so will see the target go up and down a lot when using the track feature, and won't see the BLE temp. This is perhaps what would be better switched over to the settings used by the track feature instead. A useful change I may do.

But sounds like it needs more documentation at least.

andreas-bulling, 2023-09-20:
There is lots of potential in a tighter integration with HA (and potentially others) I think - the tracking in combination with an external BLE sensor is a really cool feature that few others (maybe even none?!) have. Ideally one could control/enable/disable most/all of this through HA.

Unfortunately I cannot help directly with coding - but I am  happy to continue giving feedback/ideas based on the six Faikin modules I have bought/will have in use eventually, and testing HA stuff out.

revk, 2023-09-20:
Still feel like shit, but yes, sounds like a good idea.

revk, 2023-09-23:
OK I am back on dry land, still not well, but I agree this would be a good change, so looking in to it. It would help if you are around to test some changes at some point this week.

revk, 2023-09-23:
I have issued new code.

The HA reporting of temp and target will not be the Faikin ones if the Faikin-auto mode is engaged. It will report as *auto* mode.

The control from HA to set temp should set the target we are using in Faikin-auto.

Do test and let me know any issues.

andreas-bulling, 2023-09-23:
Thanks a lot but just a quick question:

"The HA reporting of temp and target will not be the Faikin ones if the Faikin-auto mode is engaged. It will report as auto mode."

Did you mean to say "will noW be the Faikin ones.." ?

andreas-bulling, 2023-09-23:
Flashed the newest firmware but I unfortunately can't test the new functionality out - the external sensor is not read out anymore since I had to switch to the BTHome advertisement format (see the other issue I just created).

andreas-bulling, 2023-09-24:
OK, so in Faikin-auto mode the temperature from the external BLE sensor gets transmitted to HA fine. The operation mode of the AC cannot be changed in HA anymore - it seems to be fixed to "auto" (intended?)

However, in this mode, one also cannot turn off the indoor units with the remote control anymore - only through HA. I assume this is not intended?!

andreas-bulling, 2023-09-24:
So, as an intensive HA user - here's what would be ideal IMHO. The wishlist that is ;)
Maybe it helps/gives some ideas for further improvements if you have time.

The switch to enable/disable Farkin-auto mode is exposed to HA and can be operated there without having to go to the Faikin web UI. Same goes for the target temperature, allowed temperature deviation, eco mode etc.

The Faikin-auto mode would just be another operating mode. Turning it on disables any other mode and it equally gets disabled once the user selects heat/cool explicitly. Also by the "normal" auto mode.

ALTERNATIVE: The Faikin auto mode **replaces** the "normal" auto mode. It's better after all it seems. If a BLE sensor is connected that one is being used. Otherwise the internal sensor.

andreas-bulling, 2023-09-24:
> As for remote - not even tried that before, it may be something I can change to work out you must have done it view remote.

Sorry, don't get what you mean with "you must have done it view remote"?

revk, 2023-09-24:
Thanks for feedback. I'll have to see what I can work out.

revk, 2023-09-25:
I'll re-open so I don't forget.

JeanOlivier, 2023-10-21:
I just installed a BLE sensor for both of my units and it seems to have worked alright.

The faikin auto mode is very interesting, but it seems to take over the whole unit. Ideally I'd like to be able to say "cool only" or "heat only", and turn it on/off with  the timer.

However, what would be even better IMO, it to use the BLE sensor instead of the internal Daikin one, as transparently as possible.

What I mean by that is a mode where you automatically set the internat Daikin target to the desired value +ΔT between the BLE sensor and the Daikin one, and keep track of this +ΔT over time and adjust the Daikin internal target accordingly.

I thought about implementing this in HA automations but I'm not familiar enough with it yet, and there's the issue of keeping track of the actual target set by the user vs the internal target within the Daikin.

I think simply intercepting the "set temperature" command (from remote/HA/webpage) and then passing the adjusted value (+ΔT from the BLE) periodically would work. But you'd need to keep the "last user-set target" in memory.

revk, 2023-10-21:
I think configuring the over and under temp setting controls to zero may do that. Someone else posted that worked well for their Daikin.

JeanOlivier, 2023-10-21:
You mean essentially setting the PID (or PID-like) adjustment for temperature to follow the BLE directly?

It think this might essentially be what I'm asking for, but I'd like it to be a feature enabled no matter the mode. Let's says I'm turning on heat I don't want it to ever cool, and turning the unit on/off is only done manually or through the webpage Schedule.

Can one create virtual devices in HA with internal states that can control other devices? I could maybe do a prototype this way.

JeanOlivier, 2023-10-21:
Update:

I found this: https://www.home-assistant.io/integrations/template#state-based-template-sensors

Which seems like it'd be sufficient for my goal as long as I can read the BLE sensors through HA. Could the faikin forward that information? Otherwise I'll just buy a Bluetooth adapter for the HA server.

If I can find the time to implement this, and if it works well, I'll share it wherever appropriate.

revk, 2023-10-21:
Not sure on HA side. Obviously new ways of working could be added if people really want them.

revk, 2023-10-21:
The BLE is reported when in the auto mode. I don't recall for other reports. That is probably easy to add. But afaik HA can read these BLE as well.

JeanOlivier, 2023-10-21:
> The BLE is reported when in the auto mode. I don't recall for other reports. That is probably easy to add. But afaik HA can read these BLE as well.

HA can read it but I don't have Bluetooth support on my server. That's on me though, not you, a passthrough would just be convenient.

I'll try to implement my idea in HA. It's easier for me to script "over" your firmware than to modify it. If that's a feature you find interesting, or if there a demand from someone else than me about it, then feel free to implement it.

andreas-bulling, 2023-10-22:
No need for a BT module. If you got an ESP32 flashed with ESPHome close-by you can activate a [Bluetooth proxy](https://esphome.io/components/bluetooth_proxy.html) on the device to read out all values of the sensor in HA.

When you flash your temperature sensor I recommend you select the BTHome protocol - the others didn't not work reliably for me. BTHome devices get detected within a few seconds and report all sensor readings nicely (temperature, humidity, battery voltage and percentage)

JeanOlivier, 2023-10-22:
@andreas-bulling 

I will want to control when the unit is on/off a different times during the day and ensure it's only heating during winter and warming up during summer. I'm from a very cold-winter place, starting the AC when it's -30C outside doesn't make sense. Ideally I'd also like to turn off the heat pump in favor of regular electric heating when it's under -20C outside, but that's another discussion.

I agree that the best is when technology is doing its job in the background, I just want to tune that job my liking.

Ideally the BLE sensor would just replace the internal thermometer of the unit and I'd be happy. But I understand this requires some more logic.

revk, 2023-10-22:
I have a tired on/off and have a setting to turn on/off automatically based on temperature.

I'll have to double check the settings for change from hot/cold when needed. It really should only change if needed, but there are a few settings which can be tweaked which probably stop it ever hitting that.

The advanced section of the manual lists most, but I'm not at my computer right now to check.

JeanOlivier, 2023-10-22:
I've been using the scheduled On/Off fr a while now, but the "faikin auto" mode seems to take precedence over the schedule.

revk, 2023-10-22:
I'll look when get back, but there will be settings (via MQTT) that can be used to change the way it works.

JeanOlivier, 2023-10-24:
I received my Bluetooth adapter and it seems to be working fine with the flashed BLE temperature sensors over Home Assistant.

I started digging into python scripts for HA automation, see this more flexible schedule implementation https://github.com/revk/ESP32-Faikin/discussions/132 as an example.

I'll give a shot to an auto-track-external-temperature scripts when I have the time.

closed 2023-10-28 by revk

XMaarten, 2023-10-31:
If possible I would like the Faikin module to forward the raw data from the BLE sensor to MQTT / HomeAssistant, both the temperature and the humidity if available. It saves from having an extra ESPHome module just for that functionality...

revk, 2023-11-01:
I'll see what I can do.

revk, 2023-11-01:
I have issued new code - it should show BLE temp/hum on web page and on HA. Please do let me know if that works for you as I don't have a HA set up for testing.

XMaarten, 2023-11-01:
That is great! Thank you so much! I just tested it and it works without issue in HomeAssistant. 

One small thing: you are using mdi:humidity as icon which doesn't exist, I think this should be mdi:water-percent.

revk, 2023-11-01:
Ha, that was a guess, changed in next issue (a few minutes). Thanks for testing.

andreas-bulling, 2023-11-01:
I can also confirm that it works fine in HA. Minor thing: On the Faikin webpage the BLE temperature is shown twice now - with a slash behind the internal temperature and as "BLE"

revk, 2023-11-01:
Ah, if it is used as the reference temperature, that was always shown. So yes, may be a tad confusing. I may have to clean that up a bit.

JeanOlivier, 2023-11-01:
I can also confirm it works fine (minus the icon as stated above).

It's even better than using the thermometer via HA bluetooth in my case, since my server is far away from the thermometer/Daikin and the bluetooth signal is very faint, but the Faikin sees both the wifi and thermometer perfectly.

andreas-bulling, 2023-11-01:
there is another feature waiting to be implemented then - ESP Bluetooth Proxy (see ESPHome). If Faikin supported that it could also robustly forward the BLE sensor data. And not as a separate Faiking entity but readily using the ESPHome/HA Bluetooth infrastructure.

revk, 2023-11-01:
I have an app for BLE to MQTT. It does not (yet) create HA configs, but probably could.

revk, 2023-11-01:
I may look in the morning

andreas-bulling, 2023-11-01:
> @andreas-bulling You could already use [Faikin as ESPHome module](https://github.com/joshbenner/esphome-daikin-s21) and add the Bluetooth proxy there easily.

Thanks a lot - I actually know this repository but I haven't dared trying it out given that I wasn't sure which features it supports / if it's comparable to Faikin. Faikin's BLE feature is particularly nice

revk, 2023-11-02:
Covered by issue #133

commit 7bb78134 (2023-09-23): Fix #111