revk/xESP32-Faikout #113: Support BTHome advertisement format for temperature sensors

andreas-bulling, 2023-09-23:
The BLE temperature sensors work fine with the default custom advertisement format, i.e. the external temperature is shown in the Faiking user interface. Unfortunately, the HA Bluetooth implementation is still rather buggy and the Xiaomi BLE integration doesn't work for a lot of people out there. Typical symptom: HA only shows an empty signal strength entity for the device but nothing else.

One common solution is to switch the BLE sensors to the BTHome advertisement format. This works like a charm - sensors are detected within a few seconds typically and work reliably, even with the ESPHome/HA Bluetooth Proxy.

Unfortunately, this advertisement format is not supported by Faikin - the external temperature is not shown in the user interface.

revk, 2023-09-23:
I could say "don't do that then".

But I'll look.

revk, 2023-09-24:
Well, it looks well enough documented here https://bthome.io/format/

But setting up encrypted data would be an extra complication.

andreas-bulling, 2023-09-24:
> I could say "don't do that then".
> 
> But I'll look.

well - the sensor should really work with both Faikin and HA at the same time, shouldn't it? Currently people can only go for one of them, not ideal I'd say. Thanks for taking a look!

andreas-bulling, 2023-09-24:
> Well, it looks well enough documented here https://bthome.io/format/
> 
> But setting up encrypted data would be an extra complication.
> 
> I assume you are setting BT Home v1... That seems to sent a service ID `181C` (user data) and not as documented!
> 
> Ah, found it https://bthome.io/v1/

I am running BTHome unencrypted, that's the default in the custom BLE sensor's firmware anyway.

revk, 2023-09-24:
OK, I am issuing an update now, see if it does what you need.

closed 2023-09-24 by revk

andreas-bulling, 2023-09-24:
Seems to work like a charm, thanks a lot! With the BLE sensor working, later today (once it gets warmer) I will test the new auto adjust (tracking) functionality.

ando2040, 2024-02-19:
Thanks for making Faikin aware of BTHome (before I needed it). Not sure if @revk  will see this on a closed issue, but please could the BLE documentation be updated to recommend changing to BTHome when flashing the Telink device? It would have saved me a lot of re-pairing to PC when I discovered 5 sensors needed one setting changed to be imported to Home Assistant concurrently with Faikin! Cheers.

Reason for doing this is so I can move the sensors around the house as needed and try out different positions - Faikin makes it easy to swap every active sensor it can detect, but I might not be using all of them with Faikin all the time.

revk, 2024-02-19:
It may be worth starting a wiki page on the BLE sensors and options perhaps.

ando2040, 2024-02-19:
Sounds great but the existing sentence would be easy to update. "The default custom format advertisement is understood. If you use Home Assistant, BTHome advertisement is also understood by Faikin and will be recognised automatically in HA so it may be worth sending BTHome config while you have the Telink paired."

revk, 2024-02-19:
Maybe when I get home.

revk, 2024-02-20:
Done