revk/xESP32-Faikout #115: lack of choice

ckvsoft, 2023-09-25:
thanks for the great solution.
However, my Prefera 2023 has other settings that I miss. Maybe you can include these too.
These would be
Vertical airflow direction: on, off, comfort
and streamer: on, off

I noticed that I can switch powerful on but the display immediately goes off again even though powerful is on.

The outside temperature is not displayed anywhere.

Thanks LG Chris

revk, 2023-09-25:
Basically I need the S21 messages and replies for your unit to add these.

There is a debug mode over MQTT which can be used to find clues.

ckvsoft, 2023-09-25:
Thanks for the fast reply.
How do I turn on Debug mode?

revk, 2023-09-25:
I really have to make sure this is in the manual.

Send MQTT message setting/Faikin/whatevename/debug 1

That should start logging S21 status messages.

Change settings, eg via remote, and hopefully there is a change in one of the replies.

ckvsoft, 2023-09-25:
Yes, I already found this in the description, but it doesn't seem to work.
I tried mosquito_pub but I don't see any change. Where can I see the debug information?

revk, 2023-09-25:
The debug is logged to mqtt. Even without setting debug you should see some information being posted to MQTT by the Faikin anyway. Assuming you do have it connected to your MQTT server.

ckvsoft, 2023-09-25:
I use IoBroker and can see some of Faikin's MQTT stuff. But no debug information

I'm probably too stupid for that :)

revk, 2023-09-25:
OK you should see your published setting...

Ok it may not have Faikin in it. May be `setting/whatevername/debug 1`

So try that.

(There are options to include `/Faikin` or not, but I think the default does not have it)

ckvsoft, 2023-09-25:
Do you need the
payload":{"type":"Buffer","data" ... ?
Or  the Status? Here the only change is  swingv

revk, 2023-09-25:
If you have `swingv` then that should show on the web page as an option.

I may have to check, do you not have `swingh` ? It may be that I expected to have both.

ckvsoft, 2023-09-25:
I think we're talking at cross purposes.

Of course I have swingh and swingv

Both have only true and false

In my case there are several options for swingv

With swingv it should be possible: true and false and as a further option comfort

revk, 2023-09-25:
Ah, OK, I see...

OK the debug log should include a `"G5":` value. Let me know that in all the combinations you can set with the remote, and I should be able to work it out.

ckvsoft, 2023-09-25:
I guess I have a problem now because I can't recognize the characters on G5.
Example:  "G5\":\"2?0�\"

revk, 2023-09-25:
Ok that is fun. Let me look in to improving that debug. I'll get new code shortly for that.

ckvsoft, 2023-09-25:
What is your development environment?
I usually program on Linux (kubuntu, Manjaro) in C++ with QTCreator
for ESP32 I use VSCode with PlatformIO

revk, 2023-09-25:
This is all ESP IDF (Linux or Mac) and C.

revk, 2023-09-25:
There is no published spec...

Streamer (whatever that is) looks doable. But that vertical looks like it may not report "comfort".

Sadly some things are not in S21, eg the quiet/night fan mode is not reported either. So might not be possible.

If I get the duff unicode to report in a readable way that would help. I'll do that anyway.

What is streamer anyway?

ckvsoft, 2023-09-25:
Daikin's Streamer Air Purifier is proven to remove up to 99.9% influenza viruses, bacteria, allergens and pollen, creating the ideal living space for you and your family.

revk, 2023-09-25:
Nice

ckvsoft, 2023-09-25:
> Update: hang in looks like comfort may be in G6 anyway.
> 
> It may be simpler to make streamer and comfort extra on/off, even it it goes off when vertical goes off.

Yes, that seems so. It's a good idea to make your own switch for it.

revk, 2023-09-25:
I have comfort, and made a guess at streamer, a new G6 dump from you will help confirm. Can you try - it may work anyway.

ckvsoft, 2023-09-25:
So I did a few tests.
Comfort mode seems to work.

When I turn on Streamer, quiet mode turns on automatically. "G6" then automatically goes from "0000" to "0°00"

If I only activate the Quiet Mod, "G6" goes from "0000" to "°000"

What exactly is Sensor Mode?

ckvsoft, 2023-09-25:
When I switch on the streamer from the remote control, G6 also goes to "0°00"
The Quiet Mode will then also be activated in the web app

ckvsoft, 2023-09-25:
Quiet mode from the remote control switches G6 to " "°°04", streamer off
Quiet and streamer on is G6 on "°004"

Only Streamer G6 is "0°00"

revk, 2023-09-25:
Sensor mode is one I have, but OK, I may need to tweak streamer then.
° is a surprise, you sure?

ckvsoft, 2023-09-25:
Ok, I realized that I also have the sensor mode ;-) At first I thought it only affected the comfort mode.

Yes, ° is a surprise

revk, 2023-09-25:
You have to be viewing this via something that understand JSON and sees the literal string`\u00B0` as ° which is interesting.

ckvsoft, 2023-09-25:
Yes,  \u00B0  should be the degree sign

revk, 2023-09-25:
Yes, but the actual MQTT text is `\u00B0` not `°` so something is interpreting that for you. No issue, it is what expected.

So, in summary, what is working and not working for you now?

ckvsoft, 2023-09-25:
Ok, I can also send this to you as raw if necessary.
Thought it didn't matter

revk, 2023-09-25:
Does not matter, as long as I can work it out. So what is and is not working now?

ckvsoft, 2023-09-25:
Swing
Vertical off, on, comfort working
streamer and quite mode i can not test. 
If I use the remote control to set stream, faikin recognizes it and automatically switches to quite mode. I can't tell what the right value is now.

revk, 2023-09-25:
Ok well, let me know if I need to change something, but sounds like we have managed to get a few more controls on there now, which is nice.

ckvsoft, 2023-09-25:
Unfortunately not quite yet. You can't switch streamers individually.
Can I have a version without stream and quite to recognize the values ​​from the remote control without the faikin changing anything? I then write back exactly the values ​​we need

revk, 2023-09-25:
You should be able to do that with debug and a remote, surely? It won't change back if you change stuff.

ckvsoft, 2023-09-25:
No, it will not be put back. Only when I stream on site does Daikin recognize this and automatically activate Quite Mode. and vice versa. Both are always active in the display

ckvsoft, 2023-09-25:
case '6':                // 'G6' - "powerful" mode and some others
         set_val (powerful, payload[0] & 0x02 ? 1 : 0);
         set_val (comfort, payload[0] & 0x40 ? 1 : 0);
         set_val (quiet, payload[1] & 0x80 ? 1 : 0);
         set_val (streamer, payload[1] & 0x80 ? 1 : 0);
         set_val (sensor, payload[3] & 0x08 ? 1 : 0);

Isn't the same value set for quiet and streamer here?

revk, 2023-09-25:
Ooooh, yes...

revk, 2023-09-25:
OK I'll change that now.

ckvsoft, 2023-09-25:
Ok works now.
But Sensor here do nothing on G6

ckvsoft, 2023-09-25:
I would like to say thank you very much for the quick implementation and the time you invested.

revk, 2023-09-26:
I'll recheck sensor here, but any that you don't have are likely to simply not work.

ckvsoft, 2023-09-26:
Yes, I already have the sensor and can switch it on with the remote control. But it doesn't seem to arrive on the S21.

revk, 2023-09-26:
Ah, ok, may be one of those things. I am pretty sure it works here, but I'll double check. If not then I'll remove it.

ckvsoft, 2023-09-26:
Is it possible to also display the outside temperature? As far as I can see, the information is there.

ckvsoft, 2023-09-26:
how does it work with the BLE temperature sensors? How is this recognized? Is there any information about what the mqtt message looks like? I would like to integrate my existing Raspberry running device

revk, 2023-09-26:
> Is it possible to also display the outside temperature? As far as I can see, the information is there.

Should be easy enough.

revk, 2023-09-26:
> how does it work with the BLE temperature sensors? How is this recognized? Is there any information about what the mqtt message looks like? I would like to integrate my existing Raspberry running device

The BLE is is not MQTT, it is BLE. Sorry, what are you asking? I do have code (BLE-Env) that picks them up and reports to mqtt.

ckvsoft, 2023-09-26:
I mean how does Faikin ask or how does he find such a sensor? I would like to send information from my sensors to the server using mqtt. If I do the same thing as I did with the specified sensors, Faikin should recognize them too. Where does Faikin look for what mqtt information?

revk, 2023-09-26:
> I mean how does Faikin ask or how does he find such a sensor? I would like to send information from my sensors to the server using mqtt. If I do the same thing as I did with the specified sensors, Faikin should recognize them too. Where does Faikin look for what mqtt information?

It uses BLE...

Sorry, What am I missing. There is also an option to send temp via mqtt (in the manual, honest), but the BLE sensors work over BLE, not MQTT.

ckvsoft, 2023-09-26:
> It uses BLE...
>
> Sorry, What am I missing. There is also an option to send temp via mqtt (in the manual, honest), but the BLE sensors work over BLE, not MQTT.

Yes I know that. But I would be interested to see how this is mapped in the mqtt with your software and how Faikin recognizes it. Unfortunately I don't have such a sensor to see this.

ruimmsimoes, 2023-09-26:
Would it be possible for the faikin board to communicate the temperature and humidity to home assistant regardless of the AC unit state and operating mode?

I want to use this info to make other automations in home assistant (covers, WC ventilation).
If the faikin board bridged this info to home assistant would be perfect.

revk, 2023-09-26:
> Would it be possible for the faikin board to communicate the temperature and humidity to home assistant regardless of the AC unit state and operating mode?
>
> I want to use this info to make other [...]

Home Assistant can usually "see" BT Home v1 BLE directly, so not sure it is needed is it?

ckvsoft, 2023-09-26:
> Would it be possible for the faikin board to communicate the temperature and humidity to home assistant regardless of the AC unit state and operating mode?
>
> I want to use this info to make other [...]

I have, as can be seen in the BLE Env code, is a
info/BLE-Env Object created. But no matter whether I write direct temp here or temp in my own object (info/BLE-Env/Senosr1), the selection on the Faikin website always says "None"

revk, 2023-09-26:
The Faikin uses the same BLE-Env code, so should see any sensors. Make sure you are up to date, obviously.

ruimmsimoes, 2023-09-26:
> Home Assistant can usually "see" BT Home v1 BLE directly, so not sure it is needed is it?

My Home Assistant is running on a NAS without Bluetooth and I have a big home, even if I buy a receiver wouldn't be able to connect with all devices.
Since I will put a faikin in most divisions and it already connects to the temp&humidity sensor it would be perfect if it forwarded that info to Home Assistant.

ckvsoft, 2023-09-26:
> The Faikin uses the same BLE-Env code, so should see any sensors. Make sure you are up to date, obviously.

> The Faikin uses the same BLE-Env code, so should see any sensors. Make sure you are up to date, obviously.

Kannst Du mir vielleicht ein Beispiel einer passenden MQTT Message geben? Irgendwie erkennt Faikin keinen Sensor

ckvsoft, 2023-09-28:
Is there no MQTT payload example anywhere of what a BLE Env entry looks like or do I have to find out myself?

revk, 2023-09-28:
> Is there no MQTT payload example anywhere of what a BLE Env entry looks like or do I have to find out myself?

What do you think is making the "BLE Env" "entry" exactly. The library provides data that the Faikin code uses. It logs as "temp" if being used. There is not BLE Env MQTT and not MQTT direct from the BLE device (it is BLE not MQTT).

ckvsoft, 2023-09-28:
Didn't know it was such a secret.
An info/BLE-Env/temp or info/BLE-Env/ATC-Sensor/temp is simply not recognized, so I think something different is expected here.
I read my Xiaomi with a raspberry and transfer the temp to my MQTT server. I don't want to add another ESP32 in between. But I probably won't have any other choice to find out.
Thanks

closed 2023-09-28 by ckvsoft

revk, 2023-09-28:
Not a secret, I just don't know what you are asking.

I do have code for BLE-Env, and that does log to MQTT, but that is not the Faikin app.

The Faikin app does not log the BLE temp devices it sees. It uses my BLE-Env library to find one sensor you select and use as a reference it logs as "temp" when using it as a reference.

If you are asking about my BLE-Env app, I am happy to answer. Are you? Raise an issue or discussion in that https://github.com/revk/ESP32-BLE-Env