revk/xESP32-Faikout #118: Faikin module down after some days

HV-NL, 2023-09-28:
Recently I installed a Faikin module to replace the BRP069B41 module of my FTXM42NV1B, and updated to the latest version (auto-update on: it is now running 40bca4b 2023-09-25T19:34:17).

I use an external sensor (I bought both BlueCoinT and Telink BLE sensors). But the system stops working after some days, I cannot reach the web page any more. I still can ping the module, but I cannot control it any more.

After the first 'hang' I switched from the BlueCoinT sensor to the Telink sensor, but that did not help.
5 days ago I disabled MQTT in order to exclude this as a possible reason; but that did not help either.
(There is no logging available)

As the module does not have a reset button  I have to power-off my whole Daikin system to restart the Faikin module.

What can be the cause? I run Domoticz, and have a script to request the Daikin status every 10 seconds. Is that too frequent for the Faikin module?

revk, 2023-09-28:
can you confirm, when accessing the web page, is it via the `name.local` URL. I have seen the MDNS have issues, so if you can access by IP that is better.

Or is it losing MQTT as well?

Needless to say, I have used here for literally years now.

HV-NL, 2023-09-28:
I access by IP only (configured a fixed IP address in my DHCP server).

In the meantime I restarted the system, so I cannot answer the question about losing MQTT updates.

revk, 2023-09-28:
Somewhat hard to diagnose then.

See you on MQTT

See if you can ping

HV-NL, 2023-09-28:
This afternoon I could still ping when I discovered the http session was down.
A week ago I could not ping the module at all.

I think I discovered the problem much sooner now than last week, as a week ago I configured Domoticz to send me a triggger when it cannot reach the Faikin module for 10 minutes. 
So my guess is that first the HTTP server stops working, and some time later the IP stack stops working too.

revk, 2023-09-29:
OK, tricky, I guess it could be a memory leak. The mqtt should log periodically with a memory space available, that will be a clue. If so, I'll have to look in to where it could be leaking.

HV-NL, 2023-09-29:
I created a data logger, and am currently logging the '/Faikin/#' items.
But I don't see any info about the memory space in this item. 
```
/Faikin
{"time": [2023, 9, 29, 12, 35, 0, 4, 272, 1], "topic": "Faikin/Faikin-eetkamer", "message": {"protocol": "S21", "ts": "2023-09-29T10:35:00Z", "env": [20.73, 20.75, 20.78], "online": true, "home": [20.5, 20.99, 21.0], "heat": false, "fanrpm": 0, "comp": 0, "outside": 18.0, "liquid": 22.0, "power": false, "mode": "F", "temp": 18.0, "fan": "2", "swingh": false, "swingv": false, "econo": false, "powerful": false, "comfort": false, "streamer": false, "sensor": false, "quiet": false}}
``` 

The other interesting item seems to be 308398D3D364, but that item also does not show stack info
```
/308398D3D364
{"online":true,"temp":20.00,"outside":13.00,"liquid":21.00,"comp":0,"fanrpm":0,"mode":"off","fan":"2","swing":"off","preset":"home"}
```

What subject should I log?

revk, 2023-09-29:
Looks for `+/Faikin-eetkamer/#` and you may see an `info/Faikin-eetkamer...` periodically

HV-NL, 2023-10-03:
I am monitoring since 4 days, no hang-ups occured these 4 days. The available memory seems to be around 29636 bytes. Most of the time this seems to be a rather stable number.

The lowest number I saw however was 8680, during the automatic download of a new version.
```
{"time": [2023, 9, 30, 12, 1, 52, 5, 273, 1], "topic": "state/Faikin-eetkamer", "message": {"ts": "2023-09-30T10:01:51Z", "id": "308398D3D364", "up": 5506, "mqtt-up": 5502, "restart": 31, "reason": "OTA Download", "mem": 8680, "spi": 2091116}}
```
So perhaps a previous auto-upgrade required too much memory, resulting in the system to 'hang' itself? Just a guesss....

I will keep monitoring, fingers crossed

revk, 2023-10-03:
The upgrade does use a bit but that is a lot more than I would expect.

But if stable now, the  good.

HV-NL, 2023-10-16:
An hour ago I monitored an auto-upate. During this update the memory dropped to 4364 bytes.

```
{"time": [2023, 10, 16, 10, 15, 7, 0, 289, 1], "topic": "state/Faikin-eetkamer", "message": {"ts": "2023-10-16T08:15:06Z", "id": "308398D3D364", "up": 6301, "mqtt-up": 6297, "restart": 31, "reason": "OTA Download", "mem": 8388, "spi": 2091292}}
{"time": [2023, 10, 16, 10, 15, 9, 0, 289, 1], "topic": "state/Faikin-eetkamer", "message": {"ts": "2023-10-16T08:15:07Z", "id": "308398D3D364", "up": 6302, "mqtt-up": 6298, "restart": 31, "reason": "OTA Download"}}
{"time": [2023, 10, 16, 10, 15, 9, 0, 289, 1], "topic": "state/Faikin-eetkamer", "message": {"ts": "2023-10-16T08:15:09Z", "id": "308398D3D364", "up": 6303, "mqtt-up": 6300, "restart": 31, "reason": "OTA Download", "mem": 4364, "spi": 2091292}}
{"time": [2023, 10, 16, 10, 15, 9, 0, 289, 1], "topic": "state/Faikin-eetkamer", "message": {"ts": "2023-10-16T08:15:09Z", "id": "308398D3D364", "up": 6304, "mqtt-up": 6300, "restart": 31, "reason": "OTA Download"}}
```
After the reset the free memory is 49788 (used to be around 29636)
```
{"time": [2023, 10, 16, 10, 16, 31, 0, 289, 1], "topic": "state/Faikin-eetkamer", "message": {"ts": "2023-10-16T08:16:31Z", "id": "308398D3D364", "up": 62, "mqtt-up": 58, "mem": 49788, "spi": 2070320}}
{"time": [2023, 10, 16, 11, 19, 56, 0, 289, 1], "topic": "state/Faikin-eetkamer", "message": {"ts": "2023-10-16T09:19:56Z", "id": "308398D3D364", "up": 3867, "mqtt-up": 3863, "mem": 49556, "spi": 2070320}}

```

revk, 2023-10-16:
Does it then gradually go down over time?

HV-NL, 2023-10-16:
No, it constantly was around the same value during the last week (around 29636). 
During the auto-upgrade it just fell down to 4364, and (luckily!) recovered to a much higher value

Since the reboot I now have 2 measurements available (49788 and 49556):
```
{"time": [2023, 10, 16, 10, 16, 31, 0, 289, 1], "topic": "state/Faikin-eetkamer", "message": {"ts": "2023-10-16T08:16:31Z", "id": "308398D3D364", "up": 62, "mqtt-up": 58, "mem": 49788, "spi": 2070320}}
{"time": [2023, 10, 16, 11, 19, 56, 0, 289, 1], "topic": "state/Faikin-eetkamer", "message": {"ts": "2023-10-16T09:19:56Z", "id": "308398D3D364", "up": 3867, "mqtt-up": 3863, "mem": 49556, "spi": 2070320}}
```

revk, 2023-10-16:
The drop during the upgrade is normal (though why it uses so much I don't know, all inside the esp idf somewhere). But if it is not dropping otherwise it may be Ok. Very odd.

If you find some operation that makes it drop in normal use, let me know, it will be a clue to the leak, if any.

andreas-bulling, 2023-10-17:
I experienced something similar this morning. I had changed some settings as one of the indoor units would just blink the green LED and not actually turn on. Suddenly the page of both Faikin modules couldn't be reached anymore by IP address. Several minutes later (not sure if it rebooted or something) the modules were back. But settings were gone / set to something different as they were before.

HV-NL, 2023-10-21:
> OK, tricky, I guess it could be a memory leak. The mqtt should log periodically with a memory space available, that will be a clue. If so, I'll have to look in to where it could be leaking.

Looking at the mqtt messages I see that 
```"heat": true```
sometimes send a **value** i.s.o. a **boolean**:
```"heat": 0.92
"heat": 0.6
"heat": 0.07
"heat": 0.62
```

Can this be a reason for the memory problems? Are some fields in the code in overlap or so?
B.t.w: since yesterday I nearly get no more messages containing ```mem= ```
These messages used to occur once every hour, but the last message with ```mem= ``` occurred at 11:59 today and the previous one at 00:52.

revk, 2023-10-21:
There is a one minute status post. It reports various known values. It shows the value if it is the same for whole minute. If not it shows an array of min, ave, max for numeric values. For Boolean that has changed it shows a number 0 to 1 for how much time it was true. There is linux logging and graphing code to plot this. No, that is not likely to relate to a memory leak.

HV-NL, 2023-10-25:
Yesterday my module got the weekly software update (```"version": "47dd9fc", "build-suffix": "-S1-PICO", "build": "2023-10-15T10:04:50"```). After the update the reported up-time is very low; my guess is that the module restarts frequently, 'up' many times is now given as '4' or '5', indicating a restart of the module.

But since an hour I lost all connection to the module. It also has stopped sending mqtt messages. I cannot ping it any more. The last mqtt message was ```{"time": [2023, 10, 25, 10, 52, 46, 2, 298, 1], "topic": "state/Faikin-eetkamer", "message": {"up": false}}```

The previous mqtt message indicating 'up' was 
```{"time": [2023, 10, 25, 10, 34, 2, 2, 298, 1], "topic": "state/Faikin-eetkamer", "message": {"ts": "2023-10-25T08:34:02Z", "id": "308398D3D364", "up": 237, "mqtt-up": 233, "mem": 48452, "spi": 2070320}}```

There should be enough free memory, as 'mem' fluctuates around 49500

The module appears to be dead now. It has not activated itself as a wifi accesspoint, and I cannot ping it any more.

revk, 2023-10-25:
That is rather worrying!

Do you have a picture of it, so I can check which version it is.

HV-NL, 2023-10-25:
I see I quoted the wrong mqtt message in my first sentence, as this is what my screen displays (empty screen, with this line at the bottom):
[image]

revk, 2023-10-25:
I meant picture of the Faikin module.

HV-NL, 2023-10-25:
2023-07-19 5060634238151

revk, 2023-10-25:
OK, I may be able to work it out from that, I think that pre-dates the S3 processors. Nothing in the update should kill it, obviously. Where in the world are you, may be simplest to send you a replacement.

HV-NL, 2023-10-25:
I live in the Netherlands. 
September 3 2023 I bought 2 identical modules (1 for myself, 1 for my neighbor) at amazon.co.uk (Order # 204-1451269-3753969 )

Up to now I only installed it in my own unit, as I do not want my neighbor to have a module that goes down rather frequently (he is 80+, so I want to give him as least stress as possible)

revk, 2023-10-25:
My concerns are that (a) I don't think anyone else has reported them "going down", and (b) that it has now died totally.

Can you try the other module. Does that work better. If it is literally one faulty module, a replacement is simplest answer, and I could post one.

HV-NL, 2023-10-25:
It did not die completely, as I did not yet switch off my Daikin outdoor unit yet. I was more or less waiting for a watchdog to re-enable the module (which did not happen). 

A minute ago I switched off the outdoor unit and enabled it again. Now I see the Faikin webpage again.

This is a picture of my module:
[image]

revk, 2023-10-25:
OK, so still suffering from something causing it to die...  So could be a software issue.

If you have a second one, it is worth swapping to see if that does the same, that would be a clue.

Yes, it has a watchdog, but if the issue is a memory leak or similar simply meaning the IP stack locks up, that might not kick the watchdog, sadly.

HV-NL, 2023-10-31:
I installed the other module, and it kept working for some days. But this morning the module was down again (no webpage, an I could not ping it). Like the previous time, the last message was that the module was going down: 
```"message": {"up": false}}```

These are the last 3 messages before the unit went down:
```{"time": [2023, 10, 31, 7, 28, 0, 1, 304, 0], "topic": "Faikin/Faikin-eetkamer", "message": {"protocol": "S21", "ts": "2023-10-31T06:28:00Z", "env": [16.09, 16.11, 16.13], "online": true, "home": 16.5, "heat": false, "fanrpm": 0, "comp": 0, "outside": 10.5, "liquid": 18.0, "power": false, "mode": "A", "temp": 19.0, "fan": "5", "swingh": false, "swingv": false, "econo": false, "powerful": false, "comfort": false, "streamer": false, "sensor": false, "quiet": false}}
{"time": [2023, 10, 31, 7, 29, 0, 1, 304, 0], "topic": "Faikin/Faikin-eetkamer", "message": {"protocol": "S21", "ts": "2023-10-31T06:29:00Z", "env": [16.1, 16.13, 16.14], "online": true, "home": 16.5, "heat": false, "fanrpm": 0, "comp": 0, "outside": 10.5, "liquid": 18.0, "power": false, "mode": "A", "temp": 19.0, "fan": "5", "swingh": false, "swingv": false, "econo": false, "powerful": false, "comfort": false, "streamer": false, "sensor": false, "quiet": false}}
{"time": [2023, 10, 31, 7, 30, 0, 1, 304, 0], "topic": "Faikin/Faikin-eetkamer", "message": {"protocol": "S21", "ts": "2023-10-31T06:30:00Z", "env": [16.1, 16.12, 16.14], "online": true, "home": 16.5, "heat": false, "fanrpm": 0, "comp": 0, "outside": 10.5, "liquid": 18.0, "power": false, "mode": "A", "temp": 19.0, "fan": "5", "swingh": false, "swingv": false, "econo": false, "powerful": false, "comfort": false, "streamer": false, "sensor": false, "quiet": false}}
{"time": [2023, 10, 31, 7, 30, 45, 1, 304, 0], "topic": "state/Faikin-eetkamer", "message": {"up": false}}
```

I powered-off the outdoor unit, and re-enabled it, so I could reach the webinterface again. But then something very strange happened. I instructed the module to start heating, which can be seen in the mqtt messages. But i.s.o. heating, the unit started *cooling*. So I stopped using the web interface, switched off the power of the outdoor unit, and restarted. Now it is heating again.

This is the sequence of events:

At first I powered-on the outdoor unit:
```{"time": [2023, 10, 31, 10, 30, 44, 1, 304, 0], "topic": "error/Faikin-eetkamer/comms", "message": {"protocol": "S21", "cmd": "F1", "noack": true}}
{"time": [2023, 10, 31, 10, 30, 44, 1, 304, 0], "topic": "state/Faikin-eetkamer/status", "message": {"protocol": "S21", "online": false, "autob": "A4C138DE21CE", "autor": 0.0, "autot": 0.0, "auto0": "00:00", "auto1": "00:00", "autop": false}}
{"time": [2023, 10, 31, 10, 30, 45, 1, 304, 0], "topic": "state/Faikin-eetkamer", "message": {"id": "308398CEE264", "up": 5, "mqtt-up": 1, "app": "Faikin", "version": "90af91b", "build-suffix": "-S1-PICO", "build": "2023-10-30T08:12:57", "flash": 8388608, "rst": 1, "mem": 51812, "spi": 2070408, "ssid": "LinksysAtHome", "bssid": "744D28C3F2FF", "rssi": -57, "chan": 3, "ipv4": "192.168.11.71"}}
{"time": [2023, 10, 31, 10, 30, 46, 1, 304, 0], "topic": "error/Faikin-eetkamer/comms", "message": {"protocol": "X50\u00acTx", "timeout": true}}
{"time": [2023, 10, 31, 10, 30, 49, 1, 304, 0], "topic": "error/Faikin-eetkamer/comms", "message": {"protocol": "S21\u00acTx", "cmd": "F1", "noack": true}}
{"time": [2023, 10, 31, 10, 30, 50, 1, 304, 0], "topic": "error/Faikin-eetkamer/comms", "message": {"protocol": "X50\u00acRx", "timeout": true}}
{"time": [2023, 10, 31, 10, 30, 53, 1, 304, 0], "topic": "error/Faikin-eetkamer/comms", "message": {"protocol": "S21\u00acRx", "cmd": "F1", "noack": true, "value": "7E"}}
{"time": [2023, 10, 31, 10, 30, 55, 1, 304, 0], "topic": "error/Faikin-eetkamer/comms", "message": {"protocol": "X50\u00acRx\u00acTx", "timeout": true}}
{"time": [2023, 10, 31, 10, 30, 58, 1, 304, 0], "topic": "error/Faikin-eetkamer/comms", "message": {"protocol": "S21\u00acRx\u00acTx", "cmd": "F1", "noack": true}}
{"time": [2023, 10, 31, 10, 30, 59, 1, 304, 0], "topic": "error/Faikin-eetkamer/comms", "message": {"protocol": "X50", "timeout": true}}
{"time": [2023, 10, 31, 10, 31, 3, 1, 304, 0], "topic": "state/Faikin-eetkamer/status", "message": {"protocol": "S21", "env": 16.2, "online": true, "home": 16.5, "heat": false, "fanrpm": 0, "comp": 0, "outside": 11.0, "liquid": 18.0, "power": false, "mode": "F", "fan": "5", "swingh": false, "swingv": false, "econo": false, "powerful": false, "comfort": false, "streamer": false, "sensor": false, "quiet": false, "ble": "A4C138DE21CE", "autob": "A4C138DE21CE", "autor": 0.0, "autot": 0.0, "auto0": "00:00", "auto1": "00:00", "autop": false}}
{"time": [2023, 10, 31, 10, 31, 10, 1, 304, 0], "topic": "state/Faikin-eetkamer", "message": {"id": "308398CEE264", "up": 30, "mqtt-up": 26, "mem": 49820, "spi": 2070320}}
{"time": [2023, 10, 31, 10, 31, 16, 1, 304, 0], "topic": "Faikin/Faikin-eetkamer", "message": {"protocol": "S21", "ts": "2023-10-31T09:31:16Z", "env": [16.23, 16.24, 16.26], "online": 0.64, "home": 16.5, "heat": false, "fanrpm": 0, "comp": 0, "outside": 11.0, "liquid": 18.0, "power": false, "mode": "F", "fan": "5", "swingh": false, "swingv": false, "econo": false, "powerful": false, "comfort": false, "streamer": false, "sensor": false, "quiet": false}}
```
Then I started heating:
```
[... cut ...] se, "streamer": false, "sensor": false, "quiet": false}}
```
At this point I switched off using the web interface, powered-off the system, and restarted it again.

* I have no idea why the Faikin module switch itself off
* Very strange that after restarting the indoor unit start *cooling*, when the Faikin module requested *heating*
* Can it be that the Faikin module does not communicate with the indoor unit properly? 
  (see the error messages that are sent in the startup phase, so around 10:30)

revk, 2023-10-31:
I'll look in more detail a bit later...

I'm assuming it's more some quirk with whatever dhcp server you have. I can't even remember if I have a fixed IP mode, I'll have to check, but if so, it may be worth a try.

revk, 2023-10-31:
It is possible to enable `debug` and even `dump` settings which may allow things like the "why did it do cool not heating" to be better understood, but they do generate a lot of logs. So better if you have a reproducible issue.

As for liquid temp dropping, no idea, that is very much down to working of the unit. A change from heat to cool may well mean it has to do something to adjust what it is doing. The Faikin is just reporting what it sees.

As for the original problem - we don't see logging of memory issues, so maybe this is some IP level problem - perhaps some issue with DHCP or some such, no idea.

One approach may be to try fixing the IP address and gateway. There are settings for `wifiip`, `wifigw` and `wifidns` which you can set to IPv4 addresses to force fixed IP working. See if that stays working?

HV-NL, 2023-10-31:
I enabled debug. Now I see many lines containing ```{"protocol": "S21", "badlength": "3", "command": "SD", "data": "303030"}```

```
{"time": [2023, 10, 31, 13, 13, 17, 1, 304, 0], "topic": "info/Faikin-eetkamer/s21", "message": {"G1": "14@7", "G2": "=;\u0000\u0080", "G3": "000\u0000", "G4": "0\u0000\u00800", "G5": "000\u0080", "G6": "0004", "G7": "l2AS", "G8": "0200", "GA": "74F1", "GB": "0441", "GC": "6F01", "GG": "0000", "GK": "q{51", "GM": "5372", "GN": "1000", "GP": "5C00", "GQ": "5C00", "GS": "0000", "GT": "3000", "SH": "081+", "SI": "072+", "Sa": "521+", "SL": "270", "Sd": "510", "SN": "030+", "SX": "581+", "SD": "000"}}
{"time": [2023, 10, 31, 13, 13, 20, 1, 304, 0], "topic": "error/Faikin-eetkamer/comms", "message": {"protocol": "S21", "badlength": "3", "command": "SD", "data": "303030"}}
{"time": [2023, 10, 31, 13, 13, 20, 1, 304, 0], "topic": "info/Faikin-eetkamer/s21", "message": {"G1": "14@7", "G2": "=;\u0000\u0080", "G3": "000\u0000", "G4": "0\u0000\u00800", "G5": "000\u0080", "G6": "0004", "G7": "l2AR", "G8": "0200", "GA": "74F1", "GB": "0441", "GC": "6F01", "GG": "0000", "GK": "q{51", "GM": "5372", "GN": "1000", "GP": "5C00", "GQ": "5C00", "GS": "0000", "GT": "3000", "SH": "081+", "SI": "072+", "Sa": "521+", "SL": "370", "Sd": "510", "SN": "030+", "SX": "581+", "SD": "000"}}
{"time": [2023, 10, 31, 13, 13, 20, 1, 304, 0], "topic": "state/Faikin-eetkamer/status", "message": {"protocol": "S21", "env": 16.7, "online": true, "home": 18.0, "heat": true, "fanrpm": 730, "comp": 15, "outside": 12.5, "liquid": 27.0, "power": true, "mode": "H", "temp": 18.0, "fan": "5", "swingh": false, "swingv": false, "econo": true, "powerful": false, "comfort": false, "streamer": false, "sensor": false, "quiet": false, "ble": "A4C138DE21CE", "autob": "A4C138DE21CE", "autor": 0.0, "autot": 0.0, "auto0": "00:00", "auto1": "00:00", "autop": false}}
{"time": [2023, 10, 31, 13, 13, 23, 1, 304, 0], "topic": "error/Faikin-eetkamer/comms", "message": {"protocol": "S21", "badlength": "3", "command": "SD", "data": "303030"}}
{"time": [2023, 10, 31, 13, 13, 23, 1, 304, 0], "topic": "info/Faikin-eetkamer/s21", "message": {"G1": "14@7", "G2": "=;\u0000\u0080", "G3": "000\u0000", "G4": "0\u0000\u00800", "G5": "000\u0080", "G6": "0004", "G7": "l2AR", "G8": "0200", "GA": "74F1", "GB": "0441", "GC": "6F01", "GG": "0000", "GK": "q{51", "GM": "5372", "GN": "1000", "GP": "5C00", "GQ": "5C00", "GS": "0000", "GT": "3000", "SH": "081+", "SI": "072+", "Sa": "521+", "SL": "370", "Sd": "510", "SN": "030+", "SX": "581+", "SD": "000"}}
```

I will look into a fixed-IP setting later today, I just wanted to share some first debug impressions

revk, 2023-10-31:
Bad length is odd, suggests the low level protocol is not working well.

However, you are not the only person to see bad length and a 000 payload like that recently and I wonder if we have a separate bug - i.e. that may be a total red-herring. `SD` is only sent in debug, and it could simply be that we are expecting a different length for some reason.

I'll look in to that.

HV-NL, 2023-10-31:
> P.S. Issuing new code now that expects length 3 on SD.

Check. My module just got the new version, now the errors are gone

I will now look into the static IP settings

HV-NL, 2023-11-04:
This morning the other Faikin module was down. No ping possible. No Faikin as accesspoint either. Fixed IP configured since 4 days.
Unluckily my logging did stop yesterday, probably as side-effect of an 'apt update' of my Raspberry Pi, so no details available.

revk, 2023-11-04:
Ah, I would hope logging may give  clue, sorry.

wurmtal868, 2023-11-06:
On a side note:
According to the picture above the module is missing the 5V because its connected with the 4 wire Daikin cable. On later modules the 5V are powering the TX pullup of the level shifter. I don't know if this could cause issues on the S1 modules.

HV-NL, 2023-11-07:
_Just a status update._

After setting fixed IP (`wifiip`, `wifigw `and `wifidns`) all seemed OK. But some days ago I saw that the time (`"ts"`) was `1970-01-01`. OTA also was not possible any more. After resolving `ota.revk.uk `no request was done to `excalibur.belmont.cymru`, every OTA request ended with `error -3`

As I am very very interested in the new functionality "`lockmode"` I really wanted to update. I managed to perform a factory reset, and now OTA is possible again (and the time is OK as well). 

So I currently run `Faikin-S1-PICO: cc59320 2023-11-07T07:00:47`, and directly enabled `"lockmode"`

I re-enabled debug, waiting for the next time the module goes down.

revk, 2023-11-07:
OK, good luck, and weird...

HV-NL, 2023-11-16:
This evening the Faikin module went down again. I cannot ping it, there is no new Wifi accesspoint, no LED.

Just like the previous time, the module announces that it is going down `(up=false)`, and is now effectively dead (until I disconnect the power of the outdoor unit and restart the system).
`{"time": [2023, 11, 16, 17, 30, 48, 3, 320, 0], "topic": "state/Faikin-eetkamer", "message": {"up": false}}
`

These were the last mqtt messages (including the mqtt debug messages) leading to this situation:
```{"time": [2023, 11, 16, 17, 29, 58, 3, 320, 0], "topic": "308398CEE264", "message": {"online": true, "target": 17.5, "temp": 19.0, "outside": 7.5, "liquid": 34.5, "comp": 16, "fanfreq": 14.2, "bletemp": 17.39, "blehum": 63.82, "mode": "heat", "fan": "4", "swing": "off", "preset": "eco"}}
{"time": [2023, 11, 16, 17, 30, 1, 3, 320, 0], "topic": "info/Faikin-eetkamer/s21", "message": {"G1": "14?6", "G2": "=;\u0000\u0080", "G3": "000\u0000", "G4": "0\u00b0\u00800", "G5": "000\u0080", "G6": "0004", "G7": "l2@`", "G8": "0200", "GA": "99F1", "GB": "0441", "GC": "6F01", "GG": "0000", "GK": "q{51", "GM": "C6A2", "GN": "1000", "GP": "5C00", "GQ": "5C00", "GS": "0000", "GT": "3000", "SH": "091+", "SI": "043+", "Sa": "570+", "SL": "680", "Sd": "610", "SN": "030+", "SX": "081+", "SD": "000"}}
{"time": [2023, 11, 16, 17, 30, 1, 3, 320, 0], "topic": "state/Faikin-eetkamer/status", "message": {"protocol": "S21", "env": 17.4, "online": true, "home": 19.0, "heat": true, "fanrpm": 860, "comp": 16, "outside": 7.5, "liquid": 34.0, "power": true, "mode": "H", "temp": 17.5, "fan": "4", "swingh": false, "swingv": false, "econo": true, "powerful": false, "comfort": false, "streamer": false, "sensor": false, "quiet": false, "ble": {"temp": 17.39, "hum": 63.82, "bat": 1739, "volt": 28.06}, "autob": "A4C138DE21CE", "autor": 0.0, "autot": 16.5, "auto0": "00:00", "auto1": "00:00", "autop": false}}
{"time": [2023, 11, 16, 17, 30, 1, 3, 320, 0], "topic": "308398CEE264", "message": {"online": true, "target": 17.5, "temp": 19.0, "outside": 7.5, "liquid": 34.0, "comp": 16, "fanfreq": 14.3, "bletemp": 17.39, "blehum": 63.82, "mode": "heat", "fan": "4", "swing": "off", "preset": "eco"}}
{"time": [2023, 11, 16, 17, 30, 1, 3, 320, 0], "topic": "Faikin/Faikin-eetkamer", "message": {"protocol": "S21", "ts": "2023-11-16T16:30:01Z", "env": [17.39, 17.4, 17.42], "online": true, "home": 19.0, "heat": true, "fanrpm": [850, 854, 860], "comp": 16, "outside": [7.5, 7.8, 8.0], "liquid": [34.0, 34.1, 34.5], "power": true, "mode": "H", "temp": 17.5, "fan": "4", "swingh": false, "swingv": false, "econo": true, "powerful": false, "comfort": false, "streamer": false, "sensor": false, "quiet": false}}
{"time": [2023, 11, 16, 17, 30, 1, 3, 320, 0], "topic": "308398CEE264", "message": {"online": true, "target": 17.5, "temp": 19.0, "outside": 7.5, "liquid": 34.0, "comp": 16, "fanfreq": 14.3, "bletemp": 17.39, "blehum": 63.82, "mode": "heat", "fan": "4", "swing": "off", "preset": "eco"}}
{"time": [2023, 11, 16, 17, 30, 4, 3, 320, 0], "topic": "info/Faikin-eetkamer/s21", "message": {"G1": "14?6", "G2": "=;\u0000\u0080", "G3": "000\u0000", "G4": "0\u00b0\u00800", "G5": "000\u0080", "G6": "0004", "G7": "l2@`", "G8": "0200", "GA": "99F1", "GB": "0441", "GC": "6F01", "GG": "0000", "GK": "q{51", "GM": "C6A2", "GN": "1000", "GP": "5C00", "GQ": "5C00", "GS": "0000", "GT": "3000", "SH": "091+", "SI": "043+", "Sa": "570+", "SL": "680", "Sd": "610", "SN": "030+", "SX": "081+", "SD": "000"}}
{"time": [2023, 11, 16, 17, 30, 48, 3, 320, 0], "topic": "state/Faikin-eetkamer", "message": {"up": false}}
```

I have lots and lots of earlier mqtt messages in case they are needed, just ask 

(I currently run sw version `Faikin-S1-PICO: cc59320 2023-11-07T07:00:47`)

revk, 2023-11-17:
Did we get anywhere with serial debug?

Just to clarify, it does not "announce it is going down", the "up":false is from the MQTT server when the connection times out. So it has just stopped talking / hung.

I can certainly make a build with lots more serial debug if it is possible to connect that.

andreas-bulling, 2023-11-17:
Could it just be a hardware issue? I am running two modules with the latest firmware without any problems for months now...

revk, 2023-11-17:
If working with other modules, it could be faulty - yes.

HV-NL, 2023-11-17:
It looks as if the Faikin module does not communicate  correctly with the Daikin, and then hangs itself.
Just like the previous time after repowering the system I selected 'heat' mode, and enabled the Faikin module.
But instead of heating, I rapidly saw the liquid temperature dropping (to as low as -29.0 °C)

```{"time": [2023, 11, 17, 10, 18, 5, 4, 321, 0], "topic": "308398CEE264", "message": {"online": true, "target": 16.5, "temp": 14.5, "outside": 6.0, "liquid": -29.0, "comp": 42, "fanfreq": 0.0, "bletemp": 14.37, "blehum": 65.47, "mode": "heat", "fan": "3", "swing": "off", "preset": "eco"}}
{"time": [2023, 11, 17, 10, 18, 8, 4, 321, 0], "topic": "info/Faikin-eetkamer/s21", "message": {"G1": "14=5", "G2": "=;\u0000\u0080", "G3": "000\u0000", "G4": "0\u0000\u00800", "G5": "000\u0080", "G6": "0804", "G7": "l>C0", "G8": "0200", "GA": "D9F1", "GB": "0441", "GC": "6F01", "GG": "0000", "GK": "q{51", "GM": "E8A2", "GN": "1000", "GP": "5C00", "GQ": "5C00", "GS": "0000", "GT": "3000", "SH": "541+", "SI": "092-", "Sa": "060+", "SL": "000", "Sd": "240", "SN": "030+", "SX": "591+", "SD": "000"}}
{"time": [2023, 11, 17, 10, 18, 11, 4, 321, 0], "topic": "info/Faikin-eetkamer/s21", "message": {"G1": "14=5", "G2": "=;\u0000\u0080", "G3": "000\u0000", "G4": "0\u0000\u00800", "G5": "000\u0080", "G6": "0804", "G7": "l>C0", "G8": "0200", "GA": "D9F1", "GB": "0441", "GC": "6F01", "GG": "0000", "GK": "q{51", "GM": "E8A2", "GN": "1000", "GP": "5C00", "GQ": "5C00", "GS": "0000", "GT": "3000", "SH": "541+", "SI": "092-", "Sa": "060+", "SL": "000", "Sd": "240", "SN": "030+", "SX": "591+", "SD": "000"}}
{"time": [2023, 11, 17, 10, 18, 14, 4, 321, 0], "topic": "info/Faikin-eetkamer/s21", "message": {"G1": "14=5", "G2": "=;\u0000\u0080", "G3": "000\u0000", "G4": "0\u0000\u00800", "G5": "000\u0080", "G6": "0804", "G7": "l>C0", "G8": "0200", "GA": "D9F1", "GB": "0441", "GC": "6F01", "GG": "0000", "GK": "q{51", "GM": "E8A2", "GN": "1000", "GP": "5C00", "GQ": "5C00", "GS": "0000", "GT": "3000", "SH": "541+", "SI": "092-", "Sa": "560+", "SL": "000", "Sd": "240", "SN": "030+", "SX": "591+", "SD": "000"}}
{"time": [2023, 11, 17, 10, 18, 14, 4, 321, 0], "topic": "state/Faikin-eetkamer/status", "message": {"protocol": "S21", "env": 14.4, "online": true, "home": 14.5, "heat": true, "fanrpm": 0, "comp": 42, "outside": 6.5, "liquid": -29.0, "power": true, "mode": "H", "temp": 16.5, "fan": "3", "swingh": false, "swingv": false, "econo": true, "powerful": false, "comfort": false, "streamer": false, "sensor": false, "quiet": false, "ble": {"temp": 14.37, "hum": 65.45, "bat": 1437, "volt": 27.71}, "autob": "A4C138DE21CE", "autor": 0.0, "autot": 16.5, "auto0": "00:00", "auto1": "00:00", "autop": false}}
{"time": [2023, 11, 17, 10, 18, 14, 4, 321, 0], "topic": "308398CEE264", "message": {"online": true, "target": 16.5, "temp": 14.5, "outside": 6.5, "liquid": -29.0, "comp": 42, "fanfreq": 0.0, "bletemp": 14.37, "blehum": 65.45, "mode": "heat", "fan": "3", "swing": "off", "preset": "eco"}}
```

After disabling the Faikin (using the faikin webpage) and selecting 'heat' again (the website shows 'auto' soon after switching 0/1 to 0) and enabling '1' all went well, and heating started.

This is the second time that the Faikin module instructs the Daikin to start heating, but the Daikin starts cooling instead.

I collected all mqtt messages from the moment I started heating (sorry: tried to start heating) upto the moment that I see that the value of liquid starts dropping: see 
[cool_iso_heat.txt](https://github.com/revk/ESP32-Faikin/files/13390550/cool_iso_heat.txt)

Can you see any strange commands in these mqtt debug strings?  Or can you conclude a hardware mismatch or so?
I have no serial connections. I use the 'bare' Faikin module, so I did not solder anything on the backside of the board (yet).
This is all logging I have, as I do not know if I can configure a syslog IP address.

revk, 2023-11-17:
It does not do syslog, but can be physically connected to a serial debug (pads on the back of the PCB), which would only really be useful at catching why it is hanging / locking up.

I'll see if I can see anything odd in those logs.

But if other modules are working, it may well be worth swapping around to confirm if this specific model plays up, and if so, send back and get a replacement.

revk, 2023-11-17:
OK, your log shows...

- `G1` reporting `07=5` which is *off* and mode `7` which I have as *cool*
- `D1` send `14=5` which is *on* and mode `4` which I have as *heat*
- `G1` reporting `14=5` which is *on* and mode `4` which I have as *heat*

Now, unless we have the protocol totally wrong, and mode `4` is not *heat*, this looks sensible.

Can you confirm the modes using the original remote and looking at the `G1` report following each mode?

HV-NL, 2023-11-17:
My impression is `auto`=`7`:

auto
`"G1": "07?5", "G1": "07K5"`

dry
`"G1": "02\u0080A"`

cool
`"G1": "03J5"`

heat
`"G1": "04?5"`

revk, 2023-11-17:
Hmm maybe auto is `7`, but heat is `4` which is what was set

HV-NL, 2023-11-27:
Yesterday the module was dead again. No Wifi accesspoint, no ping possible. It was completely dead, as the automated stop of my indoor unit also did not occur at the configured time.

This morning I powered-off the Daikin system and soldered some wires to the back of the Faikin board in order to be able to start logging using a serial connection. But I do not have any info about the power-levels of these connections, so at this moment I do have some wires, but nothing connected to them yet. I want to use remote logging, so yesterday I prepared an ESP8266 with ESPEasy serial-to-TCP software. 

The Faikin board indicates 'ground, 0, +, R, T' . But what are the power levels? Can I power my ESP8266 using 0 and +? And do R and T use 3,3V levels?

After powering-on the Daikin system I used the now working Faikin webpage to start heating. But like the previous time I heard the sound of cooling, and indeed, in no time the liquid temperature dropped below zero. So I used the Faikin web interface to stop (0/1 => 0).
The interior Daikin unit closed, so that part went well. But on my Domoticz system I could see that the outside unit kept on cooling really hard (48 Hz). Looking at the Faikin webpage I could confirm that the liquid temperature kept dropping. In the end I had to use the remote control to really stop the outdoor unit.

So I have the impression that the first time after a power-on of the whole Daikin system the Faikin module does not communicate properly with my FTXM42N, and as a result my MXM68N outdoor unit does not start heating, but starts cooling instead.

Again I powered-off the whole Daikin system, and after power-on I started heating using the remote control. Now the system is heating again.

1) what are the power levels of the soldering dots?
2) there really seems to be a mismatch between the Faikin and my FTXM42N at initial start-up
3) the Faikin keeps hanging itself at random moments

revk, 2023-11-27:
The 5 pin connections are `⏚` is ground, `0` is GPIO0 (can leave unconnected), `+` is power (3.3V) but will be powered by Daikin, `R` is Rx, `T` is Tx. All working on 3.3V logic. You probably only need `⏚` and `T` (to your rx) - 3.3V levels.

I still have no clue as to why you have cooling, as the message is clearly to heat!

I am hoping the serial log will give some clue why it is hanging.

HV-NL, 2023-11-29:
Around 12 hr today I replaced the ESP8266 by an ESP32, and now I do get debug output.
The first output already shows a problem (`CORRUPTED`) and after restart a watchdog related message (`task_wdt: esp_task_wdt_reset(783): task not found`), 
But this one has lead to a restart, not yet to a 'hang' situation:
 
```pi@rpi-owncloud:~ $ screen -r
21741:0x3ffee5d8 |<-CORRUPTED

ELF file SHA256: ba78b7e9d715426f

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1336
load:0x40078000,len:13840
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3036
entry 0x4008058c
E (283) task_wdt: esp_task_wdt_reset(783): task not found
E (1762) RevK: Not using LED GPIO 47
E (1763) RevK: Not using LED GPIO 47
```
Current sw version: `Faikin-S1-PICO: 4dfbddd 2023-11-18T12:22:21`

revk, 2023-11-29:
I don't yet understand the `wdt` message to be honest, I do not think it is a problem.

You must have the setting for the led wrong (`blink`) for the board, the default (for ESP32-S3 build) is indeed `47` but I assume this is running on something other than an ESP32-S3.

What would be useful would be for a higher logging debug output level - are you building code or using mine. If mine, I can sort a test with more debug if you need.

HV-NL, 2023-11-29:
As shown in the picture I took at oct. 25 (see above) the 2 (identical) modules you sent me both contain an `ESP32-PICO-MINI-02` (both Faikin board `2023-09-17`). There is no indication of 'S3' on the ESP chip. 

> I don't yet understand the wdt message to be honest, I do not think it is a problem.

In my opinion a working watchdog can possibly prevent the system from hanging in case of a crash

> are you building code or using mine

I don't have a build environment, I am using the 'official builds' (using OTA):
`"otahost": "ota.revk.uk"`

HV-NL, 2023-11-29:
FYI: In the meantime a new `CORRUPTED` crash occurred (no watchdog error message this time):

```
Backtrace: 0x40081bde:0x3ffd1340 0x400945ed:0x3ffd1360 0x40094e72:0x3ffd1380 0x400957b7:0x3ffd1400 0x40094834:0x3ffd1420 0x400947e6:0x3ffee5d4 |<-CORRUPTED

ELF file SHA256: ba78b7e9d715426f

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsik_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1336
load:0x40078000,len:13840
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3036
entry 0x4008058c
E (1762) RevK: Not using LED GPIO 47
E (1763) RevK: Not using LED GPIO 47
```

When you are ready, please give me the URL of the test server. I will then instruct my Faikin to use that test environment

revk, 2023-11-29:
Tomorrow. I have to make the test builds use different logging. That said I wonder how it is "only you"?!

HV-NL, 2023-11-29:
Perhaps related to the FTMX42N in relation to the 4 pins S21 cable, as suggested
 by wurmtal868
> [wurmtal868](https://github.com/wurmtal868) commented [Nov 6, 2023](https://github.com/revk/ESP32-Faikin/issues/118#issuecomment-1794325148)
On a side note:
According to the picture above the module is missing the 5V because its connected with the 4 wire Daikin cable. On later modules the 5V are powering the TX pullup of the level shifter. I don't know if this could cause issues on the S1 modules.

HV-NL, 2023-11-29:
3rd reboot, this time with a more informative message (stack overflow in task httpd):
```
***ERROR*** A stack overflow in task httpd has been detected.

Backtrace: 0x40081bde:0x3ffd1340 0x400945ed:0x3ffd1360 0x40094e72:0x3ffd1380 0x400957b7:0x3ffd1400 0x40094834:0x3ffd1420 0x400947e6:0x3ffee5d4 |<-CORRUPTED
```

HV-NL, 2023-11-29:
It just hung itself. Big deception: no debug output at all. I look forward at a version with more debug output (or, way better: 2 fresh Faikin modules with ESP32-S3 processors that do not hang themselves)

revk, 2023-11-30:
This really is odd, you seem to have many issues.

1. Hanging, which may be easier to understand with more debug. Can you confirm the version line on the bottom of the web interface and I'll make a test. Nobody else reporting this, so very odd.

2. Stack overflow, which is new, and not hanging. Nobody else seeing this either, so also very odd.

3. GPIO 47 logged as an issue, which makes no sense as that is for S3 code.

I suspect I can sort replacements from your Amazon order number if you can confirm it.

HV-NL, 2023-11-30:
The Faikin module crashed 10x since my last message of yesterday, then performed an OTA update, then crashed 1x after the OTA update. (`***ERROR*** A stack overflow in task httpd has been detected.`)
Current SW version is `Faikin-S1-PICO: bedcd0c 2023-11-30T09:08:25`

> I suspect I can sort replacements from your Amazon order number if you can confirm it.

See Oct 25 : _I live in the Netherlands. September 3 2023 I bought 2 identical modules (1 for myself, 1 for my neighbor) at amazon.co.uk (Order # 204-1451269-3753969 )_

revk, 2023-11-30:
My concern is "A stack overflow in task httpd has been detected." is likely to happen with the replacement if the cause is something you are doing to it differently to everyone else. So a little more investigation is probably in order.

What are you doing with its web interface?

Does it hang or crash at all if you stop doing that?

revk, 2023-11-30:
It is complicated trying to sort multiple issues, and I wonder if the issue with httpd stack could be leading to the hang ups as well. I suspect most users are not hitting the httpd interface at all that much, leaving the web page open simply holds a web socket open and does not reload anything. So it could be there is an underlying issue with httpd working.

I'll have more of a look in that area in case I can see any issues and/or change stack.

[22 more replies]

closed 2023-12-25 by revk