revk/xESP32-Faikout #123: some values to high

lorbetzki, 2023-10-05:
since the last firmware update the values RPM and comp are factor 10 too high

[image]

revk, 2023-10-05:
We have had various looks at this.

What is the reference you have that tells you the actual RPM (290?) instead?

lorbetzki, 2023-10-05:
>What is the reference you have that tells you the actual RPM (290?) instead?
1. my ears, a start of 2900! RPM is MUCH louder than 290rpm in any case ;)
2. tests and logics... look at this picture
[image]
 more than 10.000RPM to cool down from 19° to 18° ? In this case the AC must be horrbile loud. 

Comp is, as far as i know a percentual value. now i have after starting betwen 140%-320%... For the Firmware-upgrade it was by 14-32%. Thats look better.

Compared with the original Daikin Wifi module and the previous Faikin firmware the values fit. with the actual firmware exactly 10 times to high

revk, 2023-10-05:
Fair enough.

What concerns me is whether this is just wrong in my code or if it is something that varies from model to model and needs a setting.

I'll see what I can do. Trying to get some rest after a stroke at the moment, but this should be pretty simple.

revk, 2023-10-05:
OK it does indeed look like a mistake, new code issue in a few minutes.

closed 2023-10-05 by revk