revk/xESP32-Faikout #125: menuconfig setting modified by the default files

wacsy, 2023-10-11:
I am trying to run make menuconfig to change some settings but it always changed back by the default files under:
ESP32-Faikin/ESP32/ESP32-BLE-Env/sdkconfig.defaults
ESP32-Faikin/ESP32/ESP32-BLE-Env/components/ESP32-RevK/sdkconfig.defaults

Anything we can do to prevent the file changed back, please?

revk, 2023-10-11:
This is a feature of setting the target, the idf.py copies sdkconfig.defaults.

You should be ok to just `make` after doing a `make menuconfig`, if you have already set the right target.

But `make pico`, or some such, or making all targets, will end up copying from defaults.

You can change defaults if needed for your build.

closed 2023-10-11 by revk

wacsy, 2023-10-11:
> This is a feature of setting the target, the `idf.py` copies `sdkconfig.defaults`. It confused me for a while as well :-)
> 
> You should be ok to just `make` after doing a `make menuconfig`, if you have already set the right target.
> 
> But `make pico`, or some such, or making all targets, will end up copying from defaults.
> 
> You can change defaults if needed for your build.

Thanks a lot for the quickly reply. May I ask how should I set the target, please? Is that in menuconfig as well, please?

wacsy, 2023-10-11:
The problem on SPI external PSRAM occurred after the ESP32-S3 working for few hours. If I set the SPIRAM configs in menuconfig in the beginning and flash, the ESP32-S3 will reboot forever. If I make and flash it without SPIRAM config then it can boot and work with my aircon for few hours and dead. And the UART will give the external PSRAM error again and reboot forever like I use a wrong setting. I will try to erase the flash and try again.

revk, 2023-10-11:
My build assumes S3 has PSRAM, so if you don't that will need building without using my `setbuildsuffix` command being used. But I don't see why it would fail after a few hours unless some unknown memory leak is involved.

Sounds like a tricky one, sorry.