revk/xESP32-Faikout #128: Faikin auto mode

ckvsoft, 2023-10-15:
Can someone please explain to me how the Faikin auto.mode works?
I set a target temperature of 25 degrees Celsius.
Faikin auto.mode is turned on at +/* 1 degree Celsius.

But what happens?
The temperature is set at 32 degrees Celsius. Nothing happens for a while. Daikin Temp shows 18 degrees, Coil shows 30 degrees celsius.

After what feels like 5 minutes I hear the Daikin fan, which then goes to 900RPM

Shortly afterwards the temperature goes to 19 degrees, the set temperature is set to 19.5
Coil is at 35 degrees. Then the temperature is reduced again, the temperature display goes to 18.5 degrees, the coil never reaches more than 35 degrees,
This loop is repeated a few times before the set temperature is set to 16 degrees Celsius.

The room is not yet close to the set target temperature.

However, the Daikin reaches this temperature in manual mode and the coil temperature goes up to 50 degrees.

revk, 2023-10-15:
The coil temp should not come in to it. What is the room temp doing?

It sets much higher or lower than current temp until the temperature is getting close to what you want. It also tries to predict when the temp will hit target over several minutes.

Here I use with an external sensor for the room rather than the built in one but it should work the same.

My room can usually stay well within a ½C window at my bed.

It is a shame if not working well for you. Obviously the air-con built in auto mode may work better for you if not using an external sensor.

ckvsoft, 2023-10-16:
Thanks. Will try Daikin auto. Unfortunately I can't set the time. This only works via the cloud.
An external sensor may not be recognized here. There's definitely something missing from the mqtt message

closed 2023-10-16 by ckvsoft

revk, 2023-10-17:
Time "on the Daikin"? If you find what messages do that I am happy to add that.

What is missing from MQTT?