revk/xESP32-Faikout #129: Strange (?) behaviour in auto mode

andreas-bulling, 2023-10-17:
I've been playing around with the auto mode - the temperatures are low enough now to see if/how well the AC can be used instead of the radiators.

I activated auto mode and noticed that the temperature shown next to the slider in the top part of the interface suddenly jumped to 29 degrees (showing a star as well). It was unstable at first, jumping down to 11 degrees and then back up to 29 - but now it has stabilised).

Also, for some reason, the mode in the top half of the controls switched to "cool" - in HA it still shows the mode to be "auto". Why "cool" I have no idea - I set the target temperature of the auto mode to 22 degrees (+/- 0.5 degrees). In one room we are at 22.2 degress and in the other 21.8 degrees.

Not sure if this is as intended - but I thought I'd report it anyway.

Do you have a recommendation to which value to set the target temperature difference? Does 0.5 degrees work fine in practice or is that too "tight" and 1 degree is better?

andreas-bulling, 2023-10-17:
Something else - given the above settings I assume the AC should turn on and heat once the temperature is below the target - 0.5 degrees. And cool once above target + 0.5 degrees? That currently is the case but nothing happens, i.e. the AC stays off. Or do I misunderstand?

revk, 2023-10-17:
It has prediction as well, so it not a simple on/off at the thresholds. Those are also settings that can be adjusted.

This is an example of my bedroom in summer on a 1C scale.

[image]

andreas-bulling, 2023-10-17:
I see, thanks for the clarification. I've been observing the mode throughout the day and it seems to work pretty well. I am running a 1C deviation now as well to avoid switches.

One thing I did notice though: When it switches on it let's the indoor units run pretty high, i.e. the fan is at full speed. It runs for a certain amount of time, then switches of completely again. Maybe that's just my settings but it would be nicer if it reacted/switched less "strongly". Not only because the high-speed fans are rather loud, also because I could imagine that's not good for the indoor units in the long run.

revk, 2023-10-17:
There is some logic to run fan high at start until it reaches target. It then adjust fan depending on whether it is getting to target or overshooting, etc.

andreas-bulling, 2023-10-17:
OK, so it seems all is normal/works as intended. Great! 

I can only repeat myself - this is a super cool and unique feature. I can directly compare the temperatures measured by the indoor units and the external sensor. And the external one is a so much better reference.

From my perspective now it's "only" about supporting more features. And offering these in a nice way in HA.

closed 2023-10-28 by revk