revk/xESP32-Faikout #131: loopback check Rx error

wacsy, 2023-10-20:
I have managed to compile it for wroom1 and flash successfully. When I trying to do the loopback check it shows good in the beginning. Later on, it comes with follow error:
error/ductair/comms {"protocol":"loopback","badhead":true,"mismatch":true,"badrxlen":true,"badform":true,"data":"00014606AA0701"}

Any suggestion, please?

revk, 2023-10-20:
Loopback should just work, if you have looped back tx an rx. I don't suppose you have an oscilloscope. What board is it? The latest Faikin boards may need the 5V connected (for the pull up) for loop back to work reliably.

wacsy, 2023-10-20:
> Loopback should just work, if you have looped back tx an rx. I don't suppose you have an oscilloscope. What board is it? The latest Faikin boards may need the 5V connected (for the pull up) for loop back to work reliably.

Thank you for the quick reply. It has been solved after I re-soldering everything. It is a customised board connected to a ESP32 S1 V1 board. I am trying to see if it get the same memory leakage issue like what I saw on ESP32 S3 board.

Is it a little bit over kill by using ESP32 S3 board, please?

revk, 2023-10-21:
I use the S3 as the S1 boards were not in stock. Also the S3 works out just as cheap. It also saves space and money on all my other boards as it has a built in USB.

Yes, for Faikin, one of the single CPU processors would be cheaper. But the S3 with extra RAM is much more generally useful, and I still want to add Matter support which is easier with the extra memory on that module.

closed 2023-10-28 by revk