revk/xESP32-Faikout #133: BLE Device

ckvsoft, 2023-10-27:
I would be interested to know how Faikin gets the BLE data. Is this queried directly via Bluetooth or in the mqtt broker?
The strange thing is,
In Faikin "none" was always offered as a choice. Thereupon
I installed ESP32-BLE-Env on an ESP32. This was apparent relatively quickly in the mqtt broker. In Faikin it took a while until he was available for selection.

Now I sent the same values ​​to the mqtt broker via my Raspberry with onewire sensor. However, this data is not recognized by Daikin. What are you looking for/filtering for here?
I turned on the ESP32-BLE-Env for testing purposes and the BLE entry remains in Faikin. Even if I delete all related data from the mqtt broker.
Please provide some details
Thanks

revk, 2023-10-27:
The Faikin uses same library as ESP32-BLE-Env, just make sure latest version.

So it should see the same devices, directly, via BLE.

The library recognises a few different types, including ELA Bluecoin, and refreshed simple temperature sensors with ESP Home mode, and another mode.

wurmtal868, 2023-10-27:
For Xaomi device I had to flash the custom FW using the Telink Flasher webpage. I had to disconnect the Telink Flasher before the sensor appeared in the menu of the Faikin module.

revk, 2023-10-27:
The sensor has to be advertising, which normally means it has to not be in a "connected" BLE session, if that helps.

ckvsoft, 2023-10-27:
My mistake. Until now I thought I needed ESP32-BLE-Env which reads the device and forwards it to the mqtt broker.
And then Faikin gets the temperature from the mqtt brocker. I then manually sent such entries to an mqtt broker and was wondering why Faikin doesn't recognize this data.
I misunderstood and misinterpreted everything

revk, 2023-10-27:
There are also commands via MQTT to tell the Faikin current and target temp from my environmental monitor/display project - they are also in the manual.

But no, the BLE is done direct.

ckvsoft, 2023-10-27:
can you give me an example?
I can't find anything about what an mqtt message looks like

revk, 2023-10-28:
Best to look in the manual. An MQTT message has a topic (text with / separators) and a payload (which can be anything but is quite often JSON these days).

ckvsoft, 2023-10-28:
Yes, I already know what a message looks like. Just not the one for Faikin. where does Faikin expect this point to be accepted as the current temperature. I can't find any information in the description

revk, 2023-10-28:
OK, so `command/nameoffaikin/control` with payload like `{"env":20.6,"target}:[20,21]}`

[image]

ckvsoft, 2023-10-28:
Thanks. I guess I missed that :)

ckvsoft, 2023-10-28:
should I see the sent temperature on the Daikin webpage? Like with BLE-Env?

revk, 2023-10-28:
Hmm, not sure, maybe.

ckvsoft, 2023-10-28:
If you ever have time. I would like the sent remote temperature to be displayed and used in Faikin Web in the same way as the BLE device temperature is displayed and used. That would be great
Thanks

revk, 2023-10-28:
I'll have to look in to it... Should be easy enough to add to the web page.

ckvsoft, 2023-10-28:
Wenn Du schon dabei bist. Kannst Du vielleicht auch die Außentemperatur anzeigen?

revk, 2023-10-28:
I thought it did that, but OK, yeh, will look at what temperatures we have and display them all.

ckvsoft, 2023-11-01:
@revk 
Since the update (Faikin-S1-PICO: faf663c 2023-11-01T11:51:03) the coil temperature is no longer displayed instead of additional temperatures

revk, 2023-11-01:
I cannot see any changes that would have caused that!

However, I agree, mine is not showing it either!

Ah, javascript error, FFS, OK, one sec.

ckvsoft, 2023-11-01:
@revk 
Nice. Is back ;-)
Are you on the other side of the temperature display or are you busy doing something else?

revk, 2023-11-01:
Uh?

revk, 2023-11-01:
Ah, OK, is `outside` the only one left?

revk, 2023-11-01:
Ok I have added outside, making an release now.

ckvsoft, 2023-11-01:
That's great. Thanks
Now all that would be missing is the temperature for remote (command.DaikinWZ.control, DaikinWZ is my Faikin climate in the living room)

revk, 2023-11-01:
OK is that a temp we have? Do you know the S21 command to get it?

ckvsoft, 2023-11-01:
The temperature comes from the mqtt server. Or have I misunderstood that?

ckvsoft, 2023-11-01:
ESP32-Envmon sends to the mqtt server and faikin then also displays this temperature from the mqtt server. That's what I mean

revk, 2023-11-01:
Sorry, I am being thick today, ok, I'll take a look.

revk, 2023-11-01:
I may look in the morning

ckvsoft, 2023-11-01:
Thanks. No problem
Just don't stress, I'll survive until then ;-)

revk, 2023-11-02:
It is showing the external temp but next to the internal.

My plan it so re-layout all the temps we show to much clearer. Maybe even add tooltips.

Hopefully this morning.

revk, 2023-11-02:
I have changed the display for temperatures (and BLE humidity) completely. Should be a lot clearer.

[image]

ckvsoft, 2023-11-02:
That's pretty clear.
If I want to emulate an EnvMon on my Raspberry, sending the message command.Faikin.control is probably not enough. Can you tell me which message faikin is looking for to find out the status?

revk, 2023-11-02:
It just expects the control message regularly (not sure of timeout, maybe once a minute). If they stop for any length of time it reverts to normal control mode rather than trying to manage the aircon to a remote temperature/target.

ckvsoft, 2023-11-02:
Thanks.
That's strange, I send the message every second
`command/nameoffaikin/control` with payload like `{"env":20.6,"target}:[20,21]}` but it doesn't show up in Faikin.

I'm assuming that `nameoffaikin` is the name I gave my climate. In my case `DaikinWZ`

revk, 2023-11-02:
Is that a typo? Target ending } not "  ?

ckvsoft, 2023-11-02:
Yes ;-)
`{"env":21.44,"target":[22,23]}`

revk, 2023-11-02:
Working now?

ckvsoft, 2023-11-02:
Unfortunately no.
Are you sure there isn't some way to check whether a device is online? You can't tell from the command whether there are any changes coming. For example, my temperature has risen. not changed for 2 minutes.

revk, 2023-11-02:
Well, it is on-line if it is sending us messages, and not on-line if it is not. That seems pretty simple.

MQTT can do retained messages and "last will" messages, which means a device could leave a "state" in the MQTT server that says on-line or off line, but making such messages for your app will be much more work, and at the end of the day, just "whether we are seeing the messages" seems the obvious way to do this and is pretty fail safe - i.e. even if a device claims to be on-line, if not sending messages we should stop assuming it is on-line.

Example of my MQTT messages for this, if that helps

`
command/GuestAC/control {"env":19.12,"target":[16.000,32.000]}
`

No, the `.000` won't be significant. Send at least every minute (there is a config setting for the timeout I believe).

Just to check, you have BLE turned off on the Faikin, yes, as it does either MQTT external messages or BLE external, not both.

ckvsoft, 2023-11-02:
It seems like I'm too stupid for it or we're talking past each other.

Example:
I send a message to the MQTT server 10.1.1.254 port 11883

`command/DaikinWZ/control {"env":21.1,"target":[22.0,23.0]}`

These are saved in the data point.

On the other hand, Faikin is connected to this MQTT server and checks whether there is anything in the data point command/DaikinWZ/control.

However, this data point is not deleted if I no longer send messages. Aside from the fact that I don't see anything in Faikin, how would Faikin know it wasn't an old entry?

ckvsoft, 2023-11-02:
I think the data points don’t matter. They are created additionally. I was confused too.

When I look with the MQTT Explorer I see that Published.
Is there the possibility to log/debug output with Faikin?

Somehow it seems like Faikin doesn't subscribe to anything
can I see this somewhere?

revk, 2023-11-02:
Subscriptions do happen, and some mqtt brokers can let you see them.

What exactly is MQTT explorer showing you for it?

Yes, adding debug would be possible - is this running my code / release. I can set up a test release if you can change the `otahost` settings.

ckvsoft, 2023-11-02:
Yes, I think I can change the otahost.

I'm now using a different mqtt server. However, that doesn't make any difference

revk, 2023-11-02:
OK, we are once again at the *I'll be able to look tomorrow* stage of the day. You can try an `otahost` of `excalibur.belmont.cymru` and see if it will upgrade, or say up to date.

Then I can make a test release with extra debug for you.

revk, 2023-11-02:
Ok I'll look at some debug in the morning.

ckvsoft, 2023-11-02:
Ok

revk, 2023-11-03:
OK update, and it should get my test code. When we are finished, change `otahost` back to `ota.revk.uk`.

For example, this is what I see when my unit (`GuestAC`) gets a command now.

E.g. the command

`command/GuestAC/control {"env":18.77,"target":[16.000,32.000]}`

And the debug log for the command:

`info/GuestAC/debug {"prefix":"command","target":null,"suffix":"control","payload":"{\"env\":18.77,\"target\":[16.000,32.000]}"}`

ckvsoft, 2023-11-03:
I did the update. But I don't get any debug output.
debug=1 is set

It looks like no subscription or something else is expected.

revk, 2023-11-03:
You don't need to set `debug` for this, if nothing, then what you are sending is not even getting to the Faikin!

What happens if you just send a topic of say `setting/DaikinWZ` with no payload...

What do you see when the Faikin powers up, that should log something like `state/DaikinWZ`

ckvsoft, 2023-11-03:
If I change the message to test like e.g.
command/Faikin/DaikinWZ then I get an error

`DaikinWZ = {"ts":"2023-11-03T09:05:50Z","description":"Unknown","prefix":"command","suffix":"DaikinWZ","payload":{"env ":21.4,"target":[20.0,23.0]}}
`
So the messages are recognized

ckvsoft, 2023-11-03:
setting/DaikinWZ
````
{
  "rx": -27,
  "tx": -26,
  "reporting": 60,
  "tcontrol": 600,
  "tsample": 900,
  "tpredictt": 120,
  "tpredicts": 30,
  "tmax": 32,
  "tmin": 16,
  "autob": "A4C138BECD3C",
  "autot": 250,
  "push10": 1,
  "switch10": 5,
  "heatback": 6,
  "heatover": 6,
  "coolback": 6,
  "coolover": 6,
  "autop10": 5,
  "thermref": 50,
  "uart": 1,
  "ha": false,
  "debug": true,
  "protocol": 1,
  "webcontrol": 2,
  "mqtt": [
    {
      "port": 1883,
      "host": "10.1.1.109"
    }
  ],
  "wifi": {
    "pass": "AtlantisCall1",
    "ssid": "ckvsoft_ext",
    "reset": 3600
  },
  "blink": [
    -8,
    -19,
    -7
  ],
  "hostname": "DaikinWZ",
  "appname": "Faikin",
  "watchdogtime": 60,
  "tz": "CET-1CEST,M3.5.0,M10.5.0/3",
  "ntphost": "pool.ntp.org",
  "otaauto": 7,
  "otahost": "ota.revk.uk",
  "prefix": {
    "error": "error",
    "info": "info",
    "event": "event",
    "state": "state",
    "setting": "setting",
    "command": "command"
  }
}
```

ckvsoft, 2023-11-03:
state/DaikinWZ

```
{
  "ts": "2023-11-03T08:55:47Z",
  "id": "308398D3D848",
  "up": 5,
  "mqtt-up": 1,
  "app": "Faikin",
  "version": "d24d314",
  "build-suffix": "-S1-PICO",
  "build": "2023-11-03T08:31:25",
  "flash": 8388608,
  "rst": 3,
  "mem": 57560,
  "spi": 2091372,
  "ssid": "ckvsoft_ext",
  "bssid": "78D294E9E754",
  "rssi": -51,
  "chan": 6,
  "ipv4": "10.1.1.88"
}
```

revk, 2023-11-03:
OK and if you send `command/DaikinWZ/test`

ckvsoft, 2023-11-03:
status
```
{
  "protocol": "S21",
  "env": null,
  "online": true,
  "home": 13.5,
  "heat": true,
  "fanrpm": 0,
  "comp": 0,
  "outside": 9,
  "liquid": 22,
  "power": false,
  "mode": "H",
  "temp": 25.5,
  "fan": "A",
  "swingh": true,
  "swingv": false,
  "econo": false,
  "powerful": false,
  "comfort": false,
  "streamer": false,
  "sensor": false,
  "quiet": false,
  "autor": 0,
  "autot": 25,
  "auto0": "00:00",
  "auto1": "00:00",
  "autop": false
}
```

ckvsoft, 2023-11-03:
> OK and if you send `command/DaikinWZ/test`

test = {"ts":"2023-11-03T09:16:08Z","description":"Unknown","prefix":"command","suffix":"test"}

revk, 2023-11-03:
OK you do not get an `info/DaikinWZ/debug` message?!

ckvsoft, 2023-11-03:
> OK you do not get an `info/DaikinWZ/debug` message?!

no

revk, 2023-11-03:
Arg, I see  ` "otahost": "ota.revk.uk",`

I thought you changed it to get my test code!

ckvsoft, 2023-11-03:
Thought I should put it back. Sorry

ckvsoft, 2023-11-03:
switch back. now
`debug = {"prefix":"command","target":null,"suffix":"control","payload":"{\"env\":21.4,\"target\":[17.0,26.0]}"}`

[17 more replies]

closed 2023-11-03 by revk