revk/xESP32-Faikout #135: Limit allowed modes in Faikin-auto mode

JeanOlivier, 2023-10-29:
I've been experimenting with the Faikin-auto mode for a few days and it's very good.

However, I'm from a very cold place (-30°C not uncommon in winter) and it feels very inefficient when it uses the cold (and dry?) modes to bring the temperature down after a small overshoot or because I just cooked with the oven. It could just wait instead.

It would be very convenient to have "allow_cooling" and "allow_heating" flags in the WebUI so it's possible  to manually set them according to the seasons.

Thanks for considering this.

revk, 2023-10-29:
I have settings for controlling how quickly it switches.

I don't yet have a heat/cool only control, but that may be in order to add. I'll see what I can do.

JeanOlivier, 2023-10-29:
Thanks for looking into this.

I also noticed that it seems to always stabilize to the bottom of the range. E.g. when set to 22 ± 0.5 °C, it stabilizes very well, but around 21.5°C. Maybe this is because it's fairly cold today (0°C) and it's being pulled down in temp more than expected? Not a big deal since I can juste set the target 0.5°C above what I want, but just an inconvenience.

Anyways I'll look into the advanced manual for the timing commands and see if I can fix that.

revk, 2023-10-29:
I'll consider a fixed mode anyway.

It tends to be one side or the other depending on where it is coming from - it aims to be anywhere in the band and only act when it looks like it is going to go outside it (with some prediction).

JeanOlivier, 2023-10-29:
> I'll consider a fixed mode anyway.

That'd be perfect for me use case, thank you.

> It tends to be one side or the other depending on where it is coming from - it aims to be anywhere in the band and only act when it looks like it is going to go outside it (with some prediction).

In my situation — cold outside, warming up the house in the morning — it's always fighting the temperature in the same direction. If it stops heating at the bottom of the range, it will be stable there for a little while and then drop. It won't explore the available range at all.

A relatively simple fix for that would be to target the actual target temperature when starting from outside the range, then proceeding with the current algorithm once the target has been crossed, or reached within a tolerance (e.g. 0.1°C) that's not the same as the allowed range.

Thanks.

JeanOlivier, 2023-10-30:
> If it stops heating at the bottom of the range, it will be stable there for a little while and then drop. It won't explore the available range at all.

Just to add that this happens a lot now that it's colder outside.

It can turn off and on multiple times in a short timeframe, like 3 times within 5 minutes.

Some hysteresis would be really beneficial.

revk, 2023-10-31:
The current working is the result of a lot of experimentation. It works on predicting the temperature a few minutes ahead. It aims to stay in the range.

There are a number of settings you can control via MQTT to change things. That is probably where to start. See if you can find some that work better for you.

JeanOlivier, 2023-10-31:
I will play with those a little. If I only want the "faikin auto" mode on when I set the unit on myself, and not "auto turn on whenver we're way off-target", then I understand that I should look into the cool/heat over/back parameters.

Is this right?

I'll test with this and report my findings.

That said, the fixed-mode feature would still be very beneficial. This morning the AC turned on to cooldown my room from 21°C to 20°C, when it was 0°C outside. I think it overshoot in the morning because the downstairs heater started.

JeanOlivier, 2023-11-01:
Ok, I think it's the heatback and heatover that I want to change (for winter anyways).

I managed to listen to `info/daikinunit/automation` and got this:

```
{
    "hot": true,
    "approaching": 0,
    "beyond": 0,
    "initial-samples": 900,
    "period": 900,
    "temp": 20.52,
    "min": 20.1,
    "max": 21.5
}
```

Where/how can I see the current heatover/heatback values? I'd like to know where I'm starting from.

revk, 2023-11-01:
Send setting/name with no payload and it should dump all settings back.

JeanOlivier, 2023-11-01:
I'm listening to `setting/daikinchambres/#` with the `#` being a wildcard, and sending a packet to `setting/daikinchambres/heatover` with no payload.

I do see me own packet, but I don't see any reply.

revk, 2023-11-01:
Send to `setting/daikinunit`

That should give all settings.

JeanOlivier, 2023-11-01:
Alright I managed to get it. I really don't grok MQTT.

Defaults were:
```
    "heatback": 6,
    "heatover": 6,
    "coolback": 6,
    "coolover": 6,
```

I'm trying with:
```
    "heatback": 1,
    "heatover": 3,
    "coolback": 0,
    "coolover": 0,
```
for today and assessing tomorrow.

Thanks for the help.

revk, 2023-11-01:
Good luck

JeanOlivier, 2023-11-01:
Oh, I'm not complaining. I see the usefullness and it seems sane enough. It's just not a good fit for my brain.

revk, 2023-11-01:
Typo fixed - looking at other changes later (may be in morning).

JeanOlivier, 2023-11-01:
Just to be clear, if I'm staying in heating mode the relevant settings are heatover and heatback right?

revk, 2023-11-01:
I think so!

revk, 2023-11-01:
Ah, not what I would have expected. Ok. Good luck.

JeanOlivier, 2023-11-02:
Ok, so I tested with a few settings but it seems to either shoot for the moon or oscillates towards the bottom of the range.
@revk no pressure, but do you plan on allowing heat-only mode? AC started again today and that limits my capability to use the auto-faikin mode going forward.

JeanOlivier, 2023-11-03:
Sorry for the spam.

Tonight inset the mode to simply heat without theauto-faikin mode, and to my surprise it seems to have stabilized exactly at the target temperature according to the BLE thermometer.

If the BLE thermometer is connected to the faikin, is it used instead of the internal Daikin thermometer as the current temperature even in regular heat/cool mode?

revk, 2023-11-03:
We have no way to tell the Daikin the reference temperate from the BLE, so that much have been fluke.

I'll look at mode fixing.

At the very least making a setting to now switch mode should be an easy one to add.

JeanOlivier, 2023-11-03:
In though maybe you just adjusted the target internally to compensate for the delta.

So it seems it is happenstance? 

It is very surprising given that it is stable within 0.2°C of the target on the bluetooth thermometer, which is almost 3°C off from the Daikin unit (around a corner).

revk, 2023-11-03:
In the Faikin-auto mode I do adjust for the delta plus the offsets in the settings. Some people have set the offsets to zero and found it works exactly as they want.

Unless that is what you have managed to do as well.

JeanOlivier, 2023-11-03:
I did test with 0,0,0,0 and it didn't help.
I think I'm just in a very different condition than that in which you tuned the auto faikin.

I was surprised to see how well it behaved in regular heat mode as far as aiming for the BLE tempnwent. Hence why I asked if you were secretly compensating for the delta temp internally.

Anyhow, I'm now having fun implementing a PID in Python for HA, we'll see if that pans out.

I'd still appreciate a heat-only mode for the auto faikin, but I understand that you have finite time on your hands.

revk, 2023-11-03:
I think a simple setting to not change mode in auto. In settings. It is 5:30am here so later this morning.

revk, 2023-11-03:
I have added a `lockmode`, in the settings screen as well. It stops if changing heat/cool mode automatically.

closed 2023-11-03 by revk