revk/xESP32-Faikout #139: Flashing binaries directly

matt-nz, 2023-11-01:
I have one DIY Faikin setup which is just an ESP32 and shifters, this works fine and has been running flawlessly for months now, instead of building the project (I'm on Mac) I just flashed one of your prebuilt binaries and then OTA updated. I'm trying to make another one now but I'm stuck flashing can't really remember the process I took or perhaps I'm missing a boot loader? I have tried all the different binaries in GitHub.

I'm trying
`esptool.py --chip esp32 --port /dev/cu.usbserial-1410 --baud 460800 write_flash -z 0x1000 Faikin-S1.bin`

Which flashes fine but when checking serial debug all I see is a boot loop with
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:267164`

Any pointers?

revk, 2023-11-02:
I am not an expert on the flash command, I usually just use the idf.py flash.

But the main issue with my OTA are they are signed so work work if you have a bootloader that looks for signed code using a different signature.

The idf.py flash command may however show you the exact esptool flash command it uses.

I have now including the bootloader as well as my binaries in the release directory in my github though. So it should be possible to adjust that to flash my bootloader and binary and so have correct signature checking for OTA

matt-nz, 2023-11-03:
Interesting I guess that is the issue, when I flashed this a long time ago it might have been before you signed the code?

I tried 
`esptool.py --chip esp32 --port /dev/cu.usbserial-1410 --baud 460800 write_flash -z 0x1000 Faikin-S1-bootloader.bin 0x10000 Faikin-S1.bin`

But still get errors, I also tried flashing using a partition table from another project but no luck with that.

Feel this should be an easy path, can you see the exact esptool flash command and which binaries it flashes and to where?

This is on a ESP-WROOM-32 in case I'm doing anything silly.

revk, 2023-11-03:
I have always signed the code. But could be a partition table issue maybe. I can check the flash command, later this morning.

matt-nz, 2023-11-05:
Thank you for uploading the boot loader. I gave it a try using a partition table from another project but still not working. I also tried the one from https://github.com/Sonic-Amiga/ESP8266-Faikin but guessing some differences between ESP32 and 8266 bin files. I did flash https://github.com/Sonic-Amiga/ESP8266-Faikin to an 8266 successfully following the instructions there so I think this will work if you could include the `partition-table.bin` and `ota_data_initial.bin` binaries also. Sorry to be a pain I normally use platform IO and Arduino flavour of ESP so don't have the native tool chain installed and was trying to avoid having to do so.

revk, 2023-11-06:
OK, added to `release` directory.

matt-nz, 2023-11-06:
Might just be something I'm doing wrong but no luck flashing with those files. But I found a work around. 

Run prepare for first use from. https://web.esphome.io then either OTA update with the Faikin.bin or flash all other components. In my case this was: 

`esptool.py --chip esp32 -p /dev/cu.usbserial-1410 -b 460800 --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size keep 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 Faikin-S1-V1.bin`

Thanks for all your time on this great project.

closed 2023-11-06 by matt-nz

sam-m7, 2023-11-20:
I've just had the same problem trying to re-flash the binaries from the release folder after I've corrupted the settings (#151) and erasing just the settings didn't work, because the esptool wouldn't take the addresses (difference between start and end needs to be a multiple of 4096). But I've managed to get it flashed and working again using this command: `esptool.py -p /dev/ttyUSB0 write_flash 0x1000 Faikin-S1-PICO-bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 Faikin-S1-PICO.bin`

The problem was the address of the bootloader, which has to start at address 0x1000 (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/bootloader.html). Maybe this can be changed in the README of the release folder.

revk, 2023-11-20:
OK

sam-m7, 2023-11-20:
> OK

Thank you (also for the hole project in general! :) )! 
Actually the README I meant was https://github.com/revk/ESP32-Faikin/blob/main/ESP32/release/README.md, where the current command didn't work (for my PICO) because the bootloader address is 0x0.

revk, 2023-11-20:
OK next time do a pull request... done.