revk/xESP32-Faikout #140: Please make the Comfort Fan Mode and Streamer available in Home Assistant

wurmtal868, 2023-11-02:
Hello revk,
First of all, thank you for developing the Faikin Module :-)
After successful integration and testing the first module, I ordered 5 more for my remaining indoor units. 

In order to leave the Daikin cloud for good, I would be happy if could add the comfort fan and streamer mode to Home Assistant.
Both are already supported in your Web frontend.

The icing of the cake, would be an option to control the 3D sensor of my Perfera units. 
This only supported trough the IR remote, in the Onnecta App it's not available.

Regards Thomas

revk, 2023-11-02:
I thought we had these, but I'll look again.

Anything not in their app is probably not in S21 so unlikely we can support it either.

revk, 2023-11-03:
OK, happy to add to HA, if you can find what the necessary HA config is for these. Then I can add to the config we send and provide the status/commands needed.

wurmtal868, 2023-11-03:
The HACS cloud integration has added a button panel to the thermostat card.
[image]
That will do it.

revk, 2023-11-03:
Oh, interesting, that looks useful. I don't suppose you can find the JSON config needed for that?

wurmtal868, 2023-11-03:
Sorry I'm a newbie in Home Assistant. 
I'm just happy that my indoor unit popped up automatically under MQTT after initial setup. 

Eventually someone else reading this thread can help?

revk, 2023-11-03:
I may be able to find something in google... I don't use HA myself, so have had to rely on a lot of contributions for these features.

wurmtal868, 2023-11-03:
For the comfort mode it would make sense adding it to the fan swing menu, because it supersedes the h/v modes.
[image]

revk, 2023-11-03:
Ah, OK, that may be possible...

revk, 2023-11-03:
OK issued a change to add to swing settings, please can you test that?

wurmtal868, 2023-11-03:
It's working 
[image]

Thanks a lot!

revk, 2023-11-03:
OK if we work out *streamer* controls JSON for HA, I'll add that. I'll close ticket for now until we have that.

closed 2023-11-03 by revk