revk/xESP32-Faikout #141: [FEATURE] Provide a static JSON representation for status values.

ozonejunkieau, 2023-11-04:
I use Node-RED as a convenient way to extract data from MQTT and feed this data into InfluxDB for plotting in Grafana.

The status messages from Faikin are a bit challenging to deal with as numeric fields are either a single value or a list of `[min,avg,max]`. It's definitely possible to write the code that is required to handle this, but honestly it's painful and ugly, especially in Node-RED.

I was wondering if it was possible to provide a setting via MQTT that turned on a fixed representation, always returning the data as a list, just with three identical values when the value has changed over the period? This makes it much simpler to always use the middle value for entering into a monitoring database. 

My suggestion would be something like a boolean `setting/<name>/static_json` that defaults to `false`, thus maintaining compatibility.

revk, 2023-11-04:
Certainly doable - I'll have a look.

revk, 2023-11-04:
OK, done issued. Setting is `fixstatus`, and forces the array format.

closed 2023-11-04 by revk

njh, 2023-11-05:
@ozonejunkieau I am also a Ned-RED, MQTT and Faikin user 😀 

I am not sure if we are aware that the `[min,avg,max]` values published to `Faikin/...` that are published every minute are a summary for the last minute - intended for for logging/monitoring. However the [current status](https://github.com/revk/ESP32-Faikin/blob/main/Manuals/Advanced.md#status) is also supposed to be published to `state/...`.

However I had problems with it not aways publishing a new value to MQTT when something changed. My Aircon and Faikin are out of action at the moment, so I have been unable to test to see if it is working better now.

Getting a complete lump of JSON sounds like a useful new feature, thanks!
Is the new `setting/<name>/static_json` option published periodically? Or when values change?

revk, 2023-11-05:
I have not done `static_json` just changed the one monique summary to be the same format regardless when that setting is on.

There is also a live setting of some sort, yes, checking the manual, it is `livestatus`, that probably does what you want.