revk/xESP32-Faikout #142: CTXM15R2V1B not providing 14V DC on S21 port

bajo, 2023-11-05:
I'm trying to get the Faikin ESP module from Amazon (build version: 2023-07-19) up and running on this unit. 
I found CTXMxxRxV1B on https://github.com/revk/ESP32-Faikin/wiki/List-of-confirmed-working-air-con-units, thus, I assumed it to be compatible without any issue. 
On 2 other units (FTXM35R and FTXM20R) I was able to connect my DIY cables just fine.

On the CTXM15R2V1B I'm only able to measure 5V DC against GND (pin 5). As this is not enough to power on the ESP module I'm kind of stuck right now. 
I checked the module itself with a 12 V DC power supply and it is working fine. No problem in setting up my wifi, or performing OTA updates. 
Attempting to power the ESP module with 5V DC from fails, as expected. 

I found a service manual which claims that I might be able to find between 10 and 16 V DC on the connector of the original WiFi module, but I was not able to confirm that. I only get 5V, 5V, 0V, 1.25V against GND on the wifi module connector. 

Any ideas on how I could go forward from here? Appreciate any help. 
Thanks in advance.

revk, 2023-11-05:
That does seem odd, I have not seen any S21 connector without something of the order of 12V on one of the pins. 5V is enough power to run the Faikin, but the 5V on the S21 does not usually have enough current for that, so that will be why it fails.

Maybe someone else with the same model can confirm.

bajo, 2023-11-05:
@revk I tried to power the Faikin module on a lab power supply on 5V into the 12V pin. It seems to start the module in some failure mode. Current limit should not be a problem here. But it only blinks white and red at about 0.5Hz. Not really sure what that is indicating though. 

About the 5V not providing enough current on the S21 port. That means even if I would use an off-board step-down converter and feed directly into the '+' labeled header on the Faikin, it would not be able to properly run?

revk, 2023-11-05:
The Faikin has a power supply that should be OK working from 4V to 36V input, but does need up to peaks of 2.5W, which is why I suspect (when I have tried) 5V from the Daikin it did not work. It should definitely work from a bench supply or even USB lead at 5V. The latest boards have 4 pads for USB as well as 5 pads for serial (with 3.3V input), but yours may be older and only have serial pads. But it is designed for the power that is normally on an S21 connector. The one point that can fool people is pin 5 is GND, not pin 1, on the S21.

Are you sure this is not a faulty connector/cable? If the S21 does not have "power", i.e. the 10 to 16V DC, then it may not work.

wurmtal868, 2023-11-05:
I have a similar problem with an FTXM25 indoor unit. My CTXM15 unit is working fine same pcb inside the unit. 
I guess the units without the 12V are broken.
**At your own risk**, you can try to remove the 12V wire to the indoor unit and add a blue wire between 5V and 12V on the Faikin module.
This is how I got my FTXM25 working with the Faikin module.

dzungpv, 2023-11-07:
@bajo on the PCB there are are some test point they print 15V (14V) you can wire the that checkpoint and power to the faikin module, If you upload the picture I can help you identify it, Daikin also have other port to provide power to external adapter like: X35A/X18A/X9A (depending on the
indoor unit)

revk, 2023-11-07:
Ah yes, on mine (X50A) the power is from a different plug in the board. Maybe that is what you need.

codeworkx, 2023-11-11:
I have an FTXM42R which provides around 10.5V at S21. As soon as I plug in the module (S3 MINI from Amazon), the voltage drops towards 0V. Module shows no function. However, the module can be operated without any problems on a laboratory power supply unit. Firmware is from Nov 7, 2023.

revk, 2023-11-17:
Interesting.

GL640, 2024-07-13:
> The voltage dropping is most likely caused by bad solder joint on the pcb of the indoor unit. One out of six Perfera units in my installation has this issue, too. Although I was able to work around be using 5V to power the Faikin module I will start a complaint with my supplier. Because one of two identical units is working and the other is not, the one without the 14V must be broken.

so you where able to power the faikin from the 5volt pin 1 s21?

bajo, 2024-07-13:
For me it didn't work with the Faikin PCB due to another problem. See https://github.com/revk/ESP32-Faikin/issues/362 .

In the end I used a generic esp32 dev board on the 5v line with level shifters to 3.3v on rx/tx lines. 
Works without a problem after I got the configuration sorted out.

GL640, 2024-07-13:
> so you where able to power the faikin from the 5volt pin 1 s21?

well i have connected the faikin to the 5 volt pin1 and for me its working now.
i have to say that i connected the faikin to a power supply and configed it before hand ? maybe then the power draw is less at boot because it does not need a setup?
well atm its working lets so for long time.

i also have one of te latst faikins

GL640, 2024-07-14:
> For me it didn't work with the Faikin PCB due to another problem. See #362 .
> 
> In the end I used a generic esp32 dev board on the 5v line with level shifters to 3.3v on rx/tx lines. Works without a problem after I got the configuration sorted out.

so how long is that esp running now on the 5 volt line? i wonder if the power supply for the 5 volt is not build for it then it would stop working and the 5 votl is no longer present also for other parts of the a/c? what more runs from that 5 volt on the same line as the s21 5 volt?