revk/xESP32-Faikout #143: outside temp and BLE temp mixed in MQTT message

HV-NL, 2023-11-07:
I updated to `Faikin-S1-PICO: cc59320 2023-11-07T07:00:47`

Now I see that the Domoticz mqtt auto-discovery maps 2 different temperatures to the same variable. Outdoor temperature (currently 10 °C) is overwritten by the BLE temperature (18,8°C), and within the same moment is is overwritten again with 10°C. So most of the time I see 10°C, and some very short flashes of 18°C

When I disable BLE in the settings page and reboot the Faikin module the problem appears to be fixed, as no BLE temperature is sent to mqtt any more. Now I see a stable 10°C

(I was looking for the separate BLE temperature item after Domoticz had finished discovery, but when I could not find that I saw that one value very quickly is overwritten, as described above)

revk, 2023-11-08:
I know nothing of Domoticz.

Can you explain what is wrong at an MQTT level?

HV-NL, 2023-11-08:
> I know nothing of Domoticz.
> Can you explain what is wrong at an MQTT level?
> There should be, via various MQTT messages, separate "outside" and "ble" temperatures.

No, I do not have MQTT expertise, so I can only describe what I see in Domoticz. 

Just as a test I installed HomeAssistant on a fresh raspberry pi, and there I don't see this behavior. So can be a Domoticz bug with respect to the Domoticz MQTT discovery implementation.

revk, 2023-11-08:
Ok well I don't know anything of Domoticz, so if anyone else does, happy to look.

HV-NL, 2023-11-09:
I created a ticket for the Domoticz developers ([github issue 5870](https://github.com/domoticz/domoticz/issues/5870))
I hope they will be able to look into it, and give a solution.

revk, 2023-11-09:
If they see what they think is wrong with the MQTT then I may be able to make changes that are needed. Let me know.

HV-NL, 2023-11-10:
I got a reply from my Domoticz ticket: All announcements are made using the same unique identifier (I think the MAC address of the ESP, in my situation `308398CEE264`). For the current Domoticz mqtt discovery software this is a signal that all announcements belong to the same device, and it will merge the data to one temperature/humidity device.

My guess is that when you e.g. could add a letter or digit to the unique identifier to all BLE data ( `308398CEE264A`) this will effectively create a dedicated unique identifier for the BLE sensor,  and no more mixing of data will occur in Domoticz.

revk, 2023-11-10:
OK, all *are* one device, but the *device* has more than one temperature (inside, liquid, outside, etc). So sounds like Domoticz assumes that one device can *only* have one temperature which is a very wrong assumption, especially for air-conditioning.

What I don't know is whether making it seem like different devices would upset Home Assistant - does anyone familiar with HA know?

HV-NL, 2023-11-11:
The Domoticz mqtt software maintainer just solved the problem in the latest beta. When a device advertises more than one temperature it will now generate as many different devices as required. So no more mixing of outside and BL temperature!

I would suggest not to modify your advertisements, as this possibly will have an impact on HA, and my problem is solved by this Domoticz modification.

closed 2023-11-11 by HV-NL

revk, 2023-11-11:
Thank you, please pass on my appreciation. That is excellent.