revk/xESP32-Faikout #149: [BUG] Device is not working after flashing binary from release folder

UgniusBur, 2023-11-15:
**Faikin hardware**
FDXM50F9

**Daikin hardware**
2023-09-08 Faikin PCB + custom made X50A + X35A -> S21 lead.

**Describe the bug**
I have recently assembled a Faikin device to connect to my AC unit.  Everything flashes, the device creates a hotspot and successfully connects to the house wifi. After flashing and connecting the LED is always RED/WHITE it does not change any color (even after connecting wifi still red/white). If I connect 34 and 48 pins together - the LED doesn't change too. The oscilloscope on those pins is also silent. The device looks like stuck in some kind of loop(?) and doesn't change LED modes, UART is not working.  Is there any change that release firmware is not correct? I do not rule out the possibility that it is my soldering issue, but I have tried 2 different boards and the behavior is the same. Is there any way to debug print this device?

**To Reproduce**
Steps to reproduce the behavior:
1. Flash device using: command `esptool.py -p /dev/ttyACM0 -b 460800 --before default_reset --after hard_reset --chip esp32s3 write_flash --flash_mode dio --flash_size keep --flash_freq 40m 0x0 Faikin-S3-MINI-N4-R2-bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 Faikin-S3-MINI-N4-R2.bin
`
2. Connect to Faikin's Wifi
3. Enter wifi SSID  and password

**Expected behavior**
UART is sending any info. LED light changes regarding status. Loopback test is working

revk, 2023-11-16:
Sorry for delay, the loopback test may not work on latest board unless 5V is connected. Also may not show on oscilloscope unless 5V connected or connected to the Daikin which has an internal pull up.

Can you check with the 5V connected and see if there is data flowing?

UgniusBur, 2023-11-16:
Thank you for the reply.

I have tried to power up 5V. Now I see inverted UART data on lines every ~20 seconds, but no loopback mode indication.

I am a little bit concerned why my LED is not changing. The documentation mentions that it should be BLUE/WHITE(?) but the LED is always RED/WHITE no matter what.

P.S. I already tried it on Daikin - no luck. Even with RX/TX swapped. That's why I put it under the oscilloscope.

revk, 2023-11-16:
When not connected to loopback, it should cycle through inverted and non inverted and two different baud rates.

Once looped back the LEDs should change to red/green/blue cycle.

Yes, blue/white means AP and wifi mode - once connected to your wifi it should stop the AP and just be something else. red/white is non wifi config. Is it definitely connecting to your wifi?

I don't suppose you have another one in case it is faulty?

UgniusBur, 2023-11-16:
I am 100% it is connected to wifi. I can connect to Faikin control page using local network.

I have a few spare boards, they all act the same, but they are hand soldered.
Is there ant posibility to print any debug logs to the USB?

revk, 2023-11-16:
The latest boards have 4 pads for USB on the back of the processor, and allow USB logging using esptool.

Is it on mqtt? If so we can send new settings including upgrading to a test build I can make with extra debug.

UgniusBur, 2023-11-16:
I connected to the logs using ‘make monitor’.
I can see that device prints “Loopback” once, but LED is still RED.

What I also noticed:
1. TX is always HIGH (inverted uart) while rx is always LOW (normal uart). Shouldn’t it swap between interted/normal modes?
2. I can confirm that device transmits “something” and my Daikin has 5V power. But when trying to power on using webpage - nothing works. Do Faikin waiting for some kind of response?

revk, 2023-11-16:
Rx on the Faikin will be low, it expects to be drive.

Tx should (if not connected to loop back) cycle between inverted and non inverted. The MQTT should show something for this. Once it gets a sane response it knows what mode it is in and should communicate.

It could take a minute to find the right setting, but then it stores it in flash to be quick next time.

UgniusBur, 2023-11-16:
I have found why LEDs are not working:
sdkconfig defined LEDs to a single pin: `CONFIG_REVK_BLINK "47 47 47"`.
Both "Blink LED GPIO(s) defined in library" and "Blink LED update done in library" were enabled. I disabled "Blink LED GPIO(s) defined in library" and now it is working.

Now I need to find out why UART is not communicating with daikin.

revk, 2023-11-16:
Ah, sorry, missed the fact it is your build. Yes, the code has moved on with the chip design a bit.

Check the GPIOs very carefully!

Send setting/... as well or erase flash and re flash, as GPIO settings are designed to persist once first set and not follow new defaults in code.

UgniusBur, 2023-11-16:
I noticed the same behavior in release firmware too. I built it by myself only because I wanted to have more debugging options. 
I think I need to create MQTT server to be able to access settings then.

UgniusBur, 2023-11-16:
I managed to create an MQTT server.  I tried to power on Daikin using power (0/1) slider in the web application.
It instantly turns off and I can see: "{"field":"power","error":"Setting cannot be controlled"}" on topic: "error/AC/control".

How should I interpret this message?

revk, 2023-11-16:
If it has not managed to talk to the Daikin it will not know what settings can be controller. Need to get it talking first.

UgniusBur, 2023-11-16:
I swapped TX/RX lines and got messages to "error/AC/comms" topic:
`
{"protocol":"X50¬Rx","timeout":true}
{"protocol":"S21¬Rx","cmd":"F1","noack":true}
{"protocol":"X50¬Rx¬Tx","timeout":true}
{"protocol":"S21¬Rx¬Tx","cmd":"F1","noack":true}
{"protocol":"X50","timeout":true}
`

Now I have swapped lines back and this topic is silent. I assume that it already communicated.

revk, 2023-11-16:
Ok that suggests it is talking.
There is a setting for `debug` and `dump` that can be set to `1` which shows what it is sending and receiving.

UgniusBur, 2023-11-16:
I periodically get these lines:

```
{"protocol":"X50","dump":"06AA0701000146"}  
{"protocol":"X50","dump":"06AA0701210026"}
{"protocol":"X50","cmd":"AA","payload":"00"}
{"protocol":"X50","cmd":"CA","payload":"0000000000000000000000000000000000"}
{"protocol":"X50","cmd":"CB","payload":"0000"}
{"protocol":"X50","cmd":"AA","payload":"01"}
{"protocol":"X50","dump":"06AA0701000146"}
{"protocol":"X50","dump":"06AA0701210026"}
{"protocol":"X50","cmd":"AA","payload":"00"}
{"protocol":"X50","cmd":"CA","payload":"0000000000000000000000000000000000"}
{"protocol":"X50","cmd":"CB","payload":"0000"}
{"protocol":"X50","cmd":"AA","payload":"01"}
```

revk, 2023-11-16:
Hmm, does the topic explain if `tx` and/or `rx` on those? It may help.

UgniusBur, 2023-11-16:
Sorry, should have thought about it first:

```
info/AC/tx {"protocol":"X50","dump":"06AA0701000146"}
info/AC/rx {"protocol":"X50","dump":"06AA0701210026"}
info/AC/rx {"protocol":"X50","cmd":"AA","payload":"00"}
info/AC/cannot-tx {"protocol":"X50","cmd":"CA","payload":"0000000000000000000000000000000000"}
info/AC/cannot-tx {"protocol":"X50","cmd":"CB","payload":"0000"}
info/AC/tx {"protocol":"X50","cmd":"AA","payload":"01"}
info/AC/tx {"protocol":"X50","dump":"06AA0701000146"}
info/AC/rx {"protocol":"X50","dump":"06AA0701210026"}
info/AC/rx {"protocol":"X50","cmd":"AA","payload":"00"}
info/AC/cannot-tx {"protocol":"X50","cmd":"CA","payload":"0000000000000000000000000000000000"}
info/AC/cannot-tx {"protocol":"X50","cmd":"CB","payload":"0000"}
info/AC/tx {"protocol":"X50","cmd":"AA","payload":"01"}
```

revk, 2023-11-16:
OK good, data is going both ways! This is a good start...

OK, this may be something I can look in to. It may be a timing issue, as AA is getting a reply. It could be a timeout change needed.

UgniusBur, 2023-11-16:
I turned the AC unit off and on again. Looks like the output is the same...

I forgot to mention that this unit is "fresh" and was never started with any other controller. Maybe it has something to do with this "not ready" response.

revk, 2023-11-16:
OK not a scenario I have encountered. Mine had other controllers, but did not expect any sort of setting the enable them. I guess it is possible. I mean we could carry one ignoring the `00` response perhaps. Line 479-480 of `Faikin.c` if you want to play.

revk, 2023-11-16:
That would be a shame. I hope it works.

UgniusBur, 2023-11-17:
I bought an original Daikin controller. It is not able to connect either. I sniffed UART - I get the same response.
1. I got faulty Daikin.
2. I should do some kind of start-up procedure I don't know about.

revk, 2023-11-17:
Wow, and if you find an installation process for it, please do let me know :-)

UgniusBur, 2023-11-21:
Okay, so it was my mistake. Ducted Daikins must have remote controller: wall mounted or infrared in order to work. Just Wi-Fi controller is not enough for proper functionality.  They all must be purchased separately.

I have installed wall mounted controller and now Wi-Fi is working. For now I just tested the original Wi-Fi adapter, not Faikin but I already hate the genuine one. The Onecta app is terrible. If you want you change the temperature you have to set it multiple times because it likes to "wooble back" (I think it reads actual value from the cloud and sets the temperature back) and such simple operation becomes exhausting.

Also I regret my choice to install ducted Daikin. I wanted to hide it in the ceiling, and it hid well BUT IT SCREAMS ALL THE TIME. There is no auto fan mode. You just set the fan speed and it spins 24/7. No matter what temperature. It doesn't bother during the day time but that constant fan blow is terrible at night. The lowest fan is already too loud. I would love if I would be able to make it spin a little bit slower or just turn it off when the temperature is right. Now I have to manually turn on/turn off the entire AC to achieve such functionality and I don't know if it is very efficient.

UgniusBur, 2023-11-21:
By the way I am closing this issue but I still see that release firmware does not control LEDs. It just lights them all and RGB is always white. I can open another issue for this.

closed 2023-11-21 by UgniusBur

revk, 2023-11-21:
OK? But check what is set in blink matches the GPIOs