revk/xESP32-Faikout #150: [FEATURE] Home Assioistant sensor GUI clean up

wurmtal868, 2023-11-18:
The latest FW 11/7/23 shows fanfreq instead of fanrpm. Fanrpm is always zero and therefore useless information.
It would be nice if Home Assistant shows the RPM like the Faikin web page.
Another nice to have:
The bletemp should show the normal thermometer icon.

[image]

revk, 2023-11-18:
I knew this would come back to haunt me!

`fanrpm` was changed to `fanfreq`. There is no `fanrpm` now - your HA install has remembered `fanrpm` and I have no idea how you clear that.

It was changed because someone pointed out that `rpm` is not a valid unit in HA, so was causing errors. `Hz` is the unit, hence the change.

If someone with HA experience can explain how I can use `rpm` without errors, I am pretty happy to put it back, or maybe make it a setting even.

The wrong icon for `bletemp` was a copy/paste issue by me, sorry. I'll issue new code for that now.

Make sense?

wurmtal868, 2023-11-20:
Thanks for the explanation. 
I don't care whether my indoor units show Hz or rpm.
The thermometer icon helps only when you add a new unit HA. Therefore I called "Nice to have".

A broken dashboard triggered me to start this issue. I took me a while to figure out, that an automated OTA pushed this change.

I wonder if it would make sense to add a check box to disable the automated FW update.
While I appreciate your ongoing efforts in improving the Faikin FW, pushing a new FW to every one, will haunt you if breaks something that is in use by someone who has otherwise no benefit from the new version.

patrickli, 2023-11-25:
Related issue, now the following flood in home assistant log:

```
2023-11-25 21:45:35.812 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'fanrpm' when rendering '{{value_json.fanrpm}}'
2023-11-25 21:45:36.174 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'fanrpm' when rendering '{{value_json.fanrpm}}'
2023-11-25 21:45:37.174 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'fanrpm' when rendering '{{value_json.fanrpm}}'
2023-11-25 21:45:38.717 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'fanrpm' when rendering '{{value_json.fanrpm}}'
2023-11-25 21:45:39.192 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'fanrpm' when rendering '{{value_json.fanrpm}}'
2023-11-25 21:45:39.699 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'fanrpm' when rendering '{{value_json.fanrpm}}'
2023-11-25 21:45:40.174 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'fanrpm' when rendering '{{value_json.fanrpm}}'
```

It is probably due to the MQTT discovery template still refers to `fanrpm`. This error repeats every second or so and is very annoying.

revk, 2023-11-25:
The template from the Faikin no longer has it, I assume it is possible to clear from HA, and it pick up the current template.

patrickli, 2023-11-25:
It seems home assistant still remembers it even you have stopped announcing the sensor. Let me try to manually edit the entity registry

patrickli, 2023-11-25:
OK I had to delete the retained messages for fanrpm configs from MQTT. Because they are persistent and not explicitly deleted so Home Assistant keep discovering them.

revk, 2023-11-25:
Ah, makes sense. I wonder if I should set not retained. Not sure. I expect such changes will be rare.

ruimmsimoes, 2023-11-28:
Same problem here with:
Template variable warning: 'dict object' has no attribute 'outside' when rendering '{{value_json.outside}}'
Template variable warning: 'dict object' has no attribute 'comp' when rendering '{{value_json.comp}}'
Template variable warning: 'dict object' has no attribute 'fanrpm' when rendering '{{value_json.fanrpm}}'

When I read the retained message from MQTT (homeassistant/climate/308398D43A94/config) the message I get doesn't mention those entries:
```
{
    "unique_id": "308398D43A94",
    "dev": {
        "ids": [
            "308398D43A94"
        ],
        "name": "faikin1",
        "mdl": "FBQ50C7VEB",
        "sw": "4dfbddd",
        "mf": "RevK",
        "cu": "http://faikin1.local/"
    },
    "icon": "mdi:thermostat",
    "~": "command/faikin1",
    "min_temp": 16,
    "max_temp": 32,
    "temp_unit": "C",
    "temp_cmd_t": "~/temp",
    "temp_stat_t": "308398D43A94",
    "temp_stat_tpl": "{{value_json.target}}",
    "curr_temp_t": "308398D43A94",
    "curr_temp_tpl": "{{value_json.temp}}",
    "mode_cmd_t": "~/mode",
    "mode_stat_t": "308398D43A94",
    "mode_stat_tpl": "{{value_json.mode}}",
    "fan_mode_cmd_t": "~/fan",
    "fan_mode_stat_t": "308398D43A94",
    "fan_mode_stat_tpl": "{{value_json.fan}}"
}
```

Where did you remove it from?

revk, 2023-11-28:
I am not sure of the solution to retained messages like that. We should only specify `outside`, etc. in the config messages if we have them, and then we should include them in the status we send. Restarting mqtt server may clear any messages and make it cleaner.

ruimmsimoes, 2023-11-28:
I finally found how to remove it.
I had to publish a message to: 
**homeassistant/sensor/308398D43A94outside/config** with message {}
**homeassistant/sensor/308398D43A94comp/config** with message {}
**homeassistant/sensor/308398D43A94fanrpm/config** with message {}

revk, 2023-11-28:
Well done.