revk/xESP32-Faikout #155: [FEATURE] Send BLE sensor battery state to Home Assistant

ruimmsimoes, 2023-11-28:
I have a Xiaomi BLE sensor paired up to the Faikin board and now can see the temperature and humidity both in the faikin web page and Home Assistant.
In home assistant I monitor and send low battery alerts on all my battery powered devices.

It would be great if you could also send to Home Assistant the battery state of the BLE sensor.

ruimmsimoes, 2023-11-28:
I have no experience with mqtt config, but from your bletemp config I believe it should be something like this:
```
{
    "unique_id": "308398D43A94blebattery",
    "dev": {
        "ids": [
            "308398D43A94"
        ],
        "name": "faikin1",
        "mdl": "FBQ50C7VEB",
        "sw": "4dfbddd",
        "mf": "RevK",
        "cu": "http://faikin1.local/"
    },
    "icon": "mdi:battery-bluetooth-variant",
    "name": "blebattery",
    "dev_cla": "battery",
    "stat_t": "308398D43A94",
    "unit_of_meas": "%",
    "val_tpl": "{{value_json.blebattery}}"
}
```

revk, 2023-11-28:
OK let me look.

revk, 2023-11-28:
Ok I have issued new code - please test and let me know.

closed 2023-11-28 by revk

ruimmsimoes, 2023-11-28:
It worked.
Thanks.