revk/xESP32-Faikout #161: [BUG] Odd behavoir when using the auto mode with BLE temp sensor in Home Assistant

wurmtal868, 2023-12-05:
**Faikin hardware**
S1 / S3 Mini modules from Amazon

**Daikin hardware**
Daikin Perfera Mulitsplit 

**Describe the bug**
When I try to turn on the fan to stir up the air in my living room while heating with my wood stove, the Faikin module sets cooling mode on instead of turning on the fan.
Same happens, when I turn on heating in Home Assistant while the BLE temp is higher than the auto set temperature. The idea of tuning the heating on while it's warm enough, is that the AC unit should keep a minimum temperature in room when fire burns down.

This is happening despite the lock mode "on" for  heat/cool modes and is not obvious when using the HA GUI.
Home Assistant just shows auto mode although I pushed the button for heat or fan only. 

**To Reproduce**
Set Lock mode "on". Auto mode on with BLE sensor.
Turn the indoor unit off while in heat mode is on.
The Faikin GUI will show heating while the unit is off.
When the BLE temperature  is higher the auto set temperature, turn the unit on.
The mode will be switched to cooling.

**Expected behavior**
When tuning on the indoor unit, I expect the unit to stay in last mode chosen. If it's warmer than the max temperature for heating it should do nothing until the temperature drops below the lower threshold for heating.
If it's cooler than desired in cooling mode it should do nothing.

**Additional context**
I'm not sure how to address the issue when tuning on just the fan in HA. In any case the unit should not switch between heating and cooling with Lock Mode="ON"

revk, 2023-12-05:
When turning on, it sets heat/cool depending on whether hotter or colder, normally. When running it waits before "switching" to minimise switching back and forth all the time. This is intentional behaviour. It may be worth documenting.

There is now a `lockmode` (can be set from settings link on web control page) which stops it switching mode. However, looking at the code, the "initial" heat/cool is not disabled when that it set. I'll make a change for that so you can use `lockmode`.

closed 2023-12-05 by revk

wurmtal868, 2023-12-05:
I don't understand, I think I set the lockmode in the web control page.
[image]

revk, 2023-12-05:
Yes you did, but as I said, the code was not honouring that for the initial start from power off.

I have issued code now that does, so upgrade.

wurmtal868, 2023-12-05:
I updated the FW and it did not switch :-)
However,
instead of waiting for the temperature to drop the unit starts heating:
[image]

wurmtal868, 2023-12-05:
This may be related to a issue I have when I set two indoor units hooked to the same outdoor unit to keep the same temperature. Both are set to 17°C " Wohnen" is fine within 0.5° while "Kochen" is going higher and higher  :-(
[image]

When only one unit is on at the same time everything seems to be fine.

revk, 2023-12-05:
I think I have seen Daikin do that. At the end of the day this is just applying settings to the Daikin. If, when you have it set to heat to lower than current temp it still heats, that is how it works. I'm not sure what other setting we could sensibly apply in such cases.

revk, 2023-12-05:
Err, ok, then don't turn it on?

I have not tried making it cancel what you do generally. It does have an auto on/off control you can use instead. Maybe that will do what you need?

Or are you saying it did auto turn on?

Sorry if I am misunderstanding.

I may have to make that consider lock mode as well?

wurmtal868, 2023-12-05:
A multisplit will heat up all indoor at the same time and the indoor unit decides whether the fan should be on for heating. For some reason the fan turns on although the temperature is already higher than the set temperature. Is possible that the Faikin turns on the fan when it detects that the compressor is on because another indoor unit requested heat?

wurmtal868, 2023-12-05:
Do you have a minute for a call?
https://meet.jit.si/GloriousMethodsUnveilRuthlessly

revk, 2023-12-05:
You can see what it is doing to the settings now.

I am not clear what you want it to do exactly, sorry. This sounds like a work around for something the Daikin does rather than a bug at this point. I'm just going out at the moment.

wurmtal868, 2023-12-05:
Do you have access to an Daikin Multisplit set up?

wurmtal868, 2023-12-05:
I have two three way split units. Using in a hybrid heating system with my wood stove.The third indoor unit on both systems is installed in the basement and usually turned off. My idea is to set abase temperature with the other AC units and heat up rooms individually to a higher value by either increasing the set point for the room or firing up my wood stove. When I just try to maintain a minimum temperature of 17°C like above the two units hooked to same outdoor unit diverge which I don't understand, when both set to exactly the same values. The happened on both outdoor units.

revk, 2023-12-05:
I have not tried to create anything for multiple units to work together at all. The only logic I have is to report when one is not working because it wants to cool when another is heating or the other way.

It is an interesting idea, but may be better for a separate tool, e.g. something on a raspberry Pi, or using home assistant. The Faikins can simply take instructions over MQTT.

It may be possible for the Faikin to do more - we do have settings to control the offsets for "not heating" and "not cooling" which could maybe be used to adjust what it sent to the Daikin in these cases. e.g. telling it to heat to an even cooler temperature, etc. It may be worth trying the various settings.

wurmtal868, 2024-01-19:
I think, I figured out why some of the indoor units keep heating beyond the BLE temperature limit.
In my hybrid system when using my wood stove, the heat in the room does not come from the indoor unit. 
When BLE limit is reached, the Faikin module set the indoor unit to some 2K less than desired BLE temperature. 
However, 
to achieve the new temperature limit the indoor unit will heat up itself to some 3-4K more than this new limit because it tries to compensate for a temperature gradient inside of the room. Especially in my  bedroom the tmeperatur gradient is exactly the opposite from what is expect from the indoor unit, when heating with wood. So 2K delta is not enough to turn off the indoor unit.

revk, 2024-01-19:
There are settings to adjust how much over/under it aims for, see [Manual](https://github.com/revk/ESP32-Faikin/blob/main/Manuals/Advanced.md)

[image]

commit f07ebda0 (2023-12-05): Fix #161