revk/xESP32-Faikout #166: Template variable warning: 'dict object' has no attribute 'fanrpm' when rendering '{{value_json.fanrpm}}'

DRazumovskiy, 2023-12-14:
**Faikin hardware**
Faikin hardware, purchased on Amazon on the 6th of December 2023

**Daikin hardware**
S21 lead

**Describe the bug**
After startup the homeassistant log is full of messages that repeat every 5 seconds:
`WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'fanrpm' when rendering '{{value_json.fanrpm}}'
`
**To Reproduce**
Default settings on Faikin. Mosquitto MQTT broker. After each homeassistant restart the message start to appear in logs

**Expected behavior**
No error messages

**Additional context**
There's a workaround to cancel the message by sending an empty message to "homeassistant/sensor/XXXXXXXXXXXXfanrpm/config" topic. Needs to be repeated each time homeassistant is restarted

DRazumovskiy, 2023-12-14:
Seen similar issue and workaround described in issue #150

revk, 2023-12-14:
`fanrpm` was removed, but no way to remove from retained message in mqtt without restarting mqtt, so HA thinks it still has a `fanrpm` setting.

closed 2023-12-14 by revk

DRazumovskiy, 2023-12-14:
Shall I try to restart MQTT broker then?

revk, 2023-12-14:
Yes, the broker is retaining the message, so that is where the problem is.

DRazumovskiy, 2023-12-16:
Thank you! Restarting broker did not help. I had to stop it, delete the database ../mosquitto.db and restart it. Found no better option to remove retained messages. Great experience otherwise, thank you a lot for doing this project!

njh, 2024-01-21:
Retained MQTT messages can be deleted by sending an empty message with the retained flag set:

```
mosquitto_pub -h 'mqtt-server' -t 'foobar/fanrpm' -m '' -r
```