revk/xESP32-Faikout #169: [BUG] Firmware doesn't work with the latest version of the BLE thermostat sensors

andreas-bulling, 2023-12-18:
I upgraded the thermostat sensors that I've been using with the Faikin module to the latest firmware (v4.5) that supports BTHome v2: https://github.com/pvvx/ATC_MiThermometer/releases/tag/v4.5

Since then, the sensor readings from these sensors are not shown in the Faikin web interface anymore.

revk, 2023-12-18:
I'll look and see if it can support that.

revk, 2023-12-19:
I don't have a set up to test this, but I have added v2 temp/humidity now, so try latest code.

andreas-bulling, 2023-12-19:
Done (version ba6fce4) - still doesn't work unfortunately :/

revk, 2023-12-19:
Just to check, you have set it as unencrypted, yes?

I'll have to set up a test here. I was going from the spec, which should have worked.

But should not be too hard to sort.

revk, 2023-12-19:
I have finally found my spare BLE temp sensors, and flashed, and set to BT Home v2, so starting testing now.

revk, 2023-12-19:
OK, I believe I have it now, please test latest code.

Elwell, 2023-12-19:
Can I test this with a stand-alone faikin board (powered on the bench) as I
have a cluster of BTHome v2 temp/hum sensors within range?
Happy to give it a go if it's of help

revk, 2023-12-19:
You should be able to, yes.

Elwell, 2023-12-19:
Finally installed it properly :-)  - Currently running Faikin-S1-PICO: d5965b6 2023-12-19T08:53:51
and can confirm that sadly it's not seeing either of the 2 nearby BLE sensors that _are_ being picked up by a T-Internet-POE board and sent to Home Assistant via esphome/bluetooth_proxy.

Telink flasher is showing the nearest device as
```
Status: Hardware Version: LYWSD03MMC B1.4, Software Version: 4.6
Vbat: 2811 mV , Temp: 19.28°C, Humi: 57.77%, ID: 156, Reed switch counter: 0, flg: 0x0F:r1/t1
```

revk, 2023-12-19:
Well the good news is that I am an idiot, after testing the update to the library I missed pushing the commit, so the change is not in the Faikin code. Issuing new code right now.

Elwell, 2023-12-19:
That looks better, Ta!

Elwell, 2023-12-19:
That looks better! Ta!
[image]
[image]

revk, 2023-12-19:
Good

closed 2023-12-19 by revk

andreas-bulling, 2023-12-19:
Works also for me. Thanks for the super quick fix!