revk/xESP32-Faikout #171: [FEATURE] Enable PIR where fitted

Elwell, 2023-12-26:
**Daikin hardware**
FTXS71LVMA, Jumper leads, S1 Pico

**Is your feature request related to a problem? Please describe.**
Some in-room units feature a PIR sensor, which to quote the daikin blurb

> Movement sensor saves power consumption in unoccupied rooms: when the room is empty, the unit switches to economy mode after 20 minutes and restarts when a person enters the room.
It'd be nice to make use of thi feature

**Describe the solution you'd like**
Some way of enabling the "intelligent eye" with or without the comfort mode (see manual screenshot)

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Page 15 of https://www.daikin.com.au/sites/default/files/daikin-split-system-L-Series-FTX%28K%29S20-95LVMA.pdf for example
[image]

revk, 2023-12-26:
Do you know of any of the official Daikin WiFi controllers can manage this setting? There are some settings which are simply not available via the S21 interface. I think this may be one of them.

If you can work out the S21 control bit then I'll be happy to add.

gentlecat, 2024-02-06:
I have one of these units as well. Tried going through debugging steps outlined [here](https://github.com/revk/ESP32-Faikin/wiki#debugging---adding-features). Listened to `info/XXX/s21` topic and got the following as I was switching modes on the remote (don't see any obvious differences aside from temperatures):

```
{"G1":"14@A","G2":"=;\u0000\u0080","G3":"0\u00FE\u00FE\u0000","G4":"0\u0000\u0080\u0000","G5":"000\u0080","G8":"0\u0000\u0000\u0000","SH":"042+","SI":"523+","Sa":"550+","SL":"000","Sd":"000","SN":"540+","SX":"591+","SD":"470"}
{"G1":"14@A","G2":"=;\u0000\u0080","G3":"0\u00FE\u00FE\u0000","G4":"0\u0000\u0080\u0000","G5":"000\u0080","G8":"0\u0000\u0000\u0000","SH":"532+","SI":"523+","Sa":"550+","SL":"000","Sd":"000","SN":"810+","SX":"591+","SD":"470"}
{"G1":"14@A","G2":"=;\u0000\u0080","G3":"0\u00FE\u00FE\u0000","G4":"0\u0000\u0080\u0000","G5":"000\u0080","G8":"0\u0000\u0000\u0000","SH":"042+","SI":"033+","Sa":"550+","SL":"000","Sd":"000","SN":"450+","SX":"591+","SD":"470"}
{"G1":"14@A","G2":"=;\u0000\u0080","G3":"0\u00FE\u00FE\u0000","G4":"0\u0000\u0080\u0000","G5":"000\u0080","G8":"0\u0000\u0000\u0000","SH":"532+","SI":"033+","Sa":"550+","SL":"000","Sd":"000","SN":"330+","SX":"591+","SD":"470"}
{"G1":"14@A","G2":"=;\u0000\u0080","G3":"0\u00FE\u00FE\u0000","G4":"0\u0000\u0080\u0000","G5":"000\u0080","G8":"0\u0000\u0000\u0000","SH":"532+","SI":"533+","Sa":"550+","SL":"000","Sd":"000","SN":"810+","SX":"591+","SD":"470"}
{"G1":"14@A","G2":"=;\u0000\u0080","G3":"0\u00FE\u00FE\u0000","G4":"0\u0000\u0080\u0000","G5":"000\u0080","G8":"0\u0000\u0000\u0000","SH":"532+","SI":"043+","Sa":"550+","SL":"000","Sd":"000","SN":"550+","SX":"591+","SD":"470"}
{"G1":"14@A","G2":"=;\u0000\u0080","G3":"0\u00FE\u00FE\u0000","G4":"0\u0000\u0080\u0000","G5":"000\u0080","G8":"0\u0000\u0000\u0000","SH":"532+","SI":"043+","Sa":"550+","SL":"000","Sd":"000","SN":"160+","SX":"591+","SD":"470"}
{"G1":"14@A","G2":"=;\u0000\u0080","G3":"0\u00FE\u00FE\u0000","G4":"0\u0000\u0080\u0000","G5":"000\u0080","G8":"0\u0000\u0000\u0000","SH":"532+","SI":"043+","Sa":"550+","SL":"030","Sd":"000","SN":"160+","SX":"591+","SD":"470"}
```

Let me know if there's anything else I can try in case it helps.

revk, 2024-02-06:
I tried to find a code for that, I found `comfort` but not the other one... Unless we can find the S21 message, we are a tad stuck.

closed 2024-02-06 by revk

KPIotr89, 2025-06-18:
> I tried to find a code for that, I found `comfort` but not the other one... Unless we can find the S21 message, we are a tad stuck.

Hello @revk You made wonderful job in creating Faikin, I don’t want to open new thread. Have you got any good news for message to turn on/off sensor mode? FTXA35C2V1BW is not responding to Sensor command from Faikin WebUI.

revk, 2025-06-19:
We did find sensor in S21, so it will be trying.

The issue is S21 is not documented so it is all guesses, and not all Daikin handle all the messages, so on some units a feature simply does not work via S21.

Obviously if anyone finds alternate S21 messages the I can update the code.