revk/xESP32-Faikout #172: Building issue on Windows

some1sg, 2023-12-27:
I have tried to clone the code via

git clone https://github.com/revk/ESP32-Faikin.git --recursive

when I tried to build the codes in the ESP32 folder 
I get the error

C:/Users/xxx/fakin/main/bleenv.c:1:1: error: expected identifier or '(' before '.' token
    1 | ../ESP32-BLE-Env/main/bleenv.c

this is the same issue with vscode and Eclipse 
ESP-idf v5.1.1 and v5.1.2

Thanks in advance.

revk, 2023-12-27:
Looks like windows may not understand symbolic links, `bleenv.c` is a link.

some1sg, 2024-01-03:
I see, is there any other ways to work around this problem?

revk, 2024-01-03:
You could copy the file there instead of it being a link.

some1sg, 2024-01-09:
Thanks for the suggestion it got me further.
But seems to get stuck in the last part, really appreciate any help

esptool.py v4.7.dev2
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated D:/esp-idf/ESP32/build/bootloader/bootloader.bin
[107/107] cmd.exe /C "cd /D D:\esp-idf\ESP32\build\bootloader\esp-idf\esptool_py && C:\esp\tools\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 D:/esp-idf/ESP32/build/bootloader/bootloader.bin"
Bootloader binary size 0x46f0 bytes. 0x2910 bytes (37%) free.
[1031/1034] Linking CXX executable Faikin.elf
FAILED: Faikin.elf 
cmd.exe /C "cd . && C:\esp\tools\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=D:/esp-idf/ESP32/build/Faikin.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T memory.ld -T sections.ld @CMakeFiles\Faikin.elf.rsp -o Faikin.elf  && cd ."
c:/esp/tools/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: Faikin.elf section `.iram0.text' will not fit in region `iram0_0_seg'
c:/esp/tools/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: IRAM0 segment data does not fit.
c:/esp/tools/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 7632 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

revk, 2024-01-09:
The supplied sdkconfig.defaults should have the settings needed to minimised IRAM usage.

It builds fine under linux and Mac...

If the make file cannot set the target you may have to copy to sdkconfig manually or used idf.py to set the target manually.

That or google how to minimise IRAM and adjust the config, but really, the defaults should work.

closed 2024-01-09 by some1sg