revk/xESP32-Faikout #173: Would this work as BRP072A42 replacement?

Buster14, 2024-01-02:
BRP072A42 is sold out everywhere now, it used to be great solution.
The aircons i'm using are inverters starting with FTKQ
Here's the installation video
https://youtu.be/fSt3s-3MR-I

From my understanding, BRP072A42 contains 2 units,
wifi adaptor and extra pcb, depending on the aircon, you might need the extra PCB,
in my case i need it

revk, 2024-01-02:
Why is this raised as an issue. Thanks for confirming.

closed 2024-01-02 by revk

Buster14, 2024-01-02:
Sorry it's not a feature request so I'm not sure where to post..
so it's basically not possible? the wiki says to use daikin board, which of course also impossible to get,

https://community.openenergymonitor.org/t/hack-my-heat-pump-and-publish-data-onto-emoncms/2551/99

The solution provided here also requires daikin wifi board, so i'm not clear if there's another alternative

revk, 2024-01-02:
You can start a "discussion", or follow up on one.

I don't know if possible or not, but a lot of people have discussed various ways to do this in the discussion threads. I think people have done things with S403 port, but the extra board may be needed for electrical isolation.