revk/xESP32-Faikout #174: [BUG] Not possible to use env **just** as reference temperature

creatvty, 2024-01-02:
First of all, AWESOME device you made there and props for your active engagement in the project @revk 

**Daikin hardware**
FTXM35 

**Describe the bug** 
I would like to use an external reference temperature sensor without Faikin going into auto mode 

**To Reproduce**
So the use case I'm trying to realize is **just** to pass a reference temperature via MQTT (e.g. `env`) to Faikin which then uses this as a reference temperature instead of the internal temperature.

My first attempt was to pass this Payload
```
{
  "env": 20.4,
}
```

This is not accepted by Faikin. Only if I pass additionally a target with `env` the message is accepted by Faikin

```
{
  "env": 20.4,
  "target": [21, 21]
}
```
In this case though, Faikin launches into the "auto" mode without being able to prevent this. "No problem", I thought, "I just pass the target temp of the AC". After that it went into a feedback loop because Faikin would set the target state to like 32° which then would be picked up by the message again and would send MQTT messages like this:

```
{
  "env": 20.4,
  "target": [32, 32]
}
```
A workaround would be to not use the climate entity of Faikin at all and cobble it together from some helper entities. But this is not very satisfactory IMO.

**Expected behavior**
- 1 Just allow to only pass `env` as payload and use this as a reference temperature. Or allow to make the auto mode optional
- 2 When passing the `env` and `target` message,  use `env` for  `current_temperature` instead of internal temperature. And make it optional to **not** send the auto temperature back but the the middle between min and  max target temperature or something

I hope my request is clear enough. Thanks in advance

revk, 2024-01-02:
What is the issue with auto mode?

There is a setting to stop out changing heat/cool in the Faikin auto mode, but you cannot actually override the internal temperature of the air-con, so the Faikin auto mode manages that by setting high or low targets in the aircon to reach what you set in the Faikin auto mode.

Does that make sense?

creatvty, 2024-01-02:
I don't necessarily have an issue with auto mode. I just want to have the ability not to use it when I want to use a remote sensor. Why should that not be possible?

But IMO exposing the temperature that is set inside the AC while in Auto mode is unnecessarily low level. If it is in auto mode I don't need to know that internally Faikin sets the AC target temperature to 32°C. I want to know what the target is that was actually set. i.e. 21°C

revk, 2024-01-02:
Well what do you think the remote sensor value would do?

As I say, we have no way to tell the aircon the remote value, all was can do is set temperature targets. We set high and low to achieve the target, it works.

You can change some settings to as how much higher or lower we set the target, if you wish, they are in the manual, honest...

creatvty, 2024-01-02:
I see. I thought it was a native feature of the AC to allow a an external sensor. Like the one that is actually on the remote that is supplied with my AC at least. Is that not possible?

revk, 2024-01-02:
Sadly not, it would be great if it did (and if someone finds an S21 message to do that I'll code it), but sadly not, sorry.

creatvty, 2024-01-02:
Ah okay. Well here is to hoping it will be found some time. Unfortunately I have no experience in the protocol so I cannot contribute. 

So then my idea to improve usability would be to expose the high level values in MQTT that auto mode is going for. Which would be the `env` value fed back and something between min and max target. I assume there is a value like `current_target_value` or such.

[image]

revk, 2024-01-02:
It is definitely in MQTT but I thought we exposed to home assistant now anyway. I assume you are up to date.

creatvty, 2024-01-02:
I updated yesterday. So should be. Here is my config
```
unique_id: xxx
device:
  identifiers:
    - xxx
  name: Faikin Livingroom
  sw_version: 96061bd
  manufacturer: RevK
  configuration_url: http://Faikin Livingroom.local/
icon: mdi:thermostat
min_temp: 16
max_temp: 32
temperature_unit: C
temperature_command_topic: command/Faikin Livingroom/temp
temperature_state_topic: xxx
temperature_state_template: '{{value_json.target}}'
current_temperature_topic: xxx
current_temperature_template: '{{value_json.temp}}'
mode_command_topic: command/Faikin Livingroom/mode
mode_state_topic: xxx
mode_state_template: '{{value_json.mode}}'
fan_mode_command_topic: command/Faikin Livingroom/fan
fan_mode_state_topic: xxx
fan_mode_state_template: '{{value_json.fan}}'
fan_modes:
  - auto
  - '1'
  - '2'
  - '3'
  - '4'
  - '5'
  - night
swing_mode_command_topic: command/Faikin Livingroom/swing
swing_mode_state_topic: xxx
swing_mode_state_template: '{{value_json.swing}}'
swing_modes:
  - 'off'
  - H
  - V
  - H+V
  - C
preset_mode_command_topic: command/Faikin Livingroom/preset
preset_mode_state_topic: xxx
preset_mode_value_template: '{{value_json.preset}}'
preset_modes:
  - eco
  - boost
  - home
platform: mqtt
```

This is the payload I receive:
[image]

Don't see `env` there. On the other hand I do have the `env` value already since I send it via MQTT and also the target since I'm getting it from the `climate` entity of the AC. I would just like the climate entity to resemble what is actually happening. So that `current_temperature` shows the value of `env` and so on.

jf2020, 2024-01-02:
Hi, I actually have exactly the same issue. The temperature returned by the aircon is way off the actual room temperature and I would like to have the BLE env temperature displayed in Home Assistant in the climate card and use it to control the aircon (so I can set a target of 19C to get 19C and not 21 or 22 like I need to do now. 
The BLE is reported fine in the MQTT message:
[image]

As I'm compiling the software myself, I can modify `Faikin.c` to report the env temp instead in the status message (line 2172), but do I understand correctly from your answer above that this may not control the aircon properly?

Also (and sorry this is an Home Assistant question) @creatvty  how do you display the climate template in yaml?  My climate entities were created automatically and I did not find a way to edit them (I'm relatively new to HA).

revk, 2024-01-02:
I can look at an option to show the BLE as the target in HA maybe when in Faikin auto.

jf2020, 2024-01-02:
I think this is what you are doing already if I'm mot mistaken :-).  My scenario is not using Faikin auto but to have the aircon controlled by the HA thermostat, so the option would be to report the BLE/env temp always.

creatvty, 2024-01-02:
@revk that would be great, thanks! Then it would just miss the part about the target temp. 

@jf2020 Sorry, I'm pretty solid in HA but not entirely sure what you mean. But maybe we could DM somehow

revk, 2024-01-02:
I'll look tomorrow

revk, 2024-01-03:
OK we report the env if set by ble, but not if from MQTT. Next s/w release fixes.

closed 2024-01-03 by revk

JeanOlivier, 2024-01-06:
I made a prototype PID script that uses the ble sensor to control the Daikin for those interested in a tuneable way to effectively use the ble sensor instead of the AC internal temperature:

https://gist.github.com/JeanOlivier/08ebf167e77b14a6b28f1d75bd67bb18

commit 2666aba9 (2024-01-03): Update CN_WIRED Temp
Fix #174