revk/xESP32-Faikout #178: [BUG] AC turns on when it shouldn't

ruimmsimoes, 2024-01-10:
**Faikin hardware**
This is a Faikin board I ordered from amazon in September 2023.

**Daikin hardware**
The board is connected to my Daikin AC via X50

**Describe the bug**
In the past I've tried to configure the Auto mode you developed but I can't seem to understand what each option does or the behavior doesn't match what was expected.

I have the following configuration:
[image]

I also have the board communicating with Home Assistant.

**Expected behavior**
With this configuration I expect the AC only to turn on between 4AM and 6AM, and during that time heat up to 24ºC.

However every now and then the AC turns on, today it already happened two times as can be seen in the Home Assistant:
[image]

On both times I disconnected the AC.

revk, 2024-01-10:
Agreed, with this off, it should not turn on by itself. I'll look.

[image]

revk, 2024-01-10:
The software looks correct, so I am puzzled. There is no way this could be some external control, home assistant? IR remote? etc?

Can you send `setting/Faikin` on MQTT and post the response?

ruimmsimoes, 2024-01-10:
Not sure how to send the command.

The Home Assistant is receiving this messages:
**state/faikin1**
```
{
    "ts": "2024-01-10T16:29:09Z",
    "id": "308398D43A94",
    "up": 18294,
    "mqtt-up": 7495,
    "mem": 29568,
    "spi": 2070220
}
```

**308398D43A94**
```
{
    "online": true,
    "target": 24,
    "temp": 21.71,
    "inlet": 23.11,
    "liquid": 22.23,
    "fanfreq": 0,
    "bletemp": 21.71,
    "blehum": 57.56,
    "blebat": 100,
    "mode": "off",
    "fan": "low"
}
```

**Faikin/faikin1**
```
{
    "protocol": "X50A",
    "ts": "2024-01-10T16:32:00Z",
    "env": [
        21.69,
        21.7,
        21.72
    ],
    "online": true,
    "home": 21.95,
    "heat": true,
    "slave": false,
    "antifreeze": false,
    "flap": false,
    "fanrpm": 0,
    "inlet": [
        23.02,
        23.1,
        23.11
    ],
    "liquid": [
        22.14,
        22.23,
        22.23
    ],
    "power": false,
    "mode": "A",
    "temp": 24,
    "fan": "1"
}
```

Is any of these what you asked for?

revk, 2024-01-10:
OK you need a way to send an MQTT message

ruimmsimoes, 2024-01-10:
I also recieved this message:
**setting/308398D43A94**
```
{
  "rx": -27,
  "tx": -26,
  "reporting": 60,
  "tcontrol": 600,
  "tsample": 900,
  "tpredictt": 120,
  "tpredicts": 30,
  "tmax": 32,
  "tmin": 16,
  "autob": "A4C1383EDE5E",
  "autot": 240,
  "auto1": 400,
  "auto0": 600,
  "push10": 1,
  "switch10": 5,
  "heatback": 6,
  "heatover": 6,
  "coolback": 6,
  "coolover": 6,
  "autop10": 5,
  "thermref": 50,
  "uart": 1,
  "ha": true,
  "ble": true,
  "dark": true,
  "protocol": 4,
  "webcontrol": 2,
  "mqtt": [
    {
      "port": 1883,
      "pass": "<removed>",
      "user": "<removed>",
      "host": "<removed>"
    }
  ],
  "wifi": {
    "pass": "<removed>",
    "ssid": "<removed>",
    "reset": 3600
  },
  "blink": [
    -8,
    -19,
    -7
  ],
  "hostname": "faikin1",
  "appname": "Faikin",
  "watchdogtime": 60,
  "tz": "WET0WEST,M3.5.0/1,M10.5.0",
  "ntphost": "pool.ntp.org",
  "otaauto": 7,
  "otahost": "ota.revk.uk",
  "prefix": {
    "error": "error",
    "info": "info",
    "event": "event",
    "state": "state",
    "setting": "setting",
    "command": "command"
  }
}
```

Is this the message you asked for?

revk, 2024-01-10:
Yes, and oddly not seeing `"autop"` - I'll see if I can work that out.

ruimmsimoes, 2024-01-10:
I've sent the message.

When calling again `setting/Faikin` i get the exact same response as before.

In the response there is a "autop10", is this what you were refering to?

revk, 2024-01-10:
No, that is separate, hmmm

revk, 2024-01-11:
I have an idea, leave it with me.

revk, 2024-01-11:
Nope, not seeing a way this can happen - are you sure this could not be turned on by something else?

ruimmsimoes, 2024-01-11:
In Home Assistant it logs who turns it on via Home Assistant (even if it is an automation) and there is no info that it is being turned on by home assistant.

This AC has a BRC1D52 connected to it, but the only other person in the house says it didn't turn on the AC.

revk, 2024-01-11:
I'll review the code again, but really does not make a lot of sense...

revk, 2024-01-11:
Are you able to log MQTT messages? The Faikin logs if/when it turns on/off deliberately.

openKimchi, 2024-01-27:
This happens to me once the mode has been set to heating, for example, using MQTT. The topic is saved as a state in ioBroker. If the Faikin or the MQTT broker is restarted, the device switches on. I helped myself by publishing the topic for Mode again without a payload 1 second after I set it. This leads to an error in the Topic Error but then the device does not restart.
in ioBroker then it looks like this:
[image]

revk, 2024-01-30:
I think this sis sorted, it is down to sending an MQTT message that turns it on?

closed 2024-01-30 by revk