revk/xESP32-Faikout #184: [FEATURE] Adjust gerber.zip to fulfill your requirements for the logos

Auravendill, 2024-01-15:
**Is your feature request related to a problem? Please describe.**
The documentation states, that if one wants to make their own boards via jlcpcb, that they shall remove the logos, since they are trademarked. But the gerber.zip file according to the Preview in jlcpcb does include said logos.

**Describe the solution you'd like**
It would be great, if the gerber.zip file was in a state, that one just has to upload to jlcpcb. I doubt you would come into my home to look into my air-conditioner just to sue me, but since you asked to remove your logos from it, I would like to comply.

Btw, it seems a bit wasteful to only have one pcb on this big panel. Have you considered layouts with multiple?

I assume, that since you used the 70x70 panel, because "this is the minimum panel for assembly for JLCPCB", that they include all the information to activate "PCB Assembly" on their website? Maybe I'm a bit dumb, but the preview for the assembly looks empty. Is there something wrong?

**Additional context**
[image]
[image]

revk, 2024-01-15:
As I am sure you realise, the main issue is people making these and selling them. We can't allow that with our trademarks, sorry. Given how much of a pain it is to get Amazon to ship to the US, there are people doing that in the US. They removed the logo.

All you do is remove the logo in KiCad layout and save new production files using the JLCPCB plugin. I'm not planning to make two sets of production files every time I make a change, sorry.

Making panels of boards would not help me, these are sold individually in the panel, and the panel helps protect them in the packaging anyway. There is no significant cost saving in making panels of multiple boards as JLCPCB are stupidly cheap for the boards.

Your preview shows the bottom of the board which has no component on it. You need to select assembly on the top of the board where the components are located.

Hope that helps.

Auravendill, 2024-01-15:
> As I am sure you realise, the main issue is people making these and selling them. We can't allow that with our trademarks, sorry. Given how much of a pain it is to get Amazon to ship to the US, there are people doing that in the US. They removed the logo.

So when I make a few for my personal use in my personal home, this wouldn't matter? Or should I still modify the design in KiCad?

> Your preview shows the bottom of the board which has no component on it. You need to select assembly on the top of the board where the components are located.

I uploaded two pictures. The second one is the one for the PCB Assembly. I don't see the components on any of their previews, but maybe they don't show them? I have never used their services, so idk, but I am a bit concerned to get a whole batch of PCBs without anything on them.

revk, 2024-01-16:
The PCB assembly preview does not normally look like that anyway. You load the BOM and position file and it shows an image with all parts, even a 3D view.

Auravendill, 2024-01-16:
I simply uploaded the gerber.zip file, which contains bom.csv and positions.csv among others. Maybe I have done something wrong, but it doesn't seem like there was anything, I could have done differently.

[image]

The 3D view is also empty.

revk, 2024-01-16:
Err, when you select assembly, the next stage shows the bare board, and then it asks you to upload the bom and positions files, then it shows a view with components.

revk, 2024-01-16:
I added the production files to make it easy. If it is going to be a support headache it may be better if I just remove them, and leave it so that only people familiar with KiCad and PCB ordering try and order PCBs themselves.

Auravendill, 2024-01-16:
> I added the production files to make it easy. If it is going to be a support headache it may be better if I just remove them, and leave it so that only people familiar with KiCad and PCB ordering try and order PCBs themselves.

I'm so sorry, I didn't know, that they want them as a separate upload two steps further in the process. I thought, that when I went to "Next", they would just ask for shipping details, payment options and that's it. I'll stop asking further dumb questions, before I make this worse for everybody.

closed 2024-01-30 by revk