revk/xESP32-Faikout #190: [FEATURE] Change fan_modes output text to low, medium, high instead of 1, 2, 3

BenJamesAndo, 2024-01-18:
Currently the fan_modes are listed as
```
fan_modes:
  - auto
  - '1'
  - '2'
  - '3'
  - '4'
  - '5'
  - night
```
In the [2024.1](https://www.home-assistant.io/blog/2024/01/03/release-20241/) release of Home Assistant they added  climate fan mode card feature. This allows you to select the fan mode in an easy drop-down using either the Climate card or Tile card.
[image]
However it seems to only display values that contain text, nor numbers.
[image]

If I go into the additional settings I can still change the fan mode
[image]

Home Assistant has a list of officially supported Names that if used will mean there'll be an icon next to the fan mode. https://developers.home-assistant.io/docs/core/entity/climate/#fan-modes

So if there was a way to re-name the values from 1 to slow etc. that'd be amazing.

revk, 2024-01-18:
Does your device have 5 levels, that is what I think S21 allows, so not sure of names to use as they have low/medium/high, not 5 levels.

BenJamesAndo, 2024-01-18:
Ah yes mine does have 5, I just used 3 as an example. Perhaps the following could work
```
low
medium
medium_high
high
max
```

BenJamesAndo, 2024-01-18:
I have an AC with [SmartIR](https://github.com/smartHomeHub/SmartIR) and it has these options.
[image]

revk, 2024-01-18:
I'll issue a change, just a tad worried in case it upsets other people... Can you test ASAP and confirm it works better with HA than current version.

BenJamesAndo, 2024-01-18:
Hmm kinda strange. Dropdown is blank now.
[image]

night, low & medium doesn't work
low medium goes to 1
medium high goes to 3
[image]

revk, 2024-01-18:
OK changed back issued code 1-5 now.

We need to set you to load test code for this...

`setting/Faikin/outahost excalibur.belmont.cymru`

revk, 2024-01-18:
OK set `otahost` as above, and let me know.

iancoop, 2024-01-18:
I happened to see this, and updated. The dropdown worked ok here.

BenJamesAndo, 2024-01-18:
K I've switched to the test code branch and updated. So yeah turns out if I just re-enable the fan modes then even the numbers show up for me now. I think what I did was when I first set up the Faikin AC I duplicated the UI card and that would have inherited different fan modes, hence why the 1-5 didn't show up. So turns out I was wrong and Home Assistant can in fact show numbers as fan speeds in a dropdown.
[image]

revk, 2024-01-18:
OK I'll make a change and you can try, first, what build type is it (bottom of settings page on web interface)

BenJamesAndo, 2024-01-18:
Faikin-S3-MINI-N4-R2: 41d6a4b 2024-01-18T10:33:28

revk, 2024-01-18:
I could not see `FAN_LOW` in the link you provided, happy to try those, where did you find them?

revk, 2024-01-18:
I have made a build with `FAN_LOW` etc, they may not work to make changes yet, but can you check HA is happy with them?

BenJamesAndo, 2024-01-18:
I saw `FAN_LOW` at https://developers.home-assistant.io/docs/core/entity/climate/#fan-modes

Just updated and this is what I see now
[image]

BenJamesAndo, 2024-01-18:
Switching fan modes doesn't work and this is the error log in Home Assistant
2024-01-18 21:31:17.223 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid fan_modes mode: 1
2024-01-18 21:31:17.224 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'target' when rendering '{{value_json.target}}'

Seems like it still thinks it's 1 instead of FAN_LOW

revk, 2024-01-18:
OK let me try and join the dots here...

revk, 2024-01-18:
OK try and upgrade now

BenJamesAndo, 2024-01-18:
K so it still displays as FAN_LOW as before but when switching it does in fact change the fan mode.

revk, 2024-01-18:
Are you able to see the status we are sending back? It should use the exact names we set?

revk, 2024-01-18:
I wonder if it should not have `FAN_` on the front?

BenJamesAndo, 2024-01-18:
Yeah maybe it shouldn't have `FAN_`. Looking through a .json file for SmartIR config and this is how it declares it
```
   "fanModes":[
      "auto",
      "night",
      "low",
      "lowMedium",
      "medium",
      "mediumHigh",
      "high",
      "powerful"
   ],
```

revk, 2024-01-18:
I have `powerful` as a separate setting (I think it reverts to previous fan, so is not a fan speed itself, but an override).

Anyway, changed to those, try now.

BenJamesAndo, 2024-01-18:
It works! Thanks so much!
[image]

revk, 2024-01-18:
OK can you confirm changes to all options work, and if so I'll issue new code on `ota.revk.uk`

BenJamesAndo, 2024-01-18:
Yup changes to all options work.

revk, 2024-01-18:
OK new code in 5 mins, change `otahost` back to `ota.revk.uk`.

closed 2024-01-18 by revk