revk/xESP32-Faikout #191: minimum temperature

chrisguen, 2024-01-20:
Is it possible to set the minimum temperature to something like 8° for frost protection? I don't have a daikin ac yet so I'm wondering if it is possible with this module as the normal remote just goes down to 15° I think.

revk, 2024-01-20:
It can only do what the Daikin can do. There are different models with different range of settings for different climates as I understand it. The Faikin can be told what min/max your Daikin can handle. But if the remote cannot do it then the Faikin is unlikely to be able to.

MUN0X, 2024-01-20:
I believe the allowed min and max are different for heating and cooling mode. 
Could that be implemented?

revk, 2024-01-20:
You can set the range it is happy to try but it will only work when daikin accepts it.

closed 2024-01-30 by revk