revk/xESP32-Faikout #196: [FEATURE] Configure max fan setting for Faikin-auto mode

HV-NL, 2024-01-23:
**Daikin hardware**
FTXM42N (S21), Fan 1 - 5, auto, night

**Is your feature request related to a problem? Please describe.**
When using the Faikin-auto mode the automation also modifies the Fan setting, so for FTXM42N values 1 to 5.

5' is very loud. I would like to be able to reduce the max fan value for Faikin-auto mode to a lower value

**Describe the solution you'd like**
Please implement an mqtt max setting for the Fan in Faikin-auto mode (default:5)

revk, 2024-01-23:
Interesting. There are two places we adjust fan, one is when starting and way outside target, and the other is where we adjust by steps to try and get back in to range. The latter should only happen after several previous steps and so only when really necessary.

So is it the initial "fan at full until back in range" that we do, or limit always?

closed 2024-01-23 by revk

HV-NL, 2024-01-23:
I would like to reduce it always (but 2 separate mqtt settings is fine for me)

revk, 2024-01-23:
I have gone for one setting for now, let me know how it goes.