revk/xESP32-Faikout #206: General Troubleshooting

smnino, 2024-01-27:
I am having a few issues that I hope someone can assist me with.

I am no electronics major but I have a very basic understanding. I am working with a Daikin FTX18BXVJU head unit.  I purchased my Faikin from Amazon UK although I am located in the US.

I can get the board to boot via usb serial adapter by connecting the usb adapter's VCC pin and GND to the Faikin's 12V and GND. The Faikin boots up fine and I can access the web UI. The VCC pin of the adapter is providing 4.9 volts to the Faikin. The Faikin only boots when voltage is connected to the 12V pin on the board.

My AC has an S21 connector. When I measure the voltage of the AC's pins, I get 4.9 on pin 5 and 9.4 on pin 4. When I connect the GND on the AC to GND on the faikin board and pin 5 on the AC to the  12V pin on the faikin board, the AC's control board starts making a slight buzzing noise and then the pizo buzzer beeps as if the AC's board is rebooting? At no point does the faikin boot. The faikin's LED does not light up at all. I am afraid to continue attempting this as I do not want damage the AC's control board.  I literally just had this AC installed less than 2 weeks ago.

I am including a few images of the AC's control board as well as the Faikin. I have updated the Faikin to the latest OTA firmware.

[image]
[image]
[image]
[image]

revk, 2024-01-27:
The S21 is connected one to one.

The power for the failing is on the "12V" (which may be 14V even). It actually works with anything from 5V to 35V.

The 5V connection (pin 1) is for a pull up. It does not power the Faikin board. If you try to power the Faikin from this it is likely to not have enough current (hence upsetting the aircon if you do that).

Elwell, 2024-01-27:
As the daikin boards aren't terribly clearly marked, it might be worth
having a diagram/ photo in the readme indicating which end of the s21 is
pin1. I was nervous of hooking mine the wrong way round (and couldn't
trivially find a ground reference to check pinout against). Can confirm
that jumper leads fit nicely, but I checked the shape of the connector vs
references multiple times.

revk, 2024-01-27:
I could add a 1. But as the plugs only fit one way around and it is one to one wired this has not really come up before. I agree that it is very unconventional of Daikin not having GND on pin 1 though. I'll see if the manual can be a tad clearer.

closed 2024-01-27 by revk

Elwell, 2024-01-27:
Yeah, your board is fine. It's the aircon end that wasn't clear (from the top side - it may well be marked underneath but I'm not pulling it out). Knowing it's a 1:1 maping meant I was holding yours beside it when running cables

revk, 2024-01-27:
Yeh, if you have the right plugs, it should be simple to make a one to one cable. If using jumper wires, I can see it may be harder. The board does have a "1" on it, I see.

smnino, 2024-01-27:
So, I made a 1:1 cable and I get no response from the faikin or the AC. I checked the continuity of the cable and the cable is good on all connections.

The faikin does nothing...

[image]
[image]
[image]

smnino, 2024-01-27:
> Not an expert, but I would double check your measurements of the board. The 5V pin shouldn't be next to the 12V pin on the S21 connection. You are attempting to connect here, correct?
> 
> ![image](https://private-user-images.githubusercontent.com/10055220/300152952-92b262dc-5d92-440a-9e8f-e2b6084796b9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYzOTY4MTEsIm5iZiI6MTcwNjM5NjUxMSwicGF0aCI6Ii8xMDA1NTIyMC8zMDAxNTI5NTItOTJiMjYyZGMtNWQ5Mi00NDBhLTllOGYtZTJiNjA4NDc5NmI5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI3VDIzMDE1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY4MDlkNmYwNGQ0OWI2OTc1NTVmMTFiMWFjMmM5MjUyZTk0NzlhMzk4MTI3NjQ3YmY4NWI5YTVjMjZjY2U3MDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.idstg6vfdXfIOo0A26trswrQGogH6K1QsdnFxdCtT9U)

Yes, this is where I am connecting as this is labeled on the board as S21.

revk, 2024-01-28:
Can you check voltages on back of Faikin relative to GND (brown, pin 5)?

smnino, 2024-01-28:
I have gotten it to work. Seems there was something amiss with the cable I had made. When I removed the cable and used header cables, I was able to get it working.  An option to purchase an S21 cable along with the faikin would have been nice. I certainly would have paid extra for one.... 

Thank you for making this available to me & the masses.

My shipment from the UK came pretty quickly, maybe a week or 10 days and I didn't need a forwarder or have to pay anything extra. That said, if there is anything I can do to assist distribution of the fakin in the US, please let me know. The Daikins are starting to become popular here in the states and the faikin should be included with every unit in my opinion.

One more thing; Is there a way to change the display from celsius to fahrenheit?

revk, 2024-01-28:
If I could find a sensible source of S21 cables I would sell, or at least recommend/link. It does seem the most common configuration.

As for Fahrenheit - I was half expecting this one day. The underlying protocol is all Celsius, which makes it tricky.

I suspect Home Assistant can cope.

One option would be on the web interface to output the temperatures and even the settings in F along side the C, so you actually set in C but can see what that means in F. Would that help?

smnino, 2024-01-28:
Yes, seeing the conversion from C to F would be helpful. Although, it may not be worth the effort as I have it integrated into home assistant now and as you suggested, HA converts it fine.

There's very little about the faikin to be found. I did find a thread [here](https://community.home-assistant.io/t/need-daikin-wifi-use-the-open-source-faikin-esp32-hardware-instead-of-the-official-wifi-modules/644370/8) and posted the following questions. (C to F you've already answered)

- Is there a way to change the UI display from celsius to fahrenheit
- how does one set the timezone? The linked list doesn’t look anything like the default values
- I don’t understand a few of the UI icons: Green Dot/Blob, emoji with hands above head, left/right arrow, teddy bear
- what are the temps listed? They are labeled ‘Home’ & ‘Outside.’ Does the unit have an outdoor thermostat as well as one in the head unit?

revk, 2024-01-28:
The buttons all have mouse over tool tips (but not easy on a mobile).

The timezones should match the linked page, what time zone are you in? I'll look.

The temps come from the air-con, and yes there is an outside temperature sensor. Home is the inside one built in. Others relate to coolant temperature, etc.

smnino, 2024-01-28:
I did not mean to imply 'it was just you' I just didn't understand what they meant, like the teddy bear?

As for the TZ, I'm in the New York, eastern TZ.

revk, 2024-01-28:
Sorry.. fair enough. That is, I believe "comfort" mode, whatever that does.

I'll look up timezone when I get up. It is 2:30am in my timezone :-)

revk, 2024-01-30:
Ok following that link, under `America` I see

`{"name":"New_York", "tz":"EST5EDT,M3.2.0,M11.1.0"},`

So set `tz` to `EST5EDT,M3.2.0,M11.1.0`

gentlecat, 2024-02-06:
Just to add another datapoint, I ran into the same issue with buzzing sound after plugging in Faikin into S21 using jumper wires. Missed the fact that connector is supposed to be one to one. The sound seems to be happening due to backwards connection. Luckily that didn't break anything. Faikin started working fine after I flipped the wires.

[image]

[image] [image]

Wasn't sure if my models (CTXS07WVJU9, FTXS09WVJU9, FTXS15WVJU9) will have the right connector since they were not listed in the wiki, but it turned out they do. I added them to the list already.

Thanks for making this awesome alternative! Shipping from Amazon UK to Canada was pretty quick.