revk/xESP32-Faikout #208: Got Faikin working on FTX12NMVJUA with direct connection on S403

Jerther, 2024-01-28:
Not really an issue or a feature request, I just wanted to put out there that I've got Faikin working on my head unit! 

**Daikin hardware**
FTX12NMVJUA
Direct connection on S403 using:

```
Faikin - S403
GND    - GND pin 6 or 7
12v   - +14v pin 1 (BOTTOM)
48(TX) - RX pin 5
34(RX) - TX pin 4
5v     - Not Used
```

This has already been said but, to anyone who would like to do the same: WATCHOUT FOR **PIN 10 (TOP) WHICH HAS 320V ON IT**! DON'T TOUCH IT OR CONNECT AYTHING TO IT! And abviously, don't wire anything while the unit is powered!

I've then put the Faikin module in the small compartment at the bottom that is not covered by the metal shield.

Works a treat! WAY better than my attempt at an IR module that mimics the IR remote control ;)

[image]

Sytten, 2024-01-28:
Nice I have a FTX12NMVJU (not A?) and I was just wondering if it was compatible. Ideally I would to make a cable with a plug, is it a standard one?

Jerther, 2024-01-28:
I'd like to help you with the cable but as you can see I hacked something. The pins on the Faikin are standard pitch so I just used a strip of dupont connector, but on the other end I used some old motherboard connectors for front panel leds and buttons. I removed the metal connectors from the plastic housing and crimpted them a little bit. It makes a surprisingly secure connection.

Best of luck :)

closed 2024-01-30 by revk

Sytten, 2024-02-05:
I got a better view of of the board. I am still looking for a compatible connector. [image]

eskey0, 2024-03-30:
@Sytten I just setted up my faikins, I was successful with "2.5mm Pitch 6-pin Cable Matching Pair - JST XH Compatible" I had around, It's a thight fit, it even bend the plastic a little bit, but Iam 100% satisfied with the result.

A picture of the finished product 
[image]

hAMMERlized, 2024-04-17:
> @Sytten I just setted up my faikins, I was successful with "2.5mm Pitch 6-pin Cable Matching Pair - JST XH Compatible" I had around, It's a thight fit, it even bend the plastic a little bit, but Iam 100% satisfied with the result.
> 
> 
> 
> A picture of the finished product 
> 
> [image]
> 
> 

Nice one. But I don't get your wiring. Why do you have 6 wires instead of 5 at the connector? And where is the white and black cable going to? As well as the green one. As I can see there is a light and dark green one?

eskey0, 2024-04-17:
> > @Sytten I just setted up my faikins, I was successful with "2.5mm Pitch 6-pin Cable Matching Pair - JST XH Compatible" I had around, It's a thight fit, it even bend the plastic a little bit, but Iam 100% satisfied with the result.
> > A picture of the finished product
> > ![Conector](https://private-user-images.githubusercontent.com/139238654/318216672-e2aaf1e7-0fbd-4fae-b8f1-aee9e6cf4062.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzMzM5NDEsIm5iZiI6MTcxMzMzMzY0MSwicGF0aCI6Ii8xMzkyMzg2NTQvMzE4MjE2NjcyLWUyYWFmMWU3LTBmYmQtNGZhZS1iOGYxLWFlZTllNmNmNDA2Mi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxN1QwNjAwNDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iZGNkZGVmYTVhMGIyYTA1ZTcxMGUwM2EzZjQxOTUyZWMyYjcyMjZjM2ZhNTY1OTgzMDU2ZjQ3MGI1NTY1ZDU5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.3PxNqfvC0QhlLRHp13q2WEE4PhNru_SVcvfDBGkiUPU)
> 
> @eskey0
> 
> Nice one. But I don't get your wiring. Why do you have 6 wires instead of 5 at the connector? And where is the white and black cable going to? As well as the green one. As I can see there is a light and dark green one?

Thanks! The color of the wires don't match, I had both the connectors with their pigtails (I don't have the tools to make any of them) so they are soldered behind the faikin. The connector has 6 pins because you have to reach pin 6th for the ground, the unused ones are not connected, just routed behind the faikin and hidden there.

I hope I managed to explain myself, if you want I can dissasembly the unit and make some more pictures.

hAMMERlized, 2024-04-17:
@eskey0 Thanks for the quick response. I think I got what you mean but I guess some detailed pictures can't be wrong 😉. And maybe some kind of quick written "diagram" like @Jerther did at the beginning of the topic. Just to be sure which pin is used and connected to the faikin board. Thanks.

fischan, 2024-04-17:
@eskey0 which Indoor Unit do you have is this an ATXD Siesta Unit?
Looks very similar to mine ATXD35A5V1B.
Thanks!

eskey0, 2024-04-17:
> @eskey0 which Indoor Unit do you have is this an ATXD Siesta Unit? Looks very similar to mine ATXD35A5V1B. Thanks!

They are comfora, the exact model is FTXP35N5V1B, but maybe they share the board.

hAMMERlized, 2024-04-17:
Hey @eskey0 ,

how about the detailed pictures and / or wiring diagram that you wanted to upload? :) 

Thanks

eskey0, 2024-04-19:
> Hey @eskey0 ,
> 
> how about the detailed pictures and / or wiring diagram that you wanted to upload? :)
> 
> Thanks

I have to dissasembly the unit, I will upload as soon as I can, but I got my pinout from Jerther, in the first post you have all the info.

hAMMERlized, 2024-04-23:
Ok, so I have managed it to get it work with the [Lyeteung 10 Paar JST 2.54 XH 6... https://www.amazon.de/dp/B0CBWY23FN?ref=ppx_pop_mob_ap_share](following cable). Be aware to cut the 2 plastic "noses" with a sharp knife. Then the cable fits pretty well. 

For connecting the 6 Pin Cable with the Jumper Cables of the Faikin module I have used [BAURIX® Lötverbinder Set [120... https://www.amazon.de/dp/B09MJ7W41L?ref=ppx_pop_mob_ap_share](this). So no soldering and crimping is needed.

Finally some hot glue to keep the cables where I want them to be and that's it. I have attached some detailed pictures. I have used Pin 1,4,5 and 6 from the connector. 

Best Regards

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Jerther, 2024-05-29:
Any of you who got the Faikin module working on S403 have random WiFi connectivity issues? I have the module working since my original post, but once in a while the wifi would dicsonnect and reconnect after many minutes. I wonder if it may be caused by the Vcc line on the S403 port not providing enough power?

fischan, 2024-05-30:
> Any of you who got the Faikin module working on S403 have random WiFi connectivity issues? I have the module working since my original post, but once in a while the wifi would dicsonnect and [...]

I remember two times where I could not reach the web UI, this happened always after changing settings, but not sure if related to power supply. Also did not check if Faikin was still connected to Wifi.

Next time I will check.

adude00, 2024-06-08:
@hAMMERlized do I understand correctly that you're using the S403 connector that was originally meant for [wired remote control](http://www.daikinac.com/content/assets/Controls/KRP980B1-EDUS041011.pdf)?

I'm asking because [in mine](https://i.imgur.com/5tY6ruC.png) the cloud only wifi BRP069C47 [connects to the S430 connector](https://i.imgur.com/veDV51t.png).

hAMMERlized, 2024-06-08:
> @hAMMERlized do I understand correctly that you're using the S403 connector that was originally meant for [wired remote [...]

Yes

adude00, 2024-06-17:
Following @hAMMERlized example I made an S403-Faikin cable, thanks to [a set of pre-crimped jst connectors](https://www.amazon.it/dp/B0CDGC1ZQS) I found on Amazon.

The split is a Siesta series, model ATXF25E (with ARXF25E outside unit, but it shouldn't matter).

The cable:
[image]

Faikin side:
[image]

S403 (Daikin) side:
[image]

Note that I did trim both connectors as per @hAMMERlized example above.

Connected to Faikin:
[image]
[image]

Faikin details:
[image]
[image]

Before (original WiFi module only):
[image]

Test run:
[image]

Tucked in:
[image]

Shield installed:
[image]

Note that I did NOT touch the original BRP069C47 WiFi module. This is a feature for me so Alexa integration does not rely on my home automation (also, I paid for the damn thing so I might as well use it).

dodgeboy, 2024-07-19:
I have the same Daikin model and i just got mine connected via the S403 yesterday.  I'm having trouble controlling the fan speed.  It auto detected 5 fan speeds + night mode.  However, night mode does not work.  Are you able to use night mode with your Faikin?

revk, 2024-07-19:
Night mode should work, but may not report back as night mode. This is a basic issue with the S21 messaging. There are a few options to change how it is handled.

dodgeboy, 2024-07-19:
In my case, when I click night mode it does not slow down the fan and after 5-10 seconds the highlighted button switches back to the fan speed it was at.  In advanced settings, if I enable the option to always use night in auto mode, it doesn’t slow the fan down either.  So I believe it’s something about my unit that doesn’t understand the command for night mode.  

I should leave the fan steps setting at 0, right?  I tried setting it to 3, 5, 6 and 9 as an experiment and it only made things worse by just giving me low, med, high speed buttons. 

I didn’t mean for this to turn into a project for you.  I mainly just wanted to see if others with the same unit were having the same problem and determine if maybe I missed a setting or something.  I appreciate your response!

revk, 2024-07-19:
Ok maybe it does not so night mode? Strange.

There are options to debug the messaging and even send messages, using MQTT.

If you work out how to do it, then I can update code.

dodgeboy, 2024-07-19:
Great!  I’ll work on that and report back with my findings.  Thanks again!

fortellerq, 2024-08-04:
Here's a pic of my aircon unit: Daikin Sensira FTXF35E5V1B. Very basic unit as it does not have a lot of features, but Faikin gets the job done nicely. Thanks a lot for people who provided their feedback in this thread - I've bought and used the same connector as people above.
[image]

Sytten, 2024-08-04:
Its better to add images directly in github @fortellerq, third party image hoster always break after a few years
[image]

fortellerq, 2024-08-05:
Thank you @Sytten for uploading the image for me. I had some trouble uploading it to GitHub and that's why I uploaded it to 3rd party host.

hedgepigdaniel, 2024-08-09:
Just to add my experience to the mix, I've set up Faikins on CTXM25RVMA, CTXM35RVMA, CTXM60RVMA. These are all part of the Australian [Super Multi NX](https://www.daikin.com.au/sites/default/files/brochures/files/PCRAU2208A-Daikin-Super-Multi-NX-Brochure_LR.pdf) line of multi split Cora indoor units.

|Units|Experience|Pictures|
|--|--|--|
|CT(K/X)M(20/25/35/42)RVMA| According to the brochure, these all accept a BRP072C42 Wifi adapter via a BRP067A42 "Remote control PC board set", so I assume they are all equivalent. My CTXM25RVMA and CTXM35RVMA both have an S403 header accepting a XAP-10V-1 plug.|[image] [image]|
|CT(K/X)M(50/60/71)RVMA| According to the brochure, these all accept the same BRP072C42 Wifi adapter but via a different BRP980B42 "Remote control PC board set". Again, I assume these are all equivalent. My CTXM60RVMA has an S403 connector that looks like the one in [this post](https://github.com/revk/ESP32-Faikin/issues/317#issuecomment-2155758144). I also used a JST PAP-10V-S and trimmed off the retainer and clips. The pins are connected in the same order as the XAP-10V-1 connector in the smaller units.|[image] [image] [image] [image]|
|CT(K/X)M(85/95)RVMA|According to the brochure, these accept a BRP072C42 with no adapter. My best guess is that they have an S21 connector.|

In all cases, I wrapped the Faikin in some heat shrink and left it resting inside the steel shielding around the high voltage circuits, because it doesn't seem like the S403 is safely isolated from high voltage. None of my units are very far from the WiFi router, but they work perfectly - no dropouts or anything.

deadhurricane, 2024-08-20:
@hedgepigdaniel I'm going to be attempting this shortly on the 2x Single Cora units, it looks like the internal units should be identical. I'll report back once completed. 

2 questions for you:
- How did you source the Faikin board? Amazon direct to Australia, Amazon forwarding via UK or PCB Manufacture?
- Do you have a link to the crimps you used on both sides? I can see JST-XH vs JST-XA in previous comments

hedgepigdaniel, 2024-08-20:
I bought the Faikins from Amazon which (at the time) was shipping them to Australia.

I bought pre-crimped cables from Digikey. You can find them in the "Associated Product" section, or by referring to the datasheets.

ASXASXA22K203 for the XAP-10V and APAPA22K254 for the PAP-10V-S.

AndrewMBL, 2025-01-19:
I wanted to add my 2-bits I've replicated the install on my CTXM60RVMA & CTXM20RVMA for completeness I also have the 4MXM80R2VMA outdoor unit and everything is working great.

| Units | Experience | Pictures |
|--------|--------|--------|
| CTXM20RVMA | I used a JST connector for the Faikin and then stripped the pins for the s403 and covered them in heat shrink or tape which ever was on hand. I'll note again that there is dangerous voltage present DO NOT WORK LIVE! By regulation your AC has to have a shutoff either in the breaker board of next to the outdoor unit. Use it. | [image] [image] |
| CTXM60RVMA | Same as above | [image] [image] |

Michael-Yongshi, 2025-01-31:
@adude00 Thanks for your wiring setup! Yours seemed the simplest I found so far (noob here) so decided to try it.

Ordered the same for my FTXP-N and it worked blissfully (after ordering the correct cables, looking them up on the dutch amazon I accidently ordered the 2.0 instead of the 2.54...)

[image]

The side of the Faikin doesnt need cutting, but it doesnt seem to fit 'quite right' (no probs with it though; bit of brute force did the trick)

An example how to cut the plastic on the JST that goes in the Daikin would be nice though for a noob like me, so I put a picture below. I cut the 2 indentations to make it perfectly square and then it fitted perfectly there.

[image]

GobblerGuy, 2025-02-03:
I am in the process of purchase of CTXM25RVMA and CTXM50RVMA. One issue I expect to have is that the indoor units may need to be installed hard against the wall in a corner so that I cannot easily access the right hand side of the unit. If I wish to install a Faikin I suspect this will cause issues. Is this correct? Is there any advice, other than trying to avoid the right hand side of the unit hard against the wall, that I could advice the installation team that will make it easier for me to retro fit a Faikin later?

Jerther, 2025-02-03:
@GobblerGuy you could run some wires from the communication port to the front. Make them long enough so you can determine a proper location for the faikin module later. Also be careful not to run a wire for the 320 pin!

Michael-Yongshi, 2025-02-03:
@GobblerGuy 
Or make sure you install it while they are doing preperation work and havent installed the unit itself yet.

I guess once it is fully installed i would not fiddle with the main unit anymore.

You know its the 403 port for sure?

Jerther, 2025-02-03:
> I would actually go with @Jerther 's option in case the faikin fails in the future so you can replace it...

And mine did a few weeks ago. I had to pull it out and reflash it.

GobblerGuy, 2025-02-04:
I'm not sure I can get a Faikin before the install date (everything is slow to get to Australia). I'm not sure the installer would agree to install a 3rd party anyway. I think it is a 403 port per @AndrewMBL post above for CTXM20RVMA. I do wonder if I could get the wiring in place - I assume I could even ask the installer to use the same wiring lead as for the official wifi unit? need to check unless anyone here knows from their own experience. 

@Michael-Yongshi  is 10cm off the wall enough to allow access? I think I could manage that - I'd not considered the possiblity the faikin could fail (wow seems obvious when I think about it)!

deadhurricane, 2025-02-04:
@GobblerGuy 10cm should be sufficient. 

It is always good practice to have some clearance around the unit (according to the installation manual recommendations) to allow for servicing of the unit. The white cover pops off through front screws and a bump to the top, and then you just need to undo the cover for the EMI shield on the side for Faikin installation. Installer most likely won't let you touch the system during their install process, just put it in afterwards.

You probably can't get the wiring in place beforehand because you need to buy the official wifi kit to get the wiring loom, which was around $200 last I checked. 6 or 7 pin JST-XH with the tabs cut off as people have mentioned above will do the trick for port S403, got mine for around $5 from aliexpress https://www.aliexpress.com/item/1005007289626003.html

AndrewMBL, 2025-02-04:
@GobblerGuy It will be tight to install with only 10cm but as @deadhurricane said should be do-able just make sure it's all fully powered down.

Michael-Yongshi, 2025-02-06:
Make sure you can remove the metal shielding within this 10 cm. When you remove it you have to remove it in a bit of an arc, although i think not much

GobblerGuy, 2025-02-17:
https://github.com/revk/ESP32-Faikin/issues/208#issuecomment-2297981818

@hedgepigdaniel 

> I bought the Faikins from Amazon which (at the time) was shipping them to Australia.
> 
> I bought pre-crimped cables from Digikey. You can find them in the "Associated Product" section, or by referring to the datasheets.
> 
> ASXASXA22K203 for the XAP-10V and APAPA22K254 for the PAP-10V-S.

How did you go with the S21 connector at the other end? Looking at the EHR-5 on Digikey which I understand is the connector for the s21 it does not list either of the two pre-crimped cables above in the 'associated products' section. Are  ASXASXA22K203 and APAPA22K254 compatible with the EHR-5?

other option is to go with the wiring loom as noted above by @deadhurricane but that ends in bare wires at the other end and I don't have a crimping tool. I'd prefer not to buy one for such a small project.

GobblerGuy, 2025-05-09:
Successfully installed Faikin in CTXM50RVMA. All the standard controls work, but extra controls such as comfort, powerful, economy and quiet cannot be operated via the Faikin.

leowinterde, 2025-08-15:
Simply works with a 2.54mm JST-XH 6 on my Daikin ATXD25A/ARXD25A ❤️ 

[image]
[image]