revk/xESP32-Faikout #209: Fan Control [BUG]

Evilghost12, 2024-01-28:
**Faikin hardware**
Buyed at Amazon 19.01.2024

**Daikin hardware**
on Port S21 - Perfera Buyed at Oct. 2023

**Describe the bug**
Whe the MQTT Aircon control Command  fan A, Q or 1 to 5 was send - The Mode was also changed to Fan.!!

Evilghost12, 2024-01-28:
https://github.com/revk/ESP32-Faikin/assets/75432774/1a054788-fa64-42ed-bb5c-3edc029faa31

revk, 2024-01-28:
Wow, OK, will look in to it.

Evilghost12, 2024-01-28:
Could you also take a look at the "bug" - when you change the mode, the Power on status is also sent. So the indoor device is power off and I change the mode to cooling - the power status changes with it and the device is power ON

https://github.com/revk/ESP32-Faikin/assets/75432774/6cf13652-10ca-4c5d-86ea-6654fcb8229e

Evilghost12, 2024-01-28:
These two "bugs" only happen with MQTT - everything works fine on the web interface

revk, 2024-01-28:
This is very odd indeed, can you provide the exact mqtt message that are being sent with topic and payload?

Evilghost12, 2024-01-28:
Okey.. I found the Problem - stateChange mqtt.1.command.technikraum.fan: null  - was changed to FAN Mode. 

mqtt.1 | 2024-01-28 12:11:45.056 | debug | onStateChange mqtt.1.command.technikraum.fan: {"val":"A","ack":false,"ts":1706440305056,"q":0,"from":"system.adapter.admin.0","user":"system.user.djneahrsi","lc":1706440305056}
---
stateChange mqtt.1.command.technikraum.fan: {"val":"","ack":false,"ts":1706440314164,"q":0,"from":"system.adapter.admin.0","user":"system.user.djneahrsi","lc":1706440314164}
--

This is to clean up the status with no "val":"" - to prevent a power on status from being published again when the MQTT client reconnects - or a mode

Solution Idea -> not response to an empty "val":"" 

And on Mode:

2024-01-28 12:21:37.547 | debug | stateChange mqtt.1.command.technikraum.mode: {"val":"C","ack":false,"ts":1706440897546,"q":0,"from":"system.adapter.admin.0","user":"system.user.djneahrsi","lc":1706440897546}

Here is the problem with or without "null" -> empty "val":"" - Power turns on with the Command

revk, 2024-01-28:
I don't quite follow your logs, but there are two command it handles for, `fan` for example.

- `command/Faikin/fan` sets mode to fan.
- `command/Faikin/fan 1` sets fan speed to `1` without changing mode.

If you are sending the `fan` command with no payload you are changing it to fan mode.

Don't do that!

Evilghost12, 2024-01-28:
Yes, that was the problem - exactly. But with Mode it doesn't matter which command I send - whether the value is there or not - Power always goes to true

revk, 2024-01-28:
OK, that is based on the way HA works, it sees `off` as a `mode` so any `mode` other than `off` turns on.

You can send `command/Faikin/heat`, etc...

Evilghost12, 2024-01-28:
Okey - yes, there are the small differences - that's why I meant bug under quotation marks - I can deal with that and the fan issue is solved - THANK YOU for your patience.

closed 2024-01-28 by revk