revk/xESP32-Faikout #213: [BUG] Password fields exposed

ruimmsimoes, 2024-02-01:
**Faikin hardware**
(Not relevant, software issue)

**Daikin hardware**
(Not relevant, software issue)

**Describe the bug**
The config webpage of the faikin exposes the passwords in the password fields.

**To Reproduce**
When accessing via any modern browser if you inspect the password field you can easily find out the values of the password field(s).

**Expected behavior**
Either protect the webpage with a user/ password, or replace the password in the webpage with some other tag like ‘password’ or ‘********’ and only if the user posts anything different from the tag is the stored password changed.
Ideally it would be best to protect the web interface with a user/ password which would also prevent unauthorized control of the AC unit, but this probably raises the need to have a mechanism to restore factory settings if the user/ password pair is forgotten.

**Additional context**
Currently if anyone with access to my network with a simple port scanner can find the faikin board IP and find out the password for the Wifi network/ MQTT.

revk, 2024-02-01:
It is the way it is designed. You can turn off the web control settings over mqtt (via setting `webcontrol`).

The MQTT is in the clear, so anyone with access to your LAN can find the password anyway.

Have you considered locking things down, VLANs, etc.?

I appreciate it can be improved, but needs more than just hiding passwords on a web page, honest.

ruimmsimoes, 2024-02-02:
I want the web control enabled, but would like the password fields to be secured.
I can't create VLANs on my network, already considered it but my router doesn't support it.

Replacing the password with a tag on the webcontrol would reduce the vulnerability of the system to packet sniffing when initially posted the passwords and the MQTT communications since the MQTT connection is not being made with TLS.

The ideal solution is to implement HTTPS, MQTTS and add username/ password to access the web control.
I understand this might be complex and time consuming.

Replacing in the web control the password with a tag would be a quick fix that although doesn't secure the system makes it less vulnerable.

I imagine that this amounts to changing a couple lines of code.

revk, 2024-02-02:
FYI, one of the big issues with https and mqtts is memory. On the latest Faikin, mqtts is probably OK (set mqttport to 8883).

https serving brings up certificate issues as well, of course.

You can disable *settings* but allow web control for air-con with `webcontrol` set to `1` I believe.

The underlying form handling is for *any* settings, hence passwords simply being one of them. It may be a consideration to understand at the low level settings handling that a password of `********` is a special case and mean never setting that, this would allow a dummy password to be set in the web form, so maybe...

I'm away for a few days, so not right now.

revk, 2024-02-02:
OK, issuing a beta shortly, do test for me please. I have chosen `✶✶✶✶✶✶✶✶` (unicode 6 pointed stars) as the dummy password. You *can* use that as a real password but only if the password is previously blank, i.e. you cannot change from another password to `✶✶✶✶✶✶✶✶` in one go as it assumes you have not edited/changed the password.

closed 2024-02-02 by revk

ruimmsimoes, 2024-02-02:
I upgraded just now the official version and now I see the ****** instead of the actual passwords.
Thank you very much.

revk, 2024-02-02:
Just to add, this was also an issue with browsers insisting on saving passwords when it is not a "password for the site", which I also wanted to solve.

The whole thing needed my rework of underlying settings library, which I did last weekend, so only now made sense to implement.

So good timing to ask, to be honest.

https://www.revk.uk/2024/02/hiding-passwords.html