revk/xESP32-Faikout #214: After power loss back to AP-mode, but can't connect

kroon040, 2024-02-02:
Hi,

I have 2 units and 2 faikin, and was working well. The airco's did have for 10 minutes have no power, but after setting the power back on it, one faikin dindn't work anymore. It wasn't present on my network, but going o the unit, I see the led flashing Orange/Red. Getting my Phone and see the network faikin-E89F.... When try to connect to that , it doesn't work I can't get an ip adres..

Any idea what is the issue? an I give a fixed ip adres?

kroon040, 2024-02-02:
I give my 192.168.4.2 and hoped that the faikin in ap mode was 192.168.4.1, but didn't work

revk, 2024-02-02:
That is really weird...

I think it's a 10 address, but if it is not doing dhcp it may not be working.

And you won't have a suitable lead to reprogram. Where in the world are you?

kroon040, 2024-02-02:
I'm in the netherlands.

I will look if I can reprogram it I'm using the faikin-s1-pico. If the pins
are exposed it should be no problem. Is there a bin file that I could use
for it?

revk, 2024-02-02:
There are bin files in release and betarelease and there is an esptool command somewhere on github. Probably worth erase-flash first and start from scratch.

kroon040, 2024-02-02:
[image]

Are the pins from right to left , GND, GPIO0, 3v3 , RX, TX?

Flashing is gpio0 to gnd?