revk/xESP32-Faikout #215: [BUG] powerful mode is changing from on to off on its own once enabled.

turbolooser, 2024-02-02:
When I switch on powerful mode then its changing back to off and on and vice versa (flipping) every couple of seconds

[image]

revk, 2024-02-02:
Ok that is weird, I am away for a few days, but I'll see if I can reproduce it when I get back.

MuzzaMcD, 2024-02-03:
I believe the black Asterix next to the Set 28C slider control indicates that you are in Faikin Auto mode. (?)
Not sure that trying to also tell the unit to enforce 'Powerful Mode' is logical.
I see that when Faikin Auto mode is Enabled the message " ✷ Automatic control means some functions are limited" is displayed.
In other words, on the one hand the Faikin Auto setting is trying to perform gradual controls to keep your unit at a specific set temperature and in a very specific range +-1c (?)  but the 'Powerful' request button is saying 'all in' (max thrust from the engine room'), forget set temperature just pump as much heat (or cool) as possible now.

revk, 2024-02-04:
Even so, it should not be flipping continuously. Does you aircon have *powerful*, it seems some settings are used for different things on different models.

Failing that, enabling `debug` and logging MQTT messages may give a clue what is happening.

revk, 2024-02-05:
If you can get me a dump, I'll re-open. I could not reproduce it here.

closed 2024-02-05 by revk