revk/xESP32-Faikout #218: Publishing releases

Sonic-Amiga, 2024-02-03:
Hello! Some time ago i asked you about hosting ESP866 version on your OTA server and you have me a permission.
Now i've finally got to a stage where i have things to publish. Could you send me instrfuctions on how to do it ?

I know this is not really a public thing; here us my email: pavel<underscore>fedin <at> mail.ru

P.S. Yes, i have fully implemented CN_WIRED, works with the simulator.

revk, 2024-02-04:
Not 100% sure, as it has my name and registered trademark in the domain. But you have been very helpful.

If you pull request for a binary add to release directory I could include that.

Sonic-Amiga, 2024-02-05:
Thank you. Will do later; as he tested it and sending doesn't work. I guess timings are off.

Sonic-Amiga, 2024-02-05:
Closing because the question is answered

closed 2024-02-05 by Sonic-Amiga