revk/xESP32-Faikout #221: [FEATURE] Perfera Floor Mount Unit Quirks

sam-m7, 2024-02-04:
**Daikin hardware**
S21, original Module on S801

**Request/Feature**
On the Perfera Floor Mount Unit setting Sensor Mode will reduce the brightness of the status LED (there is no sensor mode on this unit). Comfort mode is also not available, but apparently does nothing.

I think it would be best to have a setting, which deactivates the Sensor Mode (and probably comfort mode, which is doing nothing/is not available on this indoor unit), and instead allows for the brightness to be adjusted (useful at night). Alternatively leaving Sensor Mode and Comfort Mode in the UI and only providing a MQTT command (/send doesn't seam to work, see below) for the brightness would be fine for me too.

In debug mode I've found the following things to change when activating Sensor Mode or changing the brightness via the remote. But I couldn't set the brightness on my own using `command/faikin0/send` with f.e. "D60°08" as payload. It seams there is some encoding difference, as the tx shown is different when instead activating Sensor Mode?

**AC S21 status from debug (info/faikin0/s21)**
**Sensor mode (values for G6):**
On: 0°08
Off: 0°00

**When changing the brightness from the remote (values for G6):**
High: 0°04
Low: 0°08
Off: 0°0<

**Other additional settings**
The Perfera Floor Mount unit also has two additional settings, which are currently not available. This is Heat+ (not that important to me and also not in the official app) and Floor Warming, which disables the upper air outlet and is quite useful. Unfortunately till now I haven't seen a S21 parameter in `info/faikin0/s21` which changes only when activating Floor Warming. The original module (connected to S801) can activate Floor Warming. Is there a way to find out the command for that (if it's available on S21) besides using the debug mode?

revk, 2024-02-04:
I agree, more "disable" options need to be sorted I think

And thanks for finding the S21 messages, should be an easy one to add.

revk, 2024-02-04:
Do we need high/low/off or is on/off OK?

revk, 2024-02-04:
I have issued a beta with a load of `no...` settings to turn off features. I'll have to then look to add the LED.

sam-m7, 2024-02-04:
Thanks! I've tested the beta, worked nicely for comfort and sensor mode.
So far I've only used off or left it at high. It might be useful to have low (at least on MQTT), if someone wants to use the AC at night but sill wants to have an indication the AC is on. With floor warming on and the LED off you wouldn't know that the AC is on otherwise. But low would also be the sensor mode ON value, so one could just use that.

revk, 2024-02-06:
I should be able to sort LED.

I think I'll do as on/off switch. But off can be dim or proper off maybe based on dark mode?

revk, 2024-02-06:
Ok for now, I'll add a setting to set the 4 value called LED. In advanced settings you can turn off the `noled` setting to enable it.

sam-m7, 2024-02-06:
I just tried it, unfortunately changing the switch doesn't do anything, even in MQTT I don't see `info/hostname/tx` changing. It does react to setting High and Low on the remote though. Changing dark mode has no effect it only reacts to low.

Could you maybe also tell me how I would send the S21 command through MQTT `command/hostname/send`? As described above for some reason (probably encoding), this doesn't even work for the `D60°08` which is the same text (`0°08`) as shown in tx when activating sensor mode and is working there. Then I (and someone else reading this) could probably integrate High, Low, Off manually in OpenHAB.

Also do you have any other ideas then watching the debug S21 messages (`info/hostname/s21`) on how to find the S21 command for the floor heating (last paragraph above)?

Btw. the new advanced settings are great :)

revk, 2024-02-06:
Maybe try just 0008 as the unicode will go wrong.

I'll look again. Yes, not done dark mode.

sam-m7, 2024-02-06:
Thanks! Ok, sending `0008`, `0004` and `000<` works, although it also changes the received G6 message to these values, which I'm not so sure is good, because maybe the first part `0°` is doing something else we don't know?

revk, 2024-02-06:
On 06/02/2024 13:57, Sam wrote:
> Thanks! Ok, sending |0008|, |0004| and |000<| works, although it also 
> changes the received G6 message to these values, which I'm not so sure 
> is good, because maybe the first part |0°| is doing something else we 
> don't know?

I am sure it will be doing something else, but for testing, this is 
useful to know.

revk, 2024-02-06:
Ok I think this beta will allow it to be set, not using dark mode as yet, just setting the '4' value part.

sam-m7, 2024-02-06:
Thanks, worked when Sensor Mode was disabled, then turning LED on sent `0°04` which set the brightness to high. Deactivating is has sent `0°00` which did nothing, `0°0<` would be the correct code for off. Weirdly I noticed this is send, when sensor mode and led was both set to on. Is this on purpose?

revk, 2024-02-06:
I need to work out a "sensor mode disabled" version. Thanks for testing.

revk, 2024-02-07:
So I have an "led" setting (on and off). I also have a separate "dark" setting which is to stop the Faikin LED (unless it is unhappy).

There are three possible states (when sensor mode disabled). So these two should cover it.

Can you think of most logical way to do these? Eg should dark make led do off and dim, but not-dark make led do dim and on? Or should that be off and bright.

Once we have a sensible logic it will be easy to code.

Also what is LED state for that last digit for each of 0, 4, 8, and < please. Ie all 4 combinations of those two bits.

revk, 2024-02-07:
OK, this is what I am going for...

- LED off is off
- LED on is dim if dark, or bright is not dark

closed 2024-02-07 by revk

sam-m7, 2024-02-07:
Thanks a lot, I've just tested it and it works as you described it :) I probably would have preferred ON always being high (`4`) and Off being low or off depending on the dark mode setting, as high is (at least on my ac) what you would normally want during the day. But also works the way you have done it. 

Since you haven't answered to my question about finding the S21 command for floor heating, I'm trying again. Any ideas on how I could find the S21 command for it? So far I've gone through the alphabet issuing `R<some letter>` and `F<some letter>` commands, seeing if anything in the response changes when changing floor heating. So far I've only found `RM` which responds with `SM`. And seams to show the angle of the swing/air outlet in some weird number, but was consistent so far. Could we maybe add this in the debug polling and if we are sure what those numbers mean adding it to the status? 

Unfortunately sending `RM<some value from the results>` didn't change anything, so I guess it's read only and I'm still looking for the actual floor heating command, but am slowly loosing hope right now. I also guess tapping into the wires of the existing module (like I've read in another issue) only works if it's also connected to S21, not S801, right?

Still if SM is what I think it is it would be helpful for some automation, where I could send an IR command through some other controller.

revk, 2024-02-07:
S801 may be same protocol, no idea, worth finding out, but careful as some of these things are mains voltage.

Yes, tapping the data, if S21 or X50 format, is best way.

If you can document the SM/RM you have seen I could add simply as a status.

sam-m7, 2024-02-07:
Ok, maybe I will try it in a few weeks, when I have more spare time and feeling adventures...

For SM/RM, I'm getting results like this:
```
Received `{"protocol":"S21","dump":"02534D3034302B5F03","SM":"040+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3034302B5F03","SM":"040+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3331302B5F03","SM":"310+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3331302B5F03","SM":"310+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3034302B5F03","SM":"040+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3034302B5F03","SM":"040+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3034302B5F03","SM":"040+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3331302B5F03","SM":"310+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3331302B5F03","SM":"310+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3831302B6403","SM":"810+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3831302B6403","SM":"810+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3831302B6403","SM":"810+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3433302B6203","SM":"430+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3433302B6203","SM":"430+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3132302B5E03","SM":"120+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3132302B5E03","SM":"120+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3935302D6B03","SM":"950-"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3331302B5F03","SM":"310+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3331302B5F03","SM":"310+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3637302B6803","SM":"670+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3331302B5F03","SM":"310+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3034302B5F03","SM":"040+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3331302B5F03","SM":"310+"}` from `info/faikin0/rx` topic
Received `{"protocol":"S21","dump":"02534D3331302B5F03","SM":"310+"}` from `info/faikin0/rx` topic
```

With `310+` being the upmost postion in swing mode, `040+` being the lowest. During swing it's changing between those two values. When holding swing, it can return another value in between. `670+` is the value for the lid being closed, so lowest position all in all, this is floor heating. I can also get negative values when in the first swing cycle (those values don't make sense anymore). After that the values seam to be consistent. 

Now for the question what unit those values have, I'm not completely sure. What makes the most sense to me from measuring the lid:
310+ = 13 degrees
040+ = 40 degrees
670+ = 76 degrees

But it could also think it to be:
310+ = 0.13 radiant => 7.4 degrees
040+ = 0.4 radiant => 22.9 degrees
670+ = 0.76 radiant => 43.5 degrees

But measuring between 670+ and 310+ gives me an angle around 60 degrees from the pivot point (with some uncertainty), so I think degrees make more sense. Although it could also be something completely different of course. The only think I'm pretty sure of now, is that it's somehow reporting the position of the (vertical) swing on my AC.

revk, 2024-02-07:
Simple, and yes you read the digits right to left like that. My guess would be degrees.

sam-m7, 2024-02-07:
Yes, I think so too. Also I've just noticed, SN seam to have the same values.

revk, 2024-02-07:
Try new beta, it may start logging `anglev`

sam-m7, 2024-02-07:
I'm having SM in the S21 debug now, but no anglev in the status messages.

revk, 2024-02-07:
Damn, missed a bit, one moment.

sam-m7, 2024-02-07:
No problem :)