revk/xESP32-Faikout #222: [FEATURE] Add rssid value to mqtt

jesusvallejo, 2024-02-04:
Hi first of all thanks for the effort and mantaining this project.
Could you please add a feature so it sends via mqtt the rssid value (wifi strength)?

Thank you very much.

revk, 2024-02-04:
I thought one of the status did? But I'll look.

revk, 2024-02-05:
It logs in a `state` message when WiFi connects.

```
2024-02-05 12:18:31.408988:state/AdrianAC {"ts":"2024-02-05T12:18:31Z","id":"0C8B9596F534","up":8,"mqtt-up":1,"app":"Faikin","version":"72db032+","build-suffix":"-S1-PICO","build":"2024-02-05T12:17:21","flash":8388608,"rst":3,"mem":50516,"spi":2094336,"ssid":"IoT","bssid":"0A9D99D6B250","rssi":-40,"chan":3,"ipv4":"10.22.107.20","ipv6":"2001:067c:2a40:0107:0e8b:95ff:fe96:f534"}
```

closed 2024-02-06 by revk