revk/xESP32-Faikout #223: [BUG] Fan Speed "Indoor quiet" cannot be activated

eupesco, 2024-02-05:
**Daikin hardware**
CTXM15R + FTXM20R (S21, BRP069C4x)

**Describe the bug**
Fan speed "Indoor quite" cannot be activated in Faikin (using night)  - and when "Indoor quiet" is activated in Onecta, Faikin reports "Auto"

**To Reproduce**
Steps to reproduce the behavior:
1. Change fan speed in Onecta App to "Indoor quiet"
2. In Faikin, Fan is "Auto"

1. Change to "night" in Faikin
2. Fan speed in Onecta is still "Auto"

**Expected behavior**
When select "night" in Faikin, fan speed should be changed to "Indoor quiet".

**Additional context**
Daikin HA-integration: fan_modes: auto, **quiet**, 1, 2, 3, 4, 5
Faikin MQTT: fan_modes: auto, low, lowMedium, medium, mediumHigh, high, **night**

revk, 2024-02-05:
As far as we know, the S21 messages do not handle reporting quite properly. We do send it, but the status does not differentiate from auto.

If your unit has S21 messages to do this, let me know.

closed 2024-02-05 by revk

revk, 2024-08-13:
We may have found an S21 message, try the latest beta and report back.

eupesco, 2024-08-14:
@revk , great, I can confirm that it works.