revk/xESP32-Faikout #225: [BUG] thermostat breaks hysteresis

miniluma, 2024-02-06:
**Describe the bug**
When _thermostat_ is not set to 1, the heating hysteresis is not working properly.
E.g. lower target temp is 20.5°C
BLE sends 20.48 -> heating starts (code goes in "if" because current < min), but hysteresis is not set to 1.
BLE toggles a bit and sends 20.52 after a second -> heating stops (code goes in "else" because current >= min)
As a result, AC is constantly going on and off within seconds around _min_.

[image]

revk, 2024-02-06:
Once it is set it is comparing to
max, surely?

miniluma, 2024-02-06:
> Once it is set it is comparing to max, surely?
> 
> 20.52 is < max (I assume) so stays on (does not do else).

Yes, after I set thermostat to 1, it worked fine.

max was 21.5

miniluma, 2024-02-06:
But was defaulted to thermostat = 0 obviously

revk, 2024-02-06:
I mis read as when `thermostat` is set, sorry.

That is not a bug, that is how it is designed. It uses some prediction so normally does not change constantly, but it makes it a very tight temperature control.

Feel free to set `thermostat` to `1` if you prefer that way of working.

closed 2024-02-06 by revk

miniluma, 2024-02-06:
> I mis read as when `thermostat` is set, sorry.
> 
> That is not a bug, that is how it is designed. It uses some prediction so normally does not change constantly, but it makes it a very tight temperature control.
> 
> Feel free to set `thermostat` to `1` if you prefer that way of working.

No, no, that's not how I prefer to work. (as you can see here as well https://github.com/revk/ESP32-Faikin/discussions/226) ;-)

Then, I don't understand how it should work with thermostat=0.
It behaved like this yesterday:
[image]
I had heatover=0, heatback=6, Automation on with Target=21 and +/-0.5°C and BLE sensor for Room temperature configured.

Maybe you have a hint how I can get it running as you expect.

revk, 2024-02-06:
Which line is temperature, the dark one? that looks pretty level. But yes, it turns on/off a lot. That is how I expect it to work, to maintain a very close temperature between min and max.

miniluma, 2024-02-06:
The black line is the inlet temp.
The red line is the settemp of the unit.
The dark red line is the fan speed.

All values are from the Daikin cloud, not from faikin, so it's a bit delayed.

miniluma, 2024-02-06:
Btw, the blue line here is the power consumption of the outdoor-unit.
[image]

revk, 2024-02-06:
OK so inlet temp was very flat. So as expected.

miniluma, 2024-02-06:
Sure, because the compressor only had a couple of seconds to a minute or two to pump something into the pipes and the fan was far below level one speed (like 400rpm).
To explain the situation:
The room was at ~20.5°C with 12°C outside air temp. The door was closed and all rooms around (below, above, and at the sides) were heated. With no active heating, it would take hours to come from e.g. 20.6 to 20.4.

The logic war jumping from 24 to 16°C unit target temp while the BLE temp was at 20.5 +/- 0.02. My desired room temp was set to 21.
So even if the inlet temp was stable, the unit wasn't really running and the algorithm wasn't doing something useful to reach towards the desired temp.
Probably, some internal values ran away ;-)

Today morning, the room was at ~19.8 and I tried again. thermostat=0, heatover=0, heatback=6, switchtemp=pushtemp=0
desired 21 +/-0.5
It set fan to 5 (I think this is expected) and did some target-temp control and heated the room to 21.4x (let's say ok). Fan was reduced to 4 an approx 21.3.
[image]
[image]
Now, it seem to cool down.
So far, so good.

One thing (refers to the discussion): The outdoor unit took >300W for nearly the whole time. With a lower target temp (inlet-2 or 3) and fan level 1, it takes 120-200W and the room temp is more stable. Of course, it takes longer to reach max, but the desired room temp is 21, not 21.5. ;-)

revk, 2024-02-06:
OK it is working perfectly for me like this, so rather puzzled.

There are settings to find turn it, to adjust where the targets are for heating and cooling, to make it close. A ±0.02 is very tight.