revk/xESP32-Faikout #227: [BUG] Daikin-target temp too low in auto-mode

turbolooser, 2024-02-06:
My current settings on Faikin are
{"tmin":10,"switchtemp":0,"pushtemp":0,"heatback":3,"heatover":6,"coolback":2,"coolover":5,"tmax":31,"tsample":600}

but what I do currently see is that the faikin module does set the daikin-target to 19.5 only. Once in a while for a few seconds it goes up to 28 for daikin-target but mostly its sitting now on 19/19.5 which is not good enough to add heat to the room which should have a BLE-target of 19.0 (18.5 up to 19.5 for automode)

[image]

revk, 2024-02-06:
There are some other debug logs that you can get that may explain it.

turbolooser, 2024-02-06:
If I just add increase my BLE-target temp by 0.5 it behaves normal again by adjusting the daikin temp to 28
Maybe that gives an idea?

revk, 2024-02-06:
Yeh I suspect you need the debug enabled to show what it is playing at. It is close so may be predicting it going in to range.

turbolooser, 2024-02-06:
I have enabled the debug so where do I find the data? on mqtt?

revk, 2024-02-06:
Yeh MQTT should show all sorts. One status should have "approaching" and "beyond" and stuff in it.

turbolooser, 2024-02-06:
As you can see... it does not stay to high daikin target temps long enough to do something about the temp changes in the room
[image]

revk, 2024-02-06:
Nothing radical, well not that should have been.

Worth double checking the settings in the new "advanced" tab.

turbolooser, 2024-02-06:
{"hot":true,"approaching":52,"beyond":0,"samples":400,"period":600,"temp":18.63,"min":18.50,"max":20.00}
(i have set switchtemp to 0.5 again as another unit is doing it fine with these settings)

The approaching seems to behave not correctly or?

revk, 2024-02-06:
I can double check the code and maybe add more clarity to that report. Worth watching for a bit and seeing it it makes any more sense.

turbolooser, 2024-02-06:
Just added 0.5 to the BLE-target (19.5 now) and immediately temps do increase to 28 on daikin-target (as expected)
{"hot":true,"approaching":93,"beyond":0,"samples":518,"period":600,"temp":18.66,"min":19.00,"max":20.50}

turbolooser, 2024-02-06:
now the automation looks fine (samples etc)
{"hot":true,"approaching":1200,"beyond":0,"samples":1200,"period":600,"temp":18.74,"min":19.00,"max":20.50}

I will now lower the temp by 0.5 again

miniluma, 2024-02-06:
> Just added 0.5 to the BLE-target (19.5 now) and immediately temps do increase to 28 on daikin-target (as expected) {"hot":true,"approaching":93,"beyond":0,"samples":518,"period":600,"temp":18.66,"min":19.00,"max":20.50}

Obviously, you changed the "mode" and gave it a push. In your  previous data, temp was above min, now it's below.

revk, 2024-02-06:
I have found one issue with `coolback`, was basing on min not max, looks like a silly typo. I'll issue a beta in a moment.

turbolooser, 2024-02-06:
Could you check why this is being seen as beyond?
BLE temp is set to 19.5. I would assume that beyond is only when its above temp-max (20)?
Why does it consider 18.75 as beyond? Its not even in te range of 19 to 20 already 

{"hot":true,"approaching":569,"beyond":12,"initial-samples":600,"period":600,"temp":18.75,"min":19.00,"max":20.00}

revk, 2024-02-06:
It uses prediction of temp, and if it is going up it can predict as beyond.

revk, 2024-02-06:
Yes

revk, 2024-02-06:
Are we any better with latest beta?

turbolooser, 2024-02-06:
still the same for me. when the min temp is reached the faikin does lower the target daikin-temp too much
{"hot":true,"approaching":963,"beyond":2,"samples":1200,"period":600,"temp":18.88,"min":19.00,"max":20.00}

target is still 19.5
but just now it has reached 19.02 (BLE) and now faikin only heats with daikin-target of 19.5 which is too low

[image]

turbolooser, 2024-02-06:
why does it stop heating on reaching the min temp only?
It needs to heat until 19.5 and then try to keep this temp coz thats my target.

revk, 2024-02-06:
Sorry the logic is to aim for the window, reaching min is good enough. There is a thermostat mode now that aims for max and then cools to min. Try that.

miniluma, 2024-02-06:
> why does it stop heating on reaching the min temp only? It needs to heat until 19.5 and then try to keep this temp coz thats my target.

Please try modifying heatback.  Just to see what happens.

turbolooser, 2024-02-06:
> > why does it stop heating on reaching the min temp only? It needs to heat until 19.5 and then try to keep this temp coz thats my target.
> 
> Please try modifying heatback. Just to see what happens.

Heatback? Thats the lowering Temp once it has reached the target Window Not sure why this should Do anything other than cooling down

turbolooser, 2024-02-06:
Changed it to this now
tpredictt":0,"tmin":10,"switchtemp":0,"pushtemp":0,"heatback":6,"heatover":6,"coolback":6,"coolover":6,"tmax":31,"tsample":900

turbolooser, 2024-02-06:
and will check on thermostat mode.
stay tuned tommorrow :-)

turbolooser, 2024-02-07:
@revk 
Could there be a mode in faikin where I only use the BLE temp for the daikin algorithm? Something like faikin-no-auto mode but still use the BLE-temp
e.g. replacing the Home-temp with the BLE-temp so that the Daikin only sees the BLE Temp as Home-temp?
Does that make sense?

revk, 2024-02-07:
There is no way to send an external reference to the Daikin, sorry. If I could, yes, that would be great.

wurmtal868, 2024-02-07:
@turbolooser 
Did you change the "demand" slider on purpose or is your "Wohnzimmer" part of a Multi-Split AC?

revk, 2024-02-07:
I change range and steps, what do you need, I can change or make a setting.

turbolooser, 2024-02-07:
> @turbolooser Did you change the "demand" slider on purpose or is your "Wohnzimmer" part of a Multi-Split AC?

I have only 1 multisplit with 3 inside units and Wohnzimmer is Part of that

wurmtal868, 2024-02-07:
I'm asking because in my multi-split AC the demand settings of the individual indoor units interfere with each other. The indoor with the lowest setting can limit the heat output from the outdoor unit even if indoor unit is in standby. The Daikin cloud does it's own mess up by pushing "demand" values unidirectional. So I just want to make sure that demand was changed on purpose and not by accident. If the indoor units have still access to the cloud, I recommend turning of the demand setting in the Onecta app for each unit.

turbolooser, 2024-02-07:
I know about this behaviour and change dem and control using ha for all units the same

wurmtal868, 2024-02-07:
Thanks, 
I did not know that this can now be controlled via HA.
Last time I checked it was only available via MQTT and I started learning Node-Red to use it.

turbolooser, 2024-02-07:
> Thanks, I did not know that this can now be controlled via HA. Last time I checked it was only available via MQTT and I started learning Node-Red to use it.

I am still setting demand control using the altherma hacs addon

closed 2024-02-12 by revk