revk/xESP32-Faikout #230: [BUG]

sjaakie09, 2024-02-07:
**Faikin hardware**
Please advise if this is a Faikin you purchased (e.g. from Amazon), had built, or some other device.
Ideally, please include picture of both sides of the board itself as this will help confirm the version.

hello bought via amazon
have issue with the setup
I get the unit working till the setup from the wifi setup
there i cant find the unit any more in lanscan or also not with the host name?
I hope i can get back to factory setup?
or how to find the unit 
the unit is now flashng red
this is the number i saw in the login
70041DCE3348
my email adres is jd@comm-co.com

thanks for your help!
jacques

**Daikin hardware**
Please advise the model of your Daikin, and how you are connected (e.g. S21 lead, X50 lead, etc).

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

sjaakie09, 2024-02-07:
https://www.reichelt.nl/nl/nl/jst-busbehuizing-1x5-polig-vh-jst-vh5p-bu-p185109.html?search=+jst-pa+5&&r=1
this is the connector i think

revk, 2024-02-08:
One way to force back to the access point mode is to turn off you WiFi. It may take a few minutes...

But it should get dhcp IP address if on your WiFi. It should show that briefly when you set the WiFi.

Sonic-Amiga, 2024-02-11:
Hello!
The easiest way to look up the address is to go to your access point/router/whatever admin page and check "dhcp clients list" or "dhcp status" or whatever else that fits this description. It should be listed there under its own hostname. For example, here is what it looks like in my D-Link. The language on the page is russian, sorry for that.
[image]

sjaakie09, 2024-02-14:
Beste, Dear,
Ok that is been solved thx
But now I still have issues with the connectors

Explan what is the type of the connector on you board?

And do you know the name of the type connector s21 on the daikin?

Still have issues to find one who fits ☹
And think I have now a bad connection
I made it with wires but now it say cant connect with the daikin (als cant switch it on)

It worked ony one day so want to open the airco again but then wont to have a good connector

Ps other questions
For the second unit it is been used for rental purpose
If i have the connection trough home assistant can I then block max hot temp and in summer max cold temp
And the airco is on worng place so I need to have a other thermostaat I see ther is an option for that
Please explain how that can work

It is been necceary to guard the usage of the airco like max temps and also shutdown for open doors

And is it also possibel to read out the kwh been used out of the interface? Line in the version for daikin?

Thx
You made nice optiion
Maybee an idea include the connector in the package 😊

Met vriendelijke groet / With kind regards,
***@***.***<https://comm-co.com/jm2/>
Jacques Duerinck CEO
***@***.******@***.***>

Comm-Co B.V.
Kreekzoom 9
4561 GX  Hulst
The Netherlands
Tel: +31 114 370030
GSM: +31 653 120 225
***@***.******@***.***<https://ewheelz.nl/>[created with MySignature.io]<https://twitter.com/atexshop>[created with ***@***.***<https://nl.linkedin.com/in/jacques-duerinck?challengeId=AQHAi-Fs0-Lh1wAAAYvdylK2ok7NzA6irP0R6i52tF08AkW-uyQBCxRVPtwfspwwReWjC6m5WFdCVpc4-Tuff_-GepJXMa6EQQ&submissionId=abb2045a-0071-9817-4aea-279bfae4233d&challengeSource=AgHJcnaeiOppRQAAAYvdyuWbNO7oSDhbmuPQZCGvfX6Uxe5Sc8Mdb7a4KhBRlzM&challegeType=AgH7_Hj09W8F_QAAAYvdyuWers9w2E30FsryCkBCK1SatbG_yKhyvXU&memberId=AgECOEUb0_Rl1AAAAYvdyuWgmmRlgfXI-eqOta-QLfeZUw0&recognizeDevice=AgEl9duzUhSpagAAAYvdyuWjz6GgpYneL-OCQDE_uC4-rxQ0V2WB>

revk, 2024-02-14:
There is a separate project for reading power, it is not in there yet.

I don't know much about using with HA.

There is a way to work with external bluetooth temperature sensor.

The connector is a JST connector, but you may be able to use jumper wires.

sjaakie09, 2024-02-14:
Beste, Dear,

see below

Met vriendelijke groet / With kind regards,
***@***.***<https://comm-co.com/jm2/>
Jacques Duerinck CEO
***@***.******@***.***>

Comm-Co B.V.
Kreekzoom 9
4561 GX  Hulst
The Netherlands
Tel: +31 114 370030
GSM: +31 653 120 225
***@***.******@***.***<https://ewheelz.nl/>[created with MySignature.io]<https://twitter.com/atexshop>[created with ***@***.***<https://nl.linkedin.com/in/jacques-duerinck?challengeId=AQHAi-Fs0-Lh1wAAAYvdylK2ok7NzA6irP0R6i52tF08AkW-uyQBCxRVPtwfspwwReWjC6m5WFdCVpc4-Tuff_-GepJXMa6EQQ&submissionId=abb2045a-0071-9817-4aea-279bfae4233d&challengeSource=AgHJcnaeiOppRQAAAYvdyuWbNO7oSDhbmuPQZCGvfX6Uxe5Sc8Mdb7a4KhBRlzM&challegeType=AgH7_Hj09W8F_QAAAYvdyuWers9w2E30FsryCkBCK1SatbG_yKhyvXU&memberId=AgECOEUb0_Rl1AAAAYvdyuWgmmRlgfXI-eqOta-QLfeZUw0&recognizeDevice=AgEl9duzUhSpagAAAYvdyuWjz6GgpYneL-OCQDE_uC4-rxQ0V2WB>

There is a way to work with external bluetooth temperature sensor. Ok great!

You mean your board ?

And on the daikin board do you also know that name?

Thx jd

revk, 2024-02-14:
From discussions, this may be it.

I think [JST, EH Female Connector Housing, 2.5mm Pitch, 5 Way, 1 Row](https://uk.rs-online.com/web/p/wire-housings-plugs/3116237)

sjaakie09, 2024-02-15:
Beste, Dear,

hello I found the richt connector on the daikin side (I belave)
Now ony an vendor who can diliver complete ones 😊
This is the one who is in the daikin male  version

[PAP-05V-S JST Sales America Inc. | Connectors, Interconnects | DigiKey]
PAP-05V-S JST Sales America Inc. | Connectors, Interconnects | DigiKey<https://www.digikey.in/en/products/detail/jst-sales-america-inc/PAP-05V-S/759977>

Met vriendelijke groet / With kind regards,
***@***.***<https://comm-co.com/jm2/>
Jacques Duerinck CEO
***@***.******@***.***>

Comm-Co B.V.
Kreekzoom 9
4561 GX  Hulst
The Netherlands
Tel: +31 114 370030
GSM: +31 653 120 225
***@***.******@***.***<https://ewheelz.nl/>[created with MySignature.io]<https://twitter.com/atexshop>[created with ***@***.***<https://nl.linkedin.com/in/jacques-duerinck?challengeId=AQHAi-Fs0-Lh1wAAAYvdylK2ok7NzA6irP0R6i52tF08AkW-uyQBCxRVPtwfspwwReWjC6m5WFdCVpc4-Tuff_-GepJXMa6EQQ&submissionId=abb2045a-0071-9817-4aea-279bfae4233d&challengeSource=AgHJcnaeiOppRQAAAYvdyuWbNO7oSDhbmuPQZCGvfX6Uxe5Sc8Mdb7a4KhBRlzM&challegeType=AgH7_Hj09W8F_QAAAYvdyuWers9w2E30FsryCkBCK1SatbG_yKhyvXU&memberId=AgECOEUb0_Rl1AAAAYvdyuWgmmRlgfXI-eqOta-QLfeZUw0&recognizeDevice=AgEl9duzUhSpagAAAYvdyuWjz6GgpYneL-OCQDE_uC4-rxQ0V2WB>

revk, 2024-02-15:
That does not look the same part, but may fit.

closed 2024-02-15 by revk

sjaakie09, 2024-03-06:
Beste, Dear,

hello i finaly get the connectors in form alixepress
5 Stuks PAP-04V-S PAP-05V-S 24awg Pa 2Mm Pap Dupont 2.54 2.0Mm Pa2.0 2/3/4/5/6/7/8/9/10/12 Positie Rechthoekige Behuizing Connector - AliExpress<https://nl.aliexpress.com/item/1005006291181398.html?spm=a2g0o.order_list.order_list_main.22.21ef79d2kpLlhg&gatewayAdapt=glo2nld>

This one fits not for the full 100% but you can use them
Only one question
Still did not get the unit online
Before with the connection to the s21 the page did not came up via brouwser
But now it is possible to see the webserver page.
But the unit does not come online any idea how to get them back working
He did worked few days but now not more?
What kind of diagnostic i dan do?
The led is flashing blue/white

I double checked the wires
Can it be that rx/tx is wrong can I switch them without problems?
Can i measure somethings?

Thanks jacques

Met vriendelijke groet / With kind regards,
***@***.***<https://comm-co.com/jm2/>
Jacques Duerinck CEO
***@***.******@***.***>

Comm-Co B.V.
Kreekzoom 9
4561 GX  Hulst
The Netherlands
Tel: +31 114 370030
GSM: +31 653 120 225
***@***.******@***.***<https://ewheelz.nl/>[created with MySignature.io]<https://twitter.com/atexshop>[created with ***@***.***<https://nl.linkedin.com/in/jacques-duerinck?challengeId=AQHAi-Fs0-Lh1wAAAYvdylK2ok7NzA6irP0R6i52tF08AkW-uyQBCxRVPtwfspwwReWjC6m5WFdCVpc4-Tuff_-GepJXMa6EQQ&submissionId=abb2045a-0071-9817-4aea-279bfae4233d&challengeSource=AgHJcnaeiOppRQAAAYvdyuWbNO7oSDhbmuPQZCGvfX6Uxe5Sc8Mdb7a4KhBRlzM&challegeType=AgH7_Hj09W8F_QAAAYvdyuWers9w2E30FsryCkBCK1SatbG_yKhyvXU&memberId=AgECOEUb0_Rl1AAAAYvdyuWgmmRlgfXI-eqOta-QLfeZUw0&recognizeDevice=AgEl9duzUhSpagAAAYvdyuWjz6GgpYneL-OCQDE_uC4-rxQ0V2WB>

revk, 2024-03-06:
Wow, OK, S21 is one to one, so doing that gets tx and rx right.

sjaakie09, 2024-03-06:
Beste, Dear,
what can been wrong then?
I did not get unit connect (online)
It worked but not anymore?

How to diagnose it?

[airco daikin 5kw.jpeg]

Met vriendelijke groet / With kind regards,
***@***.***<https://comm-co.com/jm2/>
Jacques Duerinck CEO
***@***.******@***.***>

Comm-Co B.V.
Kreekzoom 9
4561 GX  Hulst
The Netherlands
Tel: +31 114 370030
GSM: +31 653 120 225
***@***.******@***.***<https://ewheelz.nl/>[created with MySignature.io]<https://twitter.com/atexshop>[created with ***@***.***<https://nl.linkedin.com/in/jacques-duerinck?challengeId=AQHAi-Fs0-Lh1wAAAYvdylK2ok7NzA6irP0R6i52tF08AkW-uyQBCxRVPtwfspwwReWjC6m5WFdCVpc4-Tuff_-GepJXMa6EQQ&submissionId=abb2045a-0071-9817-4aea-279bfae4233d&challengeSource=AgHJcnaeiOppRQAAAYvdyuWbNO7oSDhbmuPQZCGvfX6Uxe5Sc8Mdb7a4KhBRlzM&challegeType=AgH7_Hj09W8F_QAAAYvdyuWers9w2E30FsryCkBCK1SatbG_yKhyvXU&memberId=AgECOEUb0_Rl1AAAAYvdyuWgmmRlgfXI-eqOta-QLfeZUw0&recognizeDevice=AgEl9duzUhSpagAAAYvdyuWjz6GgpYneL-OCQDE_uC4-rxQ0V2WB>

sjaakie09, 2024-03-10:
Beste, Dear,
hello one of the boards don't work
It does not communicate with the Daikin
It will not  come on line ?
Whit same cable so that cant be the issue?

It has worked 2 days but then it failed

How to solve this?

Met vriendelijke groet / With kind regards,
***@***.***<https://comm-co.com/jm2/>
Jacques Duerinck CEO
***@***.******@***.***>

Comm-Co B.V.
Kreekzoom 9
4561 GX  Hulst
The Netherlands
Tel: +31 114 370030
GSM: +31 653 120 225
***@***.******@***.***<https://ewheelz.nl/>[created with MySignature.io]<https://twitter.com/atexshop>[created with ***@***.***<https://nl.linkedin.com/in/jacques-duerinck?challengeId=AQHAi-Fs0-Lh1wAAAYvdylK2ok7NzA6irP0R6i52tF08AkW-uyQBCxRVPtwfspwwReWjC6m5WFdCVpc4-Tuff_-GepJXMa6EQQ&submissionId=abb2045a-0071-9817-4aea-279bfae4233d&challengeSource=AgHJcnaeiOppRQAAAYvdyuWbNO7oSDhbmuPQZCGvfX6Uxe5Sc8Mdb7a4KhBRlzM&challegeType=AgH7_Hj09W8F_QAAAYvdyuWers9w2E30FsryCkBCK1SatbG_yKhyvXU&memberId=AgECOEUb0_Rl1AAAAYvdyuWgmmRlgfXI-eqOta-QLfeZUw0&recognizeDevice=AgEl9duzUhSpagAAAYvdyuWjz6GgpYneL-OCQDE_uC4-rxQ0V2WB>

revk, 2024-03-11:
If it seems to be a faulty Faikin module, let me know the Amazon order number and date and I can do a refund.

sjaakie09, 2024-03-11:
Beste, Dear,

ill need a new one pease send new one sam address

***@***.***

Met vriendelijke groet / With kind regards,
***@***.***<https://comm-co.com/jm2/>
Jacques Duerinck CEO
***@***.******@***.***>

Comm-Co B.V.
Kreekzoom 9
4561 GX  Hulst
The Netherlands
Tel: +31 114 370030
GSM: +31 653 120 225
***@***.******@***.***<https://ewheelz.nl/>[created with MySignature.io]<https://twitter.com/atexshop>[created with ***@***.***<https://nl.linkedin.com/in/jacques-duerinck?challengeId=AQHAi-Fs0-Lh1wAAAYvdylK2ok7NzA6irP0R6i52tF08AkW-uyQBCxRVPtwfspwwReWjC6m5WFdCVpc4-Tuff_-GepJXMa6EQQ&submissionId=abb2045a-0071-9817-4aea-279bfae4233d&challengeSource=AgHJcnaeiOppRQAAAYvdyuWbNO7oSDhbmuPQZCGvfX6Uxe5Sc8Mdb7a4KhBRlzM&challegeType=AgH7_Hj09W8F_QAAAYvdyuWers9w2E30FsryCkBCK1SatbG_yKhyvXU&memberId=AgECOEUb0_Rl1AAAAYvdyuWgmmRlgfXI-eqOta-QLfeZUw0&recognizeDevice=AgEl9duzUhSpagAAAYvdyuWjz6GgpYneL-OCQDE_uC4-rxQ0V2WB>

revk, 2024-03-11:
Well, if arrange a refund you can order a new one.

sjaakie09, 2024-03-11:
Beste, Dear,
ill rder a new one
***@***.***

Met vriendelijke groet / With kind regards,
***@***.***<https://comm-co.com/jm2/>
Jacques Duerinck CEO
***@***.******@***.***>

Comm-Co B.V.
Kreekzoom 9
4561 GX  Hulst
The Netherlands
Tel: +31 114 370030
GSM: +31 653 120 225
***@***.******@***.***<https://ewheelz.nl/>[created with MySignature.io]<https://twitter.com/atexshop>[created with ***@***.***<https://nl.linkedin.com/in/jacques-duerinck?challengeId=AQHAi-Fs0-Lh1wAAAYvdylK2ok7NzA6irP0R6i52tF08AkW-uyQBCxRVPtwfspwwReWjC6m5WFdCVpc4-Tuff_-GepJXMa6EQQ&submissionId=abb2045a-0071-9817-4aea-279bfae4233d&challengeSource=AgHJcnaeiOppRQAAAYvdyuWbNO7oSDhbmuPQZCGvfX6Uxe5Sc8Mdb7a4KhBRlzM&challegeType=AgH7_Hj09W8F_QAAAYvdyuWers9w2E30FsryCkBCK1SatbG_yKhyvXU&memberId=AgECOEUb0_Rl1AAAAYvdyuWgmmRlgfXI-eqOta-QLfeZUw0&recognizeDevice=AgEl9duzUhSpagAAAYvdyuWjz6GgpYneL-OCQDE_uC4-rxQ0V2WB>

revk, 2024-03-11:
I did not see the amazon order number for a refund, I can do one once I get it.

sjaakie09, 2024-03-11:
Beste, Dear,

Order #
206-4331075-7289114<https://www.amazon.co.uk/gp/f.html?C=139N1T2SKF371&K=1I5FBAK48PG90&M=urn:rtn:msg:202401252032246e0c3a0c920a4c0c88d205920860p0eu&R=3PWRNA0297L5W&T=C&U=https%3A%2F%2Fbusiness.amazon.co.uk%2Fabredir%2Fgp%2Fcss%2Fsummary%2Fedit.html%3Fie%3DUTF8%26orderID%3D206-4331075-7289114%26ref_%3Dpe_27063361_485629781_TE_on_sh&H=664AKQZOP6D4EMWDSCIUQB7CUI4A&ref_=pe_27063361_485629781_TE_on_sh>

Met vriendelijke groet / With kind regards,
***@***.***<https://comm-co.com/jm2/>
Jacques Duerinck CEO
***@***.******@***.***>

Comm-Co B.V.
Kreekzoom 9
4561 GX  Hulst
The Netherlands
Tel: +31 114 370030
GSM: +31 653 120 225
***@***.******@***.***<https://ewheelz.nl/>[created with MySignature.io]<https://twitter.com/atexshop>[created with ***@***.***<https://nl.linkedin.com/in/jacques-duerinck?challengeId=AQHAi-Fs0-Lh1wAAAYvdylK2ok7NzA6irP0R6i52tF08AkW-uyQBCxRVPtwfspwwReWjC6m5WFdCVpc4-Tuff_-GepJXMa6EQQ&submissionId=abb2045a-0071-9817-4aea-279bfae4233d&challengeSource=AgHJcnaeiOppRQAAAYvdyuWbNO7oSDhbmuPQZCGvfX6Uxe5Sc8Mdb7a4KhBRlzM&challegeType=AgH7_Hj09W8F_QAAAYvdyuWers9w2E30FsryCkBCK1SatbG_yKhyvXU&memberId=AgECOEUb0_Rl1AAAAYvdyuWgmmRlgfXI-eqOta-QLfeZUw0&recognizeDevice=AgEl9duzUhSpagAAAYvdyuWjz6GgpYneL-OCQDE_uC4-rxQ0V2WB>

revk, 2024-03-11:
I have managed to work out the order number from that long mangled URL, and refund you.