revk/xESP32-Faikout #231: New variation of CN_WIRED

Sonic-Amiga, 2024-02-07:
We are testing CN_WIRED protocol implementation; and we found some AC models, which don't conform to current documentation: https://github.com/Sonic-Amiga/ESP8266-Faikin/blob/main/Tools/Simulators/CN_WIRED/README.md

Symptoms are: physical format seems to be correct, but data encoding appears totally different.
Examples are:
https://github.com/revk/ESP32-Faikin/issues/126#issuecomment-1931784397
https://github.com/revk/ESP32-Faikin/issues/126#issuecomment-1932417630

I tried calculating checksum by hands and indeed i can't see any pattern here.
Some questions: 

1. Could there be more than 8 bytes in the packet ? Have we checked ?
2. We know that A/C sends room temperature every second. Does it maybe have some display, so that we know what value we're looking for ?
3. We can try capturing what the A/C sends when remote control is used. That will give us set point value, we can look for it, this could eliminate (2)
4. @akirafrabbani: which A/C model is that ? So that we know. Also i could check whether it's on support list of my Daichi. If so, i could assist figuring out controller side.
5. Do you have any known control unit, that connects to this A/C ? Like wall panel, or maybe some 3rd party controller which supports it ? Again, we need both sides of communication.

 We mostly have to restart the research (mostly) from scratch with this A/C.

revk, 2024-02-08:
I'll double check my code, but I think we do look for "too long" as well.

revk, 2024-02-08:
set_val (swingv, (payload[5] & 0x20) ? 1 : 0);

I'm sure I read "bit 5", but OK, changed. Changed.

revk, 2024-02-08:
More examples... I'll keep the log going to try and find more.
```
1F202000005A1074
1D780400000010B2
1C780400000010C2
```

revk, 2024-02-08:
I have updated my checksum code, and made these report as unknown message type.

Sonic-Amiga, 2024-02-09:
> I'm sure I read "bit 5", but OK, changed

My bad, Hit the wrong key. Or maybe counted from 1 like a n00b. Just pushed fix for the doc.

I wonder, does this AC also transmit messages we know, or this is all it sends ?

revk, 2024-02-09:
Sends the normal type 0 (and 1 on mode change) as well but also these 2 and 4 regularly.

Sonic-Amiga, 2024-02-26:
Do we know AC types ?

Sonic-Amiga, 2024-07-25:
FTV35PBV1MF and FTKG50 send type 2 and type 4: https://github.com/Sonic-Amiga/ESP8266-Faikin/issues/19
```
	•	“dump”:“1E7D0000000000A2”
	•	“dump”:“00E20000000000B4”
	•	“dump”:“1E7D000000001092”
```
There's a suggestion that byte[0] in type 2 stands for outside temperature. Unable to verify.

revk, 2025-02-12:
Parking this.

closed 2025-02-12 by revk