revk/xESP32-Faikout #234: [BUG] Fan mode "Night" ans "Auto" is not taken over if set in App/Cloud

miniluma, 2024-02-09:
**Describe the bug**
Coming from a defined fan step (1-5): If you set Fan mode to Auto or Night via Remote control or Cloud, Faikin is not taken these values. MQTT and Web Interface keep last fan level.

miniluma, 2024-02-09:
Set Auto or Night in Faikin works fine.

revk, 2024-02-09:
If it won't change then it may be the S21 messages are not accepted. A `dump` / `debug` would help.

miniluma, 2024-02-09:
dump is coming too fast. So I took debug.

Fan 1->2 in cloud
s21_G1
14<4

Fan  2->1
s21_G1
14<3

Fan 1->Night
s21_G1
14<A

Fan Night ->Auto
s21_G1
14<A

Night and Auto comes as 'A'. It's not handled.
[image]

revk, 2024-02-09:
OK night should show as auto usually as the S21 messages do not seem to include reporting night more. E.g even if you change to night mode from the remote control it will report as auto in the messages.

Are you saying we don't see 'A' as auto?

miniluma, 2024-02-09:
> Are you saying we don't see 'A' as auto?

Right.
I assume e.g. daikin.fan equals 1 and daikin.fanrpm is 690. And then comes 'A' which is not handled at all..

revk, 2024-02-09:
Hmm... Yeh, maybe a small hole in the logging there.

revk, 2024-02-09:
OK I'll make a beta now with a change of logic, try in a few minutes..

closed 2024-02-09 by revk

miniluma, 2024-02-09:
Now it goes at least from Fan=1 to Auto. So only night is not correct, but I understand that you don't see this from the S21 message.

revk, 2024-02-09:
Thanks, yes, annoying.

revk, 2024-08-13:
We may have found an S21 message, try the latest beta and report back.

miniluma, 2024-08-13:
> We may have found an S21 message, try the latest beta and report back.

I confirm. Seems to work perfectly. In both directions: Changes made in the app are reflected in Faikin and changes made in faikin show up in the app. Great work!