revk/xESP32-Faikout #239: [BUG] ECO-mode toggle does not remain active

eupesco, 2024-02-15:
**Faikin hardware**
latest beta from today

**Daikin hardware**
Daikin 3MXM40A9 + 2x CTXM15R + 1x FTXM20R

**Describe the bug**
When I enable ECO mode in Faikin, ECO mode is enabled on the device, but the toggle does not stay enabled (so it goes back to off after a few seconds). 
When ECO mode is **on** on the device, toggle is off. When it is activated, ECO mode remains on.
When ECO mode is **off** on the device, toggle is deactivated. When it is activated, ECO mode is on but toggle goes to off after a few seconds.

revk, 2024-02-15:
It won't do if the aircon does not accept the message. You can enable debug and capture messages in MQTT and we can take a look.

eupesco, 2024-02-15:
I hope, I captured the correct messages:
Switch on ECO in Faikin:
```
{"G1":"04CA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u00B0\u00800","G5":"000\u0080","G6":"0008","G7":"l0FA","G8":"0200","GA":"4230","GB":"0460","GC":"BF21","GG":"0800","GK":"q{51","GM":"CDD0","GN":"0000","GP":"E400","GQ":"7500","GS":"4000","GT":"3000","SH":"081+","SI":"581+","Sa":"581+","SL":"000","Sd":"000","SN":"521+","SM":"521+","SX":"012+","SD":"000"}

{"G1":"04CA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u00B0\u00800","G5":"000\u0080","G6":"0008","G7":"l0FA","G8":"0200","GA":"4230","GB":"0460","GC":"BF21","GG":"0800","GK":"q{51","GM":"CDD0","GN":"0000","GP":"E400","GQ":"7500","GS":"4000","GT":"3000","SH":"081+","SI":"581+","Sa":"581+","SL":"000","Sd":"000","SN":"521+","SM":"521+","SX":"012+","SD":"000"}
```

Toggle goes off although ECO still on (after approx. 24sec):
```
{"G1":"04CA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u00B0\u00800","G5":"000\u0080","G6":"0008","G7":"l0F@","G8":"0200","GA":"4230","GB":"0460","GC":"BF21","GG":"0800","GK":"q{51","GM":"CDD0","GN":"0000","GP":"E400","GQ":"7500","GS":"4000","GT":"3000","SH":"081+","SI":"081+","Sa":"581+","SL":"000","Sd":"000","SN":"521+","SM":"521+","SX":"012+","SD":"000"}

{"G1":"04CA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u00B0\u00800","G5":"000\u0080","G6":"0008","G7":"l0FA","G8":"0200","GA":"4230","GB":"0460","GC":"BF21","GG":"0800","GK":"q{51","GM":"CDD0","GN":"0000","GP":"E400","GQ":"7500","GS":"4000","GT":"3000","SH":"081+","SI":"581+","Sa":"581+","SL":"000","Sd":"000","SN":"521+","SM":"521+","SX":"012+","SD":"000"}
```

revk, 2024-02-15:
It is normally the second digit of `G7`, which is zero.

I see a change in last character of `G7`. If you change this setting on your remote control, what does `G7` do?

It is possible this is another field on your model.

eupesco, 2024-02-15:
Ok, reason found (seems unsolvable): although Faikin sends the message correctly and the device also accepts the mode (because ECO is on in the app), the response message is not changed --> second digit of G7 stays 0. When device is switched on, the digit changes to 2. 
It seems to be the same chaos as with the remote control vs. app: ECO mode can be always activated in app, but only during on mode on the remote control.

revk, 2024-02-15:
Looks like we send `2` as well, so I would have expected it to accept it.

There is a command to send S21 messages of you choice, you could try other values like '1', etc, and see if you can make it work. If you find one that works, let me know.

closed 2024-02-15 by revk