revk/xESP32-Faikout #240: CN-WIRED ongoing work.

revk, 2024-02-15:
Closing [#126](https://github.com/revk/ESP32-Faikin/issues/126) to start again as far too long - let's start again.

# Where are we?

We have details of low level timings and codings, and dumps of messages both ways, and a good idea of the protocol. A lot done by @Sonic-Amiga researching. However, we have seen some variations which even have a different checksum system.

This Faikin s/w now has CN_WIRED which can be enabled by turning off the new `nocnwired` setting. It has basic sending and receiving of messages, but it seems sending is not entirely reliable with sent messages actually working. Given how well the low level timing is understood, this seems protocol (byte level) as a likely issue, and we do not understand all the bytes. Interestingly @Sonic-Amiga has seen 900uS mark, but others have recorded 1000uS mark, including my testing with @akifrabbani

## We now have new settings in Faikin...

- `cnsend4` Causes 4 sends of packets when sent instead of 1
- `cnmark900` Causes use of 900uS mark not 1000uS
- `cnbyte1` Allows byte 1 in sent message to be set (in hex)
- `cnbyte2` Allows byte 2 in sent message to be set (in hex)

I can add more settings and options. I expect to remove these once properly cracked.

Setting `nos21` and `nox50a` and `noswaptx` and `noswaprx` and unsettling `nocnwired` forces CN_WIRED and no others.

Sonic-Amiga, 2024-02-15:
> has seen 900uS mark, but others have recorded 1000uS mark, including my testing

This is a surprise-surprise!

I've dug down my archives and found raw_sampler dumps made by @dremugit. Here they are:

Air conditioner sending once per second to emptiness, no wall panel attached:
```
13:51:56.201 -> 0 2616 1 992 0 316 1 388 0 312 1 388 0 312 1 392 0 312 1 988 0 312 1 992 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 452 0 252 1 388 0 324 1 992 0 300 1 388 0 324 1 380 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 392 0 312 1 992 0 312 1 388 0 312 1 992 0 312 1 388 0 312 1 404 0 300 1 388 0 324 1 376 0 312 1 392 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 452 0 252 1 388 0 324 1 376 0 312 1 392 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 404 0 300 1 388 0 348 1 356 0 312 1 388 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 388 0 312 1 404 0 300 1 388 0 324 1 380 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 992 0 312 1 388 0 316 1 388 0 312 1 16324 0 2016 1 931120
                0 2616 1 992 0 312 1 392 0 312 1 388 0 312 1 388 0 316 1 988 0 312 1 992 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 992 0 312 1 388 0 312 1 392 0 312 1 424 0 276 1 388 0 328 1 376 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 388 0 316 1 988 0 312 1 392 0 312 1 992 0 308 1 392 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 400 0 304 1 388 0 372 1 328 0 312 1 392 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 400 0 304 1 388 0 324 1 376 0 312 1 392 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 404 0 300 1 388 0 324 1 376 0 312 1 392 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 424 0 280 1 388 0 324 1 376 0 316 1 388 0 312 1 992 0 312 1 388 0 312 
13:51:58.289 -> 0 2620 1 992 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 992 0 312 1 988 0 316 1 388 0 312 1 388 0 316 1 400 0 300 1 388 0 372 1 332 0 312 1 988 0 324 1 380 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 388 0 316 1 448 0 252 1 992 0 312 1 400 0 300 1 992 0 312 1 424 0 276 1 392 0 324 1 376 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 400 0 300 1 392 0 376 1 324 0 312 1 392 0 312 1 388 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 400 0 300 1 392 0 324 1 376 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 400 0 304 1 388 0 324 1 376 0 312 1 392 0 312 1 388 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 992 0 312 1 388 0 312 1 392 0 312 1 16324 0 2016 1 931376
                0 2616 1 992 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 992 0 312 1 992 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 392 0 312 1 988 0 312 1 392 0 312 1 388 0 312 1 424 0 276 1 392 0 324 1 376 0 312 1 392 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 992 0 312 1 388 0 312 1 992 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 404 0 300 1 388 0 372 1 332 0 312 1 388 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 400 0 304 1 388 0 324 1 376 0 312 1 392 0 312 1 388 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 400 0 300 1 392 0 324 1 376 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 392 0 308 1 392 0 312 1 424 0 276 1 388 0 328 1 376 0 312 1 388 0 312 1 992 0 312 1 388 0 312 
13:52:00.323 -> 0 2616 1 988 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 988 0 312 1 992 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 388 0 312 1 388 0 312 1 992 0 312 1 388 0 312 1 400 0 300 1 388 0 376 1 328 0 312 1 388 0 312 1 388 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 992 0 312 1 388 0 312 1 988 0 312 1 392 0 312 1 388 0 312 1 388 0 312 1 404 0 300 1 388 0 324 1 376 0 312 1 392 0 308 1 392 0 312 1 412 0 288 1 388 0 312 1 388 0 312 1 388 0 312 1 392 0 312 1 400 0 300 1 388 0 324 1 380 0 312 1 388 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 388 0 312 1 392 0 308 1 392 0 312 1 424 0 276 1 392 0 324 1 376 0 312 1 392 0 312 1 388 0 312 1 388 0 312 1 392 0 316 1 384 0 312 1 388 0 316 1 388 0 312 1 400 0 300 1 392 0 372 1 328 0 312 1 388 0 312 1 392 0 312 1 988 0 316 1 388 0 312 1 388 0 312 1 16320 0 2016 1 931440
                0 2616 1 992 0 312 1 400 0 300 1 388 0 324 1 380 0 312 1 988 0 376 1 940 0 300 1 392 0 320 1 376 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 988 0 312 1 392 0 312 1 388 0 312 1 388 0 312 1 388 0 312 1 392 0 312 1 400 0 300 1 388 0 324 1 376 0 316 1 388 0 312 1 988 0 316 1 388 0 312 1 988 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 388 0 312 1 388 0 312 1 388 0 316 1 388 0 312 1 400 0 300 1 392 0 320 1 380 0 312 1 388 0 312 1 388 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 388 0 312 1 388 0 312 1 448 0 256 1 388 0 324 1 376 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 388 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 400 0 300 1 388 0 372 1 332 0 312 1 388 0 312 1 388 0 312 1 388 0 312 1 392 0 312 1 388 0 312 1 388 0 312 1 388 0 316 1 400 0 300 1 988 0 312 1 452 0 252 
13:52:02.404 -> 0 2612 1 992 0 316 1 396 0 304 1 388 0 324 1 376 0 312 1 992 0 344 1 968 0 300 1 388 0 324 1 380 0 312 1 388 0 312 1 388 0 312 1 388 0 312 1 992 0 312 1 388 0 312 1 392 0 308 1 392 0 312 1 388 0 312 1 388 0 312 1 400 0 300 1 392 0 324 1 376 0 312 1 388 0 312 1 992 0 312 1 388 0 312 1 992 0 312 1 388 0 312 1 388 0 312 1 388 0 312 1 388 0 312  [... cut ...] wards compatible manner.

As to unknown bytes [1] and [2]; there definitely should be some capability flags. The AC must have way to indicate which optional features it supports (like H/V swing, LED control, etc). These flags must come with packet type 0, because that's the only data sent regularly. We can try correlating data in those packets to AC models and features they have. But that's low priority; we need to beat tx.  We're definitely doin' it wrong(tm); and this wrongness is damn simple.

Sonic-Amiga, 2024-02-15:
By the way, those who aren't afraid of Arduinos, can do two-way dump right now:
1. Download code from my repo. The sketch you need is in Tools/Simulators/CN_WIRED/bridge
2. Uncomment "#define RX2_PIN 3"
3. If you want to get raw timings dump, uncomment also "#define DUMP_TIMINGS"
4. Wire pin 2 to one line and pin 3 to another. Just in parallel with the wall panel.
5. Flash the sketch in, obviously
6. Open serial port monitor in Arduino IDE, or any standalone terminal app on Arduino's COM port
7. Watch the logs coming. It will look something like this:
```
Rx1 80 24 04 50 00 00 00 10 00 OK
Rx2 80 21 00 23 18 08 00 10 A0 OK
Rx1 80 24 04 50 00 00 00 10 00 OK
Rx2 80 21 00 23 18 08 00 10 A0 OK
Rx1 80 24 04 50 00 00 00 10 00 OK
Rx2 80 21 00 23 08 08 00 10 90 OK
```
Note this is an old dump; newer version will add microsecond timestamp to each packet, so that they can be correlated in time. Timings, if enabled, will look like long string of numbers. Something like:
```
Rx1 T 2600 1000 300 400 300 400 300 900 ...
```
(this isn't a real example, i made it up because i never logged it)

revk, 2024-02-16:
I could perhaps be electrical?

Does anyone using a Faikin board on CN_WIRED have an oscilloscope they can use to check Tx?

The board was designed for serial, where the Daikin has internal pull ups, so is driven open drain. But if there is not a pull up, then we need to ensure the 5V is connected to there Faikin board to pull up.

Sonic-Amiga, 2024-02-16:
@revk Good thing to check. You may look at schematic of "Daichi" controller in my repo under Hardware; maybe this will give you some hints. I agree that Tx circuit looks weird; but i rechecked 10 times; Q1 indeed goes to Vcc, and R2 indeed goes to Vio, which looks like +5V. I have no idea how the line is brought down. Perhaps i guessed Q1 wrong; the transistor is unmarked, i simply tried to pick by footprint something that made the most sense.

Maybe that gives you some hints. Anyways, this circuit does have pullups.

revk, 2024-02-17:
Well it also depends if the 5V has been connected to the Faikin. A scope on the Tx would help.

Sonic-Amiga, 2024-02-19:
!!! BREAKING NEWS !!! IT WORKS !!!
Glinka65 has found a blown FET on his ESP8266 board. After replacing it everything works like a charm!

https://github.com/Sonic-Amiga/ESP8266-Faikin commit 6e7f26748a7d9ca20de9497b7d76b1ef1a345a0b - this is known working version

@revk If you do everything exactly the same way; the only possible problems are indeed electrical.

I will continue trying to improve the implementation; disabling interrupts on a running OS feels very harsh; But, anyways, we've got known working implementation.

revk, 2024-02-20:
It would help to know if existing testers have 5V connected, and if any have a scope.

adamsguitar, 2024-02-20:
> It would help to know if existing testers have 5V connected, and if any have a scope.

I have 5v connected but do not have a scope, unfortunately. I can say that receiving does seem to work (though it does not seem like the mode is necessarily accurately reflected--I have mine running on Auto (not Faikin auto) and it's showing set as cool, but I'm looking at it right now and it's heating.

Happy to connect it to another MQTT server if it's helpful for debugging.

revk, 2024-02-21:
Well, connecting to `testmqtt.revk.uk` would allow me to see it, but I think we need to confirm the electrical connection somehow. Glad you have the 5V connected, that really should work then.

adamsguitar, 2024-02-21:
> Well, connecting to `testmqtt.revk.uk` would allow me to see it, but I think we need to confirm the electrical connection somehow. Glad you have the 5V connected, that really should work then.

I assume not, but anything I could do with a standard multimeter to help?

revk, 2024-02-21:
I doubt it, it works some times, so likely signal timing with the 10k pull up not being enough or some such.

Sonic-Amiga, 2024-02-22:
Hello everyone!  Recent experiment proved that 16ms delay, then 2ms low pulse after the packet is mandatory. Without them the conditioner does not accept the packet.

I am on a short trip for 2 days. See my git log and driver code for more info. It has been verified to work.

Also pay attention to recent commit regarding swing control. Apparently "Daichi" controller , from which i derived the protocol, doesn't get it 100% right.

I will update the documentation when i come back

revk, 2024-02-23:
My code does the 16ms high, 2ms low, at the end, already. Hmm.

Sonic-Amiga, 2024-02-25:
And you also have a 10K pullup; totally the same as Daichi board has...
Next week i am leaving for vacation; will be back on 16th of March. I checked ESP32 prices; i can buy a Pico devboard and do some testing with the gear i have (simulator and Arduino bridge). You really seem doing everything correctly; there's a very small thing somewhere.

Sonic-Amiga, 2024-02-25:
So, i switched TX back to timer interrupts; and it all works fine. The only thing i had to keep is 16ms high, 2ms low termination sequence.
I guess Daichi does not do this, but still works because it repeats the packet 6 times; and 2500ms start of the repeated packet is somehow taken by the AC as end pulse. Pause between repeats is 16ms; it matches. That's just for info. As i said, my implementation doesn't repeat; it just sends once. Glinka65 says it works great.

Sonic-Amiga, 2024-02-26:
Updated the protocol doc. Please pay attention to swing control details; appears quirky.

revk, 2024-02-27:
For my own reference
https://github.com/Sonic-Amiga/ESP8266-Faikin/tree/main/Tools/Simulators/CN_WIRED

Sonic-Amiga, 2024-03-01:
Hello guys!
 I am here, it's all OK, i am silent because i am busy preparing for the vacation. Leaving for Thailand, will come back on 16th. After this, if you don't fix things by yourself, i'll buy a ESP32 Pico board and try playing with this myself.

type-rke, 2024-05-05:
is this still on thé table?

revk, 2024-05-05:
We got a bit stuck really.

Recreating the protocol, but seems not to play.

We can receive.

Sonic-Amiga, 2024-05-05:
We are assuming there's an electrical incompatibility in Faikin board. And i've got an idea how to verify this.

My ESP8266 port works great on its original target HW; my user Glinka65 is happy. So, what if we take a Faikin board and transplant a 8266 module onto it ? My board is built around ESP-12F; so that's just 100% known to work. Flash with my firmware and try. If the issue persists; it's indeed electrical problem. If not - the code is broken and we're missing something out; just look real carefully.

Also i have recently purchased an ESP-32S-based devboard; so i'll be able to play with the firmware on my side. But i don't have an actual A/C; only a simulator; and Glinka doesn't live in my city so we can't meet up for hands-on testing.

revk, 2024-05-06:
Equally my code on a board with 5V level shifters may work.

Sonic-Amiga, 2024-05-06:
Looked at Faikin schematic once again... Indeed, it's possible that pull-up on TX line (from Faikin to AC) is missing. And the fix could be as simple as connecting pins 1 and 4 of Faikin connector together.

!!! Before doing this, make sure that power supply voltage from the AC is 5V and no more !!!

lank23, 2024-06-23:
Evening all.  

specs
indoor unit = FTXB18AXVJU
device = HiLetgo ESP-WROOM-32 ESP32 ESP-32D

I have a unit that only has CN-WIRE connection.  I made a cable using info found on this site and connected up the ESP32 device.  Erasing then Flashing the ESP32 device with the latest beta or main release firmware using the below commands allowed the unit two boot and operate correctly.

`esptool.py  erase_flash`

then

`esptool.py --port /dev/ttyUSB0 write_flash 0x1000 Faikin-S1-bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 Faikin-S1.bin`

After I configured the TX and RX GPIO to TX = 1 and RX = 3.  Then I set the settings to only use CN-WIRE but I continue to have offline message on the web interface.  But when I use the remote to change any setting the web page is updated in real time.

I hooked my oscilloscope up to TX pin of the ESP32 and I get nothing when changing settings on the webpage or operating the remote.  Tried a 10k pull up on TX pin, still nothing....I have tried changing many settings and different GPIO pins (some prevented boot) but still nothing makes the TX pin work!  If I connect my oscilloscope to the RX pin I get output from the unit every second and when using the remote. 

So why do I get nothing on TX pin and the web page shows offline?

Thanks.

revk, 2024-06-23:
I am not sure we finally crasked it, but you should see data sent!

lank23, 2024-06-23:
Ok so I figured maybe my esp32 device was bad or non compatible in some way, so I switched to a ESP8286 device "ESP8266 ESP-12 ESP-12F NodeMcu Mini D1 Module WeMos Lua 4M Bytes" and uploaded branched firmware [from here](https://github.com/Sonic-Amiga/ESP8266-Faikin) set TX = 1 RX = 3, selected only CN-WIRE for protocol and it is working!  So not sure if its the hardware or the firmware...Thanks

Sonic-Amiga, 2024-06-23:
Small note. My port ignores rx/tx pin setting; those are hardcoded to 3 and 1 because that's the only valid choice. In 8266 they aren't reprogrammable.

Sonic-Amiga, 2024-06-23:
@lank23 Can you describe your HW setup more precisely ? Have you taken your Faikin board and transplanted a 12F onto it; or have you done something else ?

"Offline" means you aren't receiving any data. CN-WIRED ACs don't need to be actively queried, they just send status once per second. So you should at least see data coming from the AC, and the controller should be able to decode and display current status once you control the conditioner using its remote.

lank23, 2024-06-24:
the hardware I was trying to use with the original version of Faikin is a HiLetgo ESP-WROOM-32 ESP32 ESP-32D[http://www.hiletgo.com/ProductDetail/1906566.html](url) which you can find at several locations....so once I programmed the unit using these commands

`esptool.py  erase_flash`

then

`esptool.py --port /dev/ttyUSB0 write_flash 0x1000 Faikin-S1-bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 Faikin-S1.bin`

the unit boots, the CN-Wire port TX and RX pins are connected to the TX RX pins of the device IE
CN-WIRE TX - RX device
CN-WIRE RX - TX device

In the settings the GPIO settings are TX = 1 RX = 3 and all protocols except for CN-WIRE are off.

I did see data coming from the AC unit, and it would update the status's of the webpage, but would remain offline.  I could be a hardware issue, but I tried two of the same devices and got the same results.

dremugit, 2024-06-24:
Mmm. Note that the CN_WIRED pins are 5V logic and the ESP32 is 3v3. You'll need some sort of level conversion. You may be doing that already and oversimplifying for us, but, you know, to be specific.
-- A

lank23, 2024-06-24:
I did not have any level conversion on any of the devices.  The ESP8286 works without level conversion.

dremugit, 2024-06-24:
I haven't used that device so can't speak to it, but we may be comparing apples and kumquats. The ESP32 may not be able to drive the CN_WIRED on its own whereas the other may.
I'm using an "Node32S" ESP-32, very much or identical to yours, but (1) with level conversion and (2) my own software, so I'm not much help. But that hardware can both speak and listen to the CN_WIRED, so another data point.
-- A

Sonic-Amiga, 2024-06-24:
> I did see data coming from the AC unit, and it would update the status's of the webpage, but would remain offline.

Hm, looks like SW bug. Once data is received, Faikin should not be offline any more.
I definitely need to defeat my procrastination and find it out. I do have ESP32 lying around on my desk.

lank23, 2024-06-25:
Anything I can do to help let me know!

Sonic-Amiga, 2024-07-10:
Hello there!
@revk You've merged important fixes; could you release a new beta ? I think it's time to test.

revk, 2024-07-11:
Done

Sonic-Amiga, 2024-07-11:
Hello! Anyone still here ?
We've just released a new beta, in which i made some fixes regarding CN_WIRED and successfully tested it on my simulator.
Anyone to test ?

type-rke, 2024-07-11:
If you can tell me how i can put it on my board

what is the server adress?

Sonic-Amiga, 2024-07-11:
Default one. http://ota.revk.uk/ . You should enable Beta in advanced settings AFAIK.

type-rke, 2024-07-11:
[image]

Sonic-Amiga, 2024-07-11:
@revk ???

type-rke, 2024-07-11:
[image]
[image]

strange

Sonic-Amiga, 2024-07-11:
Or maybe open a new issue regarding update ?

revk, 2024-07-12:
Weird, I'll look.

revk, 2024-07-12:
I have rebuilt and checked from my units that beta is OK loading now.

type-rke, 2024-07-12:
strange for me it keeps failed

revk, 2024-07-12:
Do you know what IP you would be coming from, I can check logs. But popping out for a bit right now.

type-rke, 2024-07-12:
Can i send it to you in private?

would not expose my IP to the whole world :-)

type-rke, 2024-07-12:
ok, ive managed to update via hotspot

indeed there is something blocking the server

When i use the remote everything is changing, but when i change someting in the web screen nothing happens

[image]

Sonic-Amiga, 2024-07-13:
Aha.
Thank you for testing. There's now only a very small amount of difference between 8266 and ESP32 versions, so we'll eventually get it going.
And i even have some ideas.

revk, 2024-07-13:
I don't remember, I think it was based on timings people made.

Sonic-Amiga, 2024-07-13:
@revk After some thinking, i came up with this: https://github.com/revk/ESP32-Faikin/pull/392 Pretty simple, eh ?

Sonic-Amiga, 2024-07-13:
@revk You perhaps didn't get the exact question. Why this:
```
   seq[p].duration0 = CN_WIRED_SYNC - 1000;  // 2500us low - do in two parts? so we start with high for data
   seq[p].level0 = TX_LOW;
   seq[p].duration1 = 1000;
   seq[p++].level1 = TX_LOW;
```
instead of just one full-length interval ?

If you repeat "don't remember" that's okay; i actually have possibilities to play with this myself.

revk, 2024-07-13:
Well it is a once sided pulse or something, I do forget.

Sonic-Amiga, 2024-07-15:
I checked; and tx (Faikin -> AC) stops working with simulator if i try to encode start bit as just one symbol.
Apparently there's some limit (TL;DR full rmt docs, sorry :) )
This means that CN_WIRED_IDLE also doesn't work correctly. I'll extend by bridge to track that and try to fix.

Sonic-Amiga, 2024-07-16:
So far, all protocol code is the same between 8266 version (which we know works) and ESP32. I verified all timings again, it's all good. My ESP32 devboard works great with my simulator.

The only option left is HW incompatibility. And with the latest board that should be simple.

@type-rke Are you brave enough to try a simple hardware mod ? You need to connect together pins 1 and 3 on Faikin connector. Assuming the board is latest revision.

Something like this:

[image]

type-rke, 2024-07-16:
Did not do anything except my faikin gets warm so i disconected,  will it fry?

type-rke, 2024-07-16:
what is strange, back in ferbuary i was able to control for 2 days now i can only do a readout, so it was working back then

type-rke, 2024-07-16:
this is my board, is it OK that i connect these together?

[image]

Sonic-Amiga, 2024-07-16:
Where is pin1 (marked 5V) connected to ? CN_WIRED connector only has 4 pins.
Do not do anything before answering this question.

The board itself looks proper. To double-check, this pin should only be connected to a resistor on the board and nothing else. Other end of that resistor sits on the power line (marked +12V).
@revk Please confirm to triple-check.

[93 more replies]