revk/xESP32-Faikout #242: [BUG] Unable to set target temperature via Home Assistant when Faikin Auto mode enabled

shaohung, 2024-02-17:
**Faikin hardware**
Amazon

**Daikin hardware**
S21

**Describe the bug**
Unable to set target temperature via Home Assistant when Faikin Auto mode enabled. 

The target value (from `autot`?) continues to override the new target value set by from the Home Assistant UI. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Faikin device in HA 
2. Click on temperature to open up the control UI
3. Update the target temperature via the UI
4. Refresh the page
5. Notice that the target temperature changed back to the original value

**Expected behavior**

target temperature (`autot`) set to the new value from HA

**Additional context**

if I'm understanding the code correctly, we'll need to update the `temperature_command_topic` [here](https://github.com/revk/ESP32-Faikin/blob/24f46ba446bda6eaa2f5ef6266d1618be352b075/ESP32/main/Faikin.c#L2204) to `command/<device_name>/autot` when Faikin Auto mode is enabled.

revk, 2024-02-17:
I thought the back end handled the normal topic for that. I'll take a look.

revk, 2024-02-17:
Try latest beta in a few minutes.

closed 2024-02-17 by revk

shaohung, 2024-02-17:
Thanks for taking a look so fast.  I updated to the beta version and gave it a try, but it seems to still have the same behavior most of the time.  I was able to change the value 2 out of 10+ times. 

Here's the recent 3 messages to `command/faikin-living-room/temp`
```
Transmitted 2:00:20 AM
QoS: 0
Payload: 18.333333333333332

Transmitted 2:01:40 AM
QoS: 0
Payload: 21.11111111111111

Transmitted 2:04:37 AM
QoS: 0
Payload: 20.555555555555554
```
And here's the message HA received 
```
Received 2:04:37 AM
QoS: 0
Payload: online: true
target: 17
temp: 19.28
outside: 2.5
liquid: 22.5
comp: 0
fanfreq: 0
mode: heat
fan: auto
swing: 'off'
preset: home

Received 2:04:40 AM
QoS: 0
Payload: online: true
target: 17
temp: 19.28
outside: 2.5
liquid: 23
comp: 0
fanfreq: 0
mode: heat
fan: auto
swing: 'off'
preset: home

Received 2:04:41 AM
QoS: 0
Payload: online: true
target: 17
temp: 19.28
outside: 2.5
liquid: 23
comp: 0
fanfreq: 0
mode: heat
fan: auto
swing: 'off'
preset: home
```

revk, 2024-02-17:
Hmm, the settings code will probably barf at too many places. Can you try with one decimal place?

revk, 2024-02-17:
I have reverted my change as I found that I had, indeed, already got handling for setting `temp` to apply to `autot` when using *Faikin auto* mode.

So I suspect the issue is the settings logic does not like the number of places you are passing.

It may be I can do something with that, but it certainly was not expected anyone would send temperatures to 15 places!

revk, 2024-02-17:
OK I have made a beta to round to 1 decimal place before saving. Try that.

shaohung, 2024-02-17:
Yep, that was it. Problem with Fahrenheit conversion :(   Thanks again for the quick fix

commit 59d304f9 (2024-02-17): fix #242